UniversalRobots/Universal_Robots_ROS2_Driver #270: ursim_driver missing on the foxy branch

slabua, 2021-12-23:
Does it need to be back ported, perhaps?

destogl, 2022-01-04:
Hi @slabua,

can you plaease check if #272 is working for you?

slabua, 2022-01-05:
Hi @destogl,
Thank you, it's working for me.

One more question. Could you provide a use case for the ursim?
Currently, I have been using rviz as a front-end for moveit2, and I am able to see the mirrored movement into ursim by running the external control program.
Perhaps the usefulness of it is when it is used in the opposite direction? Namely, when controlling the robot from the sim itself, or?
Thank you.

destogl, 2022-01-18:
On the second terminal is actually command-line access to the execution. You can mostly ignore it.

ursim is used to test the robot's controller without an access to a physical robot. It provides you with dynamics and commands as real robot would do.

slabua, 2022-01-19:
@destogl thank you for the clarification~

closed 2022-01-19 by slabua