UniversalRobots/Universal_Robots_ROS2_Driver #271: Source build failling on foxy

urmahp, 2022-01-04:
I believe that the source build on foxy is failing because of this [line](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/foxy/.github/workflows/foxy-source-build.yml#L38). I think that the Universal_Robots_ROS2_Driver.repos is collected from the master branch instead of the foxy branch. I could suspect that a similar issue is present in the rolling source build. 

The pipeline fails because a function isn't member of the rclcpp::Duration, see log output below, the function is added on the master branch of ros2_control, but not the foxy branch. 

```bash
  [  4%] Building CXX object CMakeFiles/controller_manager.dir/src/controller_manager.cpp.o
  In file included from /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/src/ros2_control/controller_manager/src/controller_manager.cpp:15:
  /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/src/ros2_control/controller_manager/include/controller_manager/controller_manager.hpp:131:58: error: ‘from_nanoseconds’ is not a member of ‘rclcpp::Duration’
    131 |     const rclcpp::Duration & timeout = rclcpp::Duration::from_nanoseconds(kInfiniteTimeout));
        |                                                          ^~~~~~~~~~~~~~~~
  In file included from /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/install/rclcpp/include/rclcpp/node.hpp:1224,
                   from /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/install/rclcpp/include/rclcpp/executors/single_threaded_executor.hpp:28,
                   from /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/install/rclcpp/include/rclcpp/executors.hpp:22,
                   from /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/install/rclcpp/include/rclcpp/rclcpp.hpp:146,
                   from /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/install/controller_interface/include/controller_interface/controller_interface.hpp:28,
                   from /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/src/ros2_control/controller_manager/include/controller_manager/controller_manager.hpp:23,
                   from /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/src/ros2_control/controller_manager/src/controller_manager.cpp:15:
  /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/install/rclcpp/include/rclcpp/node_impl.hpp: In instantiation of ‘auto rclcpp::Node::declare_parameter(const string&, const ParameterT&, const ParameterDescriptor&, bool) [with ParameterT = rclcpp::ParameterType; std::string = std::__cxx11::basic_string<char>; rcl_interfaces::msg::ParameterDescriptor = rcl_interfaces::msg::ParameterDescriptor_<std::allocator<void> >]’:
  /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/src/ros2_control/controller_manager/src/controller_manager.cpp:206:74:   required from here
  /home/runner/work/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver/ros_ws/install/rclcpp/include/rclcpp/node_impl.hpp:169:23: error: no matching function for call to ‘rclcpp::ParameterValue::get<rclcpp::ParameterType>() const’
    169 |     ).get<ParameterT>();
```

aujincheng, 2022-01-21:
I just tried to colcon build the foxy branch, and the process is complete.

But when I tried to call the robots into rviz, by using:
`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true
`

It turns out to be an empty world without the universal robot arm.

fmauch, 2022-06-20:
This should be resolved by now.

PR #273 (2022-01-05): Fix foxy source build to use correct .repos file
Resolves #271

@urrsk we should try to merge this and then wait a day to see if everything gets green.
closed 2022-06-20 by fmauch