UniversalRobots/Universal_Robots_ROS2_Driver #274: No kinematics plugin defiend.

antopius, 2022-01-07:
# Environment
- ROS Distro: Galactic (Binary via Debian Packages)
- OS Version: Ubuntu 20.04
- Branch: 'main'
- Driver.repos: tried 'released' and 'not-released' (is 'released' the right one for me?)

# Description 
I'm very new to ROS and Github, so please feel free to correct me, if there is something unusual here.

After following the steps in [Getting Started](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/README.md#getting-started) and the moveit examples:

    ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false

    ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur5e robot_ip:="xxx.xxx" use_fake_hardware:=true launch_rviz:=true

the same Error as in #265 and #268 occurs. Then I have merged #268 by editing the launch files (also tried the 'small-launch-file-fixes' branch). With that the robot-model shows up in RViz, but the interactive-markers are missing. (see RViz)

<details>
  <summary> RViz </summary>

[image]

</details>

I think this Error is related to a warning in the console output from ur_moveit_launch.py 

    [WARN] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!

This message also appears in the 'main' branch (without #268).

Maybe the namespace collision is also a problem?

    SEVERE WARNING!!! A namespace collision has occured with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay.

For more details here is the full console output

<details>
  <summary> console output: ur_moveit_launch.py </summary>
    
```
1641497059.3043668 [INFO] [launch]: All log files can be found below /home/tony/.ros/log/2022-01-06-20-24-19-299322-antopius-102316
1641497059.3046501 [INFO] [launch]: Default logging verbosity is set to INFO
1641497060.9627249 [INFO] [move_group-1]: process started with pid [102324]
1641497060.9632144 [INFO] [mongo_wrapper_ros.py-2]: process started with pid [102326]
1641497060.9636350 [INFO] [rviz2-3]: process started with pid [102328]
1641497060.9638207 [INFO] [static_transform_publisher-4]: process started with pid [102330]
1641497060.9639544 [INFO] [servo_node_main-5]: process started with pid [102332]
1641497060.9931874 [static_transform_publisher-4] [INFO] [1641497060.992752781] [static_transform_publisher]: Spinning until killed publishing transform from 'world' to 'base_link'
1641497061.4333007 [rviz2-3] [INFO] [1641497061.432959786] [rviz2]: Stereo is NOT SUPPORTED
1641497061.4336653 [rviz2-3] [INFO] [1641497061.433056782] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
1641497061.4585600 [rviz2-3] [INFO] [1641497061.458234808] [rviz2]: Stereo is NOT SUPPORTED
1641497061.6414342 [rviz2-3] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
1641497061.6416452 [rviz2-3]          at line 253 in /opt/ros/galactic/include/class_loader/class_loader_core.hpp
1641497064.6672881 [rviz2-3] [ERROR] [1641497064.666863011] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
1641497064.7546139 [rviz2-3] [INFO] [1641497064.754259727] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00643986 seconds
1641497064.7550166 [rviz2-3] [INFO] [1641497064.754351377] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
1641497064.7800896 [rviz2-3] [WARN] [1641497064.779761664] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
1641497064.8443103 [rviz2-3] [INFO] [1641497064.843868459] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
1641497064.8462896 [rviz2-3] [INFO] [1641497064.845966941] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
1641497065.9465888 [rviz2-3] [INFO] [1641497065.946274839] [interactive_marker_display_94451179729456]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
1641497065.9497626 [rviz2-3] Link base had 0 children
1641497065.9500604 [rviz2-3] Link base_link_inertia had 1 children
1641497065.9501479 [rviz2-3] Link shoulder_link had 1 children
1641497065.9502277 [rviz2-3] Link upper_arm_link had 1 children
1641497065.9503055 [rviz2-3] Link forearm_link had 1 children
1641497065.9503813 [rviz2-3] Link wrist_1_link had 1 children
1641497065.9504697 [rviz2-3] Link wrist_2_link had 1 children
1641497065.9505465 [rviz2-3] Link wrist_3_link had 1 children
1641497065.9506197 [rviz2-3] Link flange had 1 children
1641497065.9506993 [rviz2-3] Link tool0 had 0 children
[... 54 lines cut ...]
1641496968.8193777 [spawner-4] [INFO] [1641496968.819093075] [spawner_joint_state_broadcaster]: [94mLoaded joint_state_broadcaster[0m
1641496968.8259437 [spawner-5] [INFO] [1641496968.824388669] [spawner_io_and_status_controller]: [92mConfigured and started [96mio_and_status_controller[0m
1641496968.8731582 [spawner-6] [INFO] [1641496968.872799921] [spawner_speed_scaling_state_broadcaster]: [94mLoaded speed_scaling_state_broadcaster[0m
1641496968.8746965 [spawner.py-8] [INFO] [1641496968.873418299] [spawner_forward_position_controller]: [94mLoaded forward_position_controller[0m
1641496968.8794971 [spawner-9] [INFO] [1641496968.879287389] [spawner_joint_trajectory_controller]: [94mLoaded joint_trajectory_controller[0m
1641496968.8888462 [spawner-7] [INFO] [1641496968.888534554] [spawner_force_torque_sensor_broadcaster]: [94mLoaded force_torque_sensor_broadcaster[0m
1641496968.8972573 [spawner-4] [INFO] [1641496968.896139981] [spawner_joint_state_broadcaster]: [92mConfigured and started [96mjoint_state_broadcaster[0m
1641496968.9285071 [spawner-6] [INFO] [1641496968.928199489] [spawner_speed_scaling_state_broadcaster]: [92mConfigured and started [96mspeed_scaling_state_broadcaster[0m
1641496968.9483364 [spawner-7] [INFO] [1641496968.948009787] [spawner_force_torque_sensor_broadcaster]: [92mConfigured and started [96mforce_torque_sensor_broadcaster[0m
1641496968.9691079 [spawner-9] [INFO] [1641496968.968791251] [spawner_joint_trajectory_controller]: [92mConfigured and started [96mjoint_trajectory_controller[0m
1641496968.9837098 [INFO] [spawner-5]: process has finished cleanly [pid 102055]
1641496969.0324657 [spawner.py-8] /home/tony/workspace2/ros_ur_driver/install/controller_manager/lib/controller_manager/spawner.py:204: DeprecationWarning: 'spawner.py' is deprecated, please use 'spawner' (without .py extension)
1641496969.0326426 [spawner.py-8]   warnings.warn(
1641496969.0470595 [INFO] [spawner.py-8]: process has finished cleanly [pid 102061]
1641496969.1052749 [INFO] [spawner-4]: process has finished cleanly [pid 102053]
1641496969.1531901 [INFO] [spawner-6]: process has finished cleanly [pid 102057]
1641496969.1586440 [INFO] [spawner-7]: process has finished cleanly [pid 102059]
1641496969.1642439 [INFO] [spawner-9]: process has finished cleanly [pid 102090]
```

</details>

Does anyone have the same problem or have i messed up something here?

I would appreciate any help!

closed 2022-01-08 by antopius

destogl, 2022-01-18:
@antopius is this still an issue or is this closed?

antopius, 2022-01-18:
@destogl 
It's closed for me as my solution works. But this is maybe more like a workaround.
I edited the title to the more related error message.
I expect that #265 is the same issue and my solution also works for @NebuDa.
This occurs also in the moveit2 Tutorial and only on Galactic, so could be something to do with the moveit Galactic branch.