UniversalRobots/Universal_Robots_ROS2_Driver #275: Servoing / TCP positioning demo

slabua, 2022-01-11:
Is there any package provided for testing the servoing behaviour?
MoveIt2 ships with a demo for the Panda robot so I thought perhaps there is something included in the UR driver as well, but I seem I can't find it.

AndyZe, 2022-01-11:
We added moveit_servo to the launch file recently.  (https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/239)

If you follow the README to launch MoveIt, then do `ros2 node list`, you should see it. Are you on Rolling or Foxy? Either way, I think you need a service call to activate the servo node:

`ros2 service call /servo_node/start_servo std_srvs/srv/Trigger {}`

You can send TwistStamped commands to it on this topic:  `/servo_node/delta_twist_cmds`

Servo tutorial here. (I'd suggest the Rolling version. I'm not sure Foxy is up to date)

https://moveit.picknik.ai/main/doc/realtime_servo/realtime_servo_tutorial.html

slabua, 2022-01-11:
@AndyZe Thanks for the information! I am using Foxy at the moment and I have been running the tutorial from that link with keyboard teleoperation. I was missing the call for the node which perhaps is not running by default on Foxy, I will check it later and update this issue.

As a note, I think the reason why I am using Foxy is that because on Galactic or Rolling, MoveIt2 wasn't playing well with Rviz, but it has been a while. Once I get things up and hopefully running as intended I might try switching back to Rolling and see how it performs.

slabua, 2022-01-11:
Foxy is missing `forward_velocity_controller` and `forward_position_controller`.
I will check on Rolling later.

slabua, 2022-01-12:
@AndyZe I have installed Rolling and recompiled the UR driver appropriately, once again with MoveIt support.
I have followed the steps mentioned in the referenced pull request, making sure that the correct robot controller is being used, and also used your simple_servo_twist_pub package to test the result. Although the data is being sent to the correct topic, the robot is not moving in rviz.
I will try testing with the actual robot.

AndyZe, 2022-01-12:
Did you remember the service call to activate Servo?

slabua, 2022-01-12:
Yes I did. Although it responds only after the first trigger. If I call it a second time it will wait forever, so I tried with and without the call but no change in the result.
Perhaps the node dies for some reason after the first call?

AndyZe, 2022-01-12:
Alright, I'll give it a quick test tomorrow. I have everything installed already.

slabua, 2022-01-12:
Thanks, I will also try on a different machine connected to the robot.

slabua, 2022-01-12:
@AndyZe I found my issue. For some reason the controller was listed as inactive, therefore even after switching to it, it had no use.
Launching ur_control with the robot_controller:=forward_position_controller seems to actually disable the controller. Launching it instead with the default controller and switching it afterwards seems to work as expected.
Thank you for your support!

closed 2022-01-12 by slabua

AndyZe, 2022-01-12:
Good to hear. I could have saved you some trouble with `ros2 control list_controllers`. Oh well, now the next person will know.  :+1:

slabua, 2022-01-13:
Thank you. One more thing about servoing in general.
As I understand, it is used to send incremental motion commands to the tcp, but is there a way to send the actual absolute coordinates in the robot space? For example with reference to the base link, or I am supposed to play with the tf?

AndyZe, 2022-01-13:
There's a feature called `pose tracker` that does that. It  uses PID controllers to seek the target pose.

(Target pose) --> (PID control) --> twist command --> Servo

There's a pretty good demo here:  https://github.com/ros-planning/moveit2/blob/main/moveit_ros/moveit_servo/src/cpp_interface_demo/pose_tracking_demo.cpp

You'll see the launch files are really similar to Servo so I don't think you'll have any trouble getting started.

slabua, 2022-01-13:
@AndyZe thanks! I'll have a look into that too~

slabua, 2022-01-18:
@AndyZe I could run the demo you linked above by its launch file for the panda robot and I understood the logic within it and the node itself, but perhaps I am missing something to be able to use it with the UR robot.

If I am not mistaken, the demo node itself should be robot-agnostic, so I tried to run the demo in the following manner:

I thought to add the missing part to the ur_moveit.launch.py file:
```
    # Pose Tracking node
    pose_tracking_yaml = load_yaml(
        "moveit_servo", "config/pose_tracking_settings.yaml")
    pose_tracking_params = {"moveit_servo": pose_tracking_yaml}

    pose_tracking_node = Node(
        package="moveit_servo",
        executable="servo_pose_tracking_demo",
        name="pose_tracking_node",
        # prefix=['xterm -e gdb -ex run --args'],
        output="screen",
        parameters=[
            robot_description,
            robot_description_semantic,
            kinematics_yaml,
            pose_tracking_params,
            servo_params,
            joint_limit_params,
        ],
    )
```
and return it accordingly together with the list of the other nodes to be started, then execute it as usual:
```
$ ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:="xxx" use_fake_hardware:=true launch_rviz:=false
$ ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur10e robot_ip:="xxx" use_fake_hardware:=true launch_rviz:=true

# with and without the following commands
$ ros2 control switch_controllers --start forward_position_controller --stop joint_trajectory_controller
$ ros2 service call /servo_node/start_servo std_srvs/srv/Trigger {}
```
but there is no pose_tracking_node.
Perhaps I am just missing something or I might as well just be on the wrong track.

AndyZe, 2022-01-18:
Did you also launch the Servo node? It should be one or the other-- Servo or pose tracking. Since pose tracking encapsulates Servo.

slabua, 2022-01-18:
Yes the servo node is already included in the moveit.launch file (the servo sine demo works with that after switching the controller).

slabua, 2022-01-19:
Hm, the `servo_node_main` node is not automatically launched by the `servo_pose_tracking_demo` node.

slabua, 2022-01-19:
@AndyZe I figured it out, I was sending an unstructured joint_limit context straight from joint_limit_params instead of reading it properly from the yaml file.
Thanks for your support~