UniversalRobots/Universal_Robots_ROS2_Driver #276: ROS2 Ur_Driver include pfads are missing

kolasha, 2022-01-11:
During installing ros2 UR_Driver for Galactic branch, I get the following error
lifecycle_msgs/msg/state.hpp: No such file or directory | #include "lifecycle_msgs/msg/state.hpp" 
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
->the package rcl_lifecycle_msgs is properly installed and dependencies in package.xml and Cmakelist are also correct

Does anyone have the same problem?

jonipol, 2022-01-22:
I had this same error when I tried to build the driver on top of ros2 galactic that I built from source. But when I tried building the driver on a docker image based on `ros:galactic`  the build passed without errors. I guess it is a issue with sourcing maybe? If someone has more insight would be interested to learn what is the issue here.

kolasha, 2022-01-23:
Hello jonipol:
i solved this problem by doing the following steps
At the first:
I'm reinstall the ros2 galactic and took the installation via Debian.
At the second:
---------->sudo apt install python3-colcon-common-extensions python3-vcstool
---------->export COLCON_WS=~/workspace/ros_ur_driver
---------->mkdir -p $COLCON_WS/src
---------->cd $COLCON_WS
---------->cd src
You should clone this link of the git repository:(over the button "code" copy)
---------->https://github.com/intel/ros2_grasp_library.git   (in your src)
---------->rosdep install --ignore-src --from-paths src -y -r

> do not build at the first:
jump to _MoveiT Installation and take the following repository step
-----vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/MoveIt_Support.repos.
Then you have to remove the following folder:
---------->rm -rf src/moveit2/
---------->rm -rf src/moveit_msgs/
---------->rm -rf src/geometric_shapes/
and install:
----------> sudo apt-get install ros-galactic-moveit* (in your src) (as descripted on the Movelt2 page)
than:
----------> rosdep update  and rosdep init and sudo apt-get update
and set:
----------> rosdep install --ignore-src --from-paths src -y -r –rosdistro=GALACTIC (if it is not done by ros2 installation )//(just for the case)

then:
----------> colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
(you should have 37 packesges:)

once the build is successful. Open new terminaln (this is very importante)
jump to ~/workspace/ros_ur_driver$ (and paste). install/setup.bash        //space behind the point 
in the same terminal:
----------> ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true

open an new Terminal: jump to  ~/workspace/ros_ur_driver$ (and paste). install/setup.bash
and put: 
----------> ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur5e robot_ip:="xxx.xxx" use_fake_hardware:=true launch_rviz:=true

i hope, it is helpfull

fmauch, 2023-05-09:
closing this, since it was more of a workspace setup thing and the correct way to handle this was posted.

closed 2023-05-09 by fmauch