UniversalRobots/Universal_Robots_ROS2_Driver #290: ros_planning error while building

taanielk, 2022-02-06:
Hello

I did a fresh Galactic+MoveIt install. While trying to build the moveit part of ur driver, I get this error: 

```
[Processing: moveit_ros_planning]                                          
--- stderr: moveit_ros_planning                                            
/home/taaniel/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp: In member function ‘void planning_scene_monitor::PlanningSceneMonitor::excludeWorldObjectFromOctree(const ObjectConstPtr&)’:
/home/taaniel/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:942:81: error: ‘using element_type = const struct collision_detection::World::Object’ {aka ‘const struct collision_detection::World::Object’} has no member named ‘global_shape_poses_’; did you mean ‘shape_poses_’?
  942 |       collision_body_shape_handles_[obj->id_].push_back(std::make_pair(h, &obj->global_shape_poses_[i]));
      |                                                                                 ^~~~~~~~~~~~~~~~~~~
      |                                                                                 shape_poses_
/home/taaniel/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp: In member function ‘bool planning_scene_monitor::PlanningSceneMonitor::getShapeTransformCache(const string&, const rclcpp::Time&, occupancy_map_monitor::ShapeTransformCache&) const’:
/home/taaniel/workspace/ros_ur_driver/src/moveit2/moveit_ros/planning/planning_scene_monitor/src/planning_scene_monitor.cpp:1134:47: error: ‘const class moveit::core::AttachedBody’ has no member named ‘getShapePosesInLinkFrame’
 1134 |             attached_body_shape_handle.first->getShapePosesInLinkFrame()[attached_body_shape_handle.second[k].second];
      |                                               ^~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/build.make:63: planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/src/planning_scene_monitor.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:872: planning_scene_monitor/CMakeFiles/moveit_planning_scene_monitor.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< moveit_ros_planning [2min 54s, exited with code 2]
                                         
Summary: 59 packages finished [17min 40s]
  1 package failed: moveit_ros_planning
  5 packages had stderr output: launch_param_builder moveit_configs_utils moveit_ros_planning ros2_control_test_nodes ros2controlcli
  19 packages not processed
```

How to solve it?
Thanks in advance.

destogl, 2022-02-07:
This is questions for the MoveIt2 and not here!

Still, I have a guess. Do you have MoveIt installed on your system? If so remove it.

taanielk, 2022-02-07:
Since this is the wrong place, I will close this question. I will edit this comment if I have this problem fixed.

closed 2022-02-07 by taanielk

destogl, 2022-02-07:
You can still write. Did you try my proposal?

taanielk, 2022-02-07:
Yes, I just got to my computer and did a new purge. It seems that I failed to purge old MoveIt completely and that messed up my compile. Thanks for the help.