UniversalRobots/Universal_Robots_ROS2_Driver #294: executable '{}' not found on the PATH

lydiazoghbi, 2022-02-17:
Hi! Any idea what could possibly be causing the following error? I'm running ubuntu 20.04, foxy, and all the packages were on foxy branch and built fine. ros2 and install/setup.bash are sourced as well. This is the first time I'm getting this error, I've had successful previous installations in the past. Thanks!

`imerse@imerse-Alienware-m15:~/ros2_ws/ros_ur_driver$ ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=yyy.yyy.yyy.yyy
[INFO] [launch]: All log files can be found below /home/imerse/.ros/log/2022-02-17-17-04-03-332691-imerse-Alienware-m15-170705
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure("executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0x7fea4bca47f0>]' not found on the PATH")>
Traceback (most recent call last):
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 1 more time]
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/home/imerse/ros2_foxy/install/launch_ros/lib/python3.8/site-packages/launch_ros/actions/node.py", line 440, in execute
    self._perform_substitutions(context)
  File "/home/imerse/ros2_foxy/install/launch_ros/lib/python3.8/site-packages/launch_ros/actions/node.py", line 395, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/home/imerse/ros2_foxy/install/launch_ros/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/home/imerse/ros2_foxy/install/launch_ros/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 71, in evaluate_parameter_dict
    evaluated_value = perform_substitutions(context, list(value))
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/launch_context.py", line 197, in perform_substitution
    return substitution.perform(self)
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/substitutions/command.py", line 94, in perform
    command_str = perform_substitutions(context, self.command)
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/launch_context.py", line 197, in perform_substitution
    return substitution.perform(self)
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/substitutions/path_join_substitution.py", line 45, in perform
    performed_substitutions = [sub.perform(context) for sub in self.__substitutions]
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/substitutions/path_join_substitution.py", line 45, in <listcomp>
    performed_substitutions = [sub.perform(context) for sub in self.__substitutions]
  File "/home/imerse/ros2_foxy/install/launch/lib/python3.8/site-packages/launch/substitutions/find_executable.py", line 66, in perform
    raise SubstitutionFailure("executable '{}' not found on the PATH".format(self.name))
launch.substitutions.substitution_failure.SubstitutionFailure: executable '[<launch.substitutions.text_substitution.TextSubstitution object at 0x7fea4bca47f0>]' not found on the PATH
`

destogl, 2022-02-23:
Maybe `xacro` or something. Check that you compiled and sourced your workspace before trying to execute this.

jackdario96, 2022-05-10:
Hey, I had the same issue with the Rolling distro.  The error message is kinda criptic but in my case I solved it installing the xacro package for ros2. 
`sudo apt install ros-<ros distro>-xacro`

fmauch, 2022-05-11:
This should not really be an issue since we have the xacro dependency in our [`package.xml`](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/b085d50a935a1e58d22099d056c40b91e8790124/ur_bringup/package.xml#L29) (same for rolling).

I would assume, that you didn't execute the `rosdep install ...` step or that step produced some errors? I'll mark this issue invalid and close it, as I cannot imagine how that should happen when following the installation / build instructions. If you have any insights in which case this can still happen or steps to reproduce it, please comment and / or reopen.

closed 2022-05-11 by fmauch

gavanderhoorn, 2022-05-11:
> I would assume, that you didn't execute the `rosdep install ...` step or that step produced some errors? I'll mark this issue invalid and close it, as I cannot imagine how that should happen when following the installation / build instructions.

the readme asks users to run `rosdep install --ignore-src --from-paths src -y -r`, which includes `-r`. This makes `rosdep` "continue installing despite errors".

I would recommend to remove that switch, as it makes diagnosing these kinds of issues much harder.

If `rosdep` simply failed with an error, it would be clear what's happening. With `-r`, it looks like things are OK, and only be carefully examining the output of `rosdep` would you discover it wasn't able to install something.