UniversalRobots/Universal_Robots_ROS2_Driver #296: Fresh galactic install fails to run examples

maffettone, 2022-02-18:
When trying to run the initial examples with ros2 galactic, I get the following error. 

I removed all other ros installations `sudo apt remove ~nros-foxy-* && sudo apt autoremove` and `sudo apt remove ~nros-galactic-* && sudo apt autoremove`, and started from scratch installing only galactic on the system. Using Ubuntu 20.04.3 LTS. 

```bash
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/pmm/.ros/log/2022-02-17-20-09-30-567585-ROS-Workstation-83035
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=SubstitutionFailure("executable 'spawner.py' not found on the libexec directory '/home/pmm/Development/ros_ur_driver/install/controller_manager/lib/controller_manager' ")>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 2 more times]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/actions/node.py", line 484, in execute
    ret = super().execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 837, in execute
    self.__expand_substitutions(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 681, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 681, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_context.py", line 197, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/substitutions/executable_in_package.py", line 84, in perform
    raise SubstitutionFailure(
launch.substitutions.substitution_failure.SubstitutionFailure: executable 'spawner.py' not found on the libexec directory '/home/pmm/Development/ros_ur_driver/install/controller_manager/lib/controller_manager' 
```

Inside that directory referenced in the error: 
```bash
$ ls -l
total 48
-rwxr-xr-x 1 pmm pmm 37720 Feb 17 19:40 ros2_control_node
-rwxrwxr-x 1 pmm pmm   415 Feb 17 20:08 spawner
-rwxrwxr-x 1 pmm pmm   419 Feb 17 20:08 unspawner
```

fmauch, 2022-02-18:
This seems to be related to #293

maffettone, 2022-02-18:
Thanks for the quick reply. Following the branch in #297 solved the issue.

closed 2022-02-23 by destogl