UniversalRobots/Universal_Robots_ROS2_Driver #299: Provide more suport material

Maltergate, 2022-02-19:
Hi there,

I'm trying to control a UR5 with ROS2 Galactic and provided drivers. Despite the telatively well done README, it's quite hard to understand how to use the drivers. It may be really nice and handy to integrate a few more hands-on tutorials on how to use UR robots (for example, set a Cartesian pose of the end-effector, or how to connect to the gazebo simulator, or how to use it with MoveIt).

Cheers.

destogl, 2022-02-20:
Hi Thomas,

did you manage to use the driver? If so, would you like to contribute some hands-on tutorial what you did? For us, is sometimes hard to provide good beginners tutorials because we simply know the driver and all its use-cases in and out.

As an open-source project, we live from contributions and the one you mentioned would be very, very welcome!

Maltergate, 2022-02-20:
Hi Denis,

Thanks very much for your reply. I did manage to install the driver and play a bit around using MoveIt and RViZ, but nothing more. I don't have access to the real robot yet, but my plan is to learn and be ready for when I'll have the robot.

I did the installation as explained and I can launch nodes using 
`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5 robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true`

However, in the [README](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/README.md?plain=1#L214), line214, why do we have `use_fake_hardware:=true`and then `(then start the external_control URCap program from the pendant, as described above)`? I'm confused.

I tried to set up the docker simulator (compiling everything with Ubuntu 20.04) but somehow the compilation fails (with ur_sim). I guess I'll open another ticket with appropriate debug logs attached. I tried with [Gazebo](urlhttps://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation) and MoveIt and couldn't get anything to move.

So, I come back to the original installation as described in the README, I launched `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5 robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true` then I listed the nodes to see what actions, what topics etc. were available, but I can't get a hint of what's going on. Would it be possible to specify for example how to send some joint positions to the robot?

I think I understood that using MoveIt allows performing further computation "under the hood" such as path-planning, collision avoidance, using the IK (that the ur_controller doesn't if I understood well... Though I'm not quite sure because I saw somewhere on the GitHub, can't find it again, that there are new ur_controllers using Cartesian coordinates for the end-effector... so it means that there is some IK somewhere). A second very helpful example for me would be to show how to set specific Cartesian pose (position orientation) of the end-effector using MoveIt (and, second example, using the ur_controller if it's possible).

Maybe all of that is already available somewhere, but I couldn't find it. If at some point I get to the point that I can do it myself, I'll be pleased to make a tutorial to explain it. Meanwhile, though, every bit of help is warmly welcome!

destogl, 2022-02-23:
@Maltergate and @benediktkreis, please check #300 and give your feedback on it. I tried to address some of your comments there. For the rest, see below.

> So, I come back to the original installation as described in the README, I launched ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5 robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true [...]

I addressed the first part in #300. The other part about sending command was already there. There are example scripts for default controllers that send commands. Please give me feedback, what would you exactly like to have there. The best would be exact comment to my PR (or even better, you can submit your proposal changes).

> I think I understood that using MoveIt allows performing further computation "under the hood" such as path-planning, collision avoidance, using the IK (that the ur_controller doesn't if I understood [...]

For all of this, you should have more knowledge about ros2_control and MoveIt. To make it short, currently ros2_control does not provide Cartesian control now. It will at some point, and then we will use IK from the UR controllers.

MoveIt is a different story. It uses IK as defined in robot setup/configuration. There are multiple options available. If you what to learn more about MoveIt, then check [moveit_tutorials](https://lmddgtfy.net/?q=moveit2_tutorials).

> I agree with @Maltergate. Some more hands on tutorials would be extremely useful, especially when it comes to end effectors.

What do you mean exactly by that? How to control endeffector through UR driver or how to control endeffector with ROS2? For this, we would need an example endeffector because the configuration is depending very much on hardware configuration.

> At the moment I'm struggling to understand the communication with the OnRobot RG2-FT gripper.

Generally, robot driver does not have anything to do with gripper. Except if you want to control it through URCap. This means that you also connect robot to UR IOs and control them by them. Or you can connect it using external cabling directly to the robot. In any case, you will need driver and controller for it. You can do some custom solution like hack something in python or use it properly through ros2_control. In any case, you will need to implement something. If you want to implement it by yourself, then the community could help you to answer your question (ros2_control is probably the best place for this). If you require additional guidance, feel free to contact me directly.

> What is the standard approach to control a gripper with the Universal_Robots_ROS2_Driver? As I understand it, the https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/202 differs between the standard and the e-series. I'm working with the standard UR5.

UR support can give you the best answer about this. For the control with ROS I described you approach above.

closed 2022-02-28 by fmauch

benediktkreis, 2022-03-01:
Sorry for the late reply. The new documentation merged from [#300](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/pull/300) looks better than before.

> What do you mean exactly by that? How to control endeffector through UR driver or how to control endeffector with ROS2? For this, we would need an example endeffector because the configuration is depending very much on hardware configuration.

I mean how to control an end effector with ROS2. I understand that UR doesn't sell grippers and that the UR driver focuses on controlling the UR arms, but most people won't use just the arm, they will mount a gripper to it in order to perform manipulation tasks. Franka Emika's Panda and UFACTORY's xArm have the advantage that they come with a standard gripper which is quite attractive if you want to work with ROS2 out of the box. This applies to working with the real robot as well as the simulated one.

Of course there are too many grippers out there to create a tutorial for all of them, but it would help to have at least one with a commonly used "standard" two-finger gripper (such as the OnRobot RG2 or the Robotiq 2F-85), because it can serve as a starting point for other gripper implementations. Since UR tries to ease the integration of add-on equipment like grippers with URCaps, maybe there is a generalized way to access URCaps for controlling grippers with ROS2? 

Furthermore, many people work with a simulation environment like Ignition or Gazebo. That is why it would be helpful to point out differences in a gripper tutorial between a simulated and a real environment (e.g. in the robot description and launch files).

I've recently stumbled over the [Universal_Robots_ROS_Tutorials](https://github.com/UniversalRobots/Universal_Robots_ROS_Tutorials). Maybe this would be a good place for a gripper tutorial which can be linked in the ROS2 driver repository.