UniversalRobots/Universal_Robots_ROS2_Driver #301: Galactic: Controller example Can't activate controller 'scaled_joint_trajectory_controller'

lmendyk, 2022-02-23:
When following the example I get the following error, which is the reason for  move it example to fail execute a plan.

`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false
[INFO] [launch]: All log files can be found below /home/lmendyk/.ros/log/2022-02-23-11-14-26-240247-lmendykLinux-17283
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [17287]
[INFO] [robot_state_publisher-2]: process started with pid [17289]
[INFO] [spawner-3]: process started with pid [17291]
[INFO] [spawner-4]: process started with pid [17293]
[INFO] [spawner-5]: process started with pid [17297]
[INFO] [spawner-6]: process started with pid [17306]
[INFO] [spawner.py-7]: process started with pid [17315]
[INFO] [spawner-8]: process started with pid [17321]
[ros2_control_node-1] [INFO] [1645611267.502405838] [controller_manager]: update rate is 500 Hz
[robot_state_publisher-2] Link base_link had 2 children
[robot_state_publisher-2] Link base had 0 children
[robot_state_publisher-2] Link base_link_inertia had 1 children
[robot_state_publisher-2] Link shoulder_link had 1 children
[robot_state_publisher-2] Link upper_arm_link had 1 children
[robot_state_publisher-2] Link forearm_link had 1 children
[robot_state_publisher-2] Link wrist_1_link had 1 children
[robot_state_publisher-2] Link wrist_2_link had 1 children
[robot_state_publisher-2] Link wrist_3_link had 2 children
[robot_state_publisher-2] Link flange had 1 children
[robot_state_publisher-2] Link tool0 had 0 children
[robot_state_publisher-2] Link ft_frame had 0 children
[robot_state_publisher-2] [INFO] [1645611267.485396000] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1645611267.485565494] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1645611267.485588164] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1645611267.485602485] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1645611267.485614468] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1645611267.485627241] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-2] [INFO] [1645611267.485639979] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1645611267.485652699] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1645611267.485665210] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1645611267.485677833] [robot_state_publisher]: got segment world
[robot_state_publisher-2] [INFO] [1645611267.485689902] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1645611267.485702807] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1645611267.485715166] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1645611267.970868036] [controller_manager]: Loading controller 'io_and_status_controller'
[ros2_control_node-1] [INFO] [1645611268.017414364] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[spawner-4] [INFO] [1645611268.028567146] [spawner_io_and_status_controller]: Loaded io_and_status_controller
[ros2_control_node-1] [INFO] [1645611268.033609067] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[ros2_control_node-1] [INFO] [1645611268.052736438] [controller_manager]: Configuring controller 'io_and_status_controller'
[ros2_control_node-1] [ERROR] [1645611268.068258268] [controller_manager]: Can't activate controller 'io_and_status_controller': Command interface with 'gpio/standard_digital_output_cmd_0' does not exist
[spawner-4] [INFO] [1645611268.071324449] [spawner_io_and_status_controller]: Configured and started io_and_status_controller
[spawner-6] [INFO] [1645611268.075902209] [spawner_force_torque_sensor_broadcaster]: Loaded force_torque_sensor_broadcaster
[ros2_control_node-1] [INFO] [1645611268.078464426] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'
[ros2_control_node-1] [INFO] [1645611268.089087930] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-5] [INFO] [1645611268.104143585] [spawner_speed_scaling_state_broadcaster]: Loaded speed_scaling_state_broadcaster
[ros2_control_node-1] [INFO] [1645611268.105320048] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[ros2_control_node-1] [INFO] [1645611268.105404952] [speed_scaling_state_broadcaster]: Publisher rate set to : 100.0 Hz
[spawner-6] [INFO] [1645611268.105533752] [spawner_force_torque_sensor_broadcaster]: Configured and started force_torque_sensor_broadcaster
[ros2_control_node-1] [ERROR] [1645611268.108163462] [controller_manager]: Can't activate controller 'speed_scaling_state_broadcaster': State interface with key 'speed_scaling/speed_scaling_factor' does not exist
[spawner-5] [INFO] [1645611268.113189044] [spawner_speed_scaling_state_broadcaster]: Configured and started speed_scaling_state_broadcaster
[spawner-3] [INFO] [1645611268.124001681] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-1] [INFO] [1645611268.135888304] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1645611268.135982756] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-3] [INFO] [1645611268.147439802] [spawner_joint_state_broadcaster]: Configured and started joint_state_broadcaster
[ros2_control_node-1] [INFO] [1645611268.192502222] [controller_manager]: Loading controller 'forward_position_controller'
[ros2_control_node-1] [INFO] [1645611268.209727902] [controller_manager]: Loading controller 'scaled_joint_trajectory_controller'
[spawner.py-7] [INFO] [1645611268.232307955] [spawner_forward_position_controller]: Loaded forward_position_controller
[ros2_control_node-1] [INFO] [1645611268.234826920] [controller_manager]: Configuring controller 'forward_position_controller'
[ros2_control_node-1] [INFO] [1645611268.235340852] [forward_position_controller]: configure successful
[spawner-8] [INFO] [1645611268.242768555] [spawner_scaled_joint_trajectory_controller]: Loaded scaled_joint_trajectory_controller
[ros2_control_node-1] [INFO] [1645611268.254174868] [controller_manager]: Configuring controller 'scaled_joint_trajectory_controller'
[ros2_control_node-1] [INFO] [1645611268.254305931] [scaled_joint_trajectory_controller]: Command interfaces are [position] and and state interfaces are [position velocity].
[ros2_control_node-1] [INFO] [1645611268.255379344] [scaled_joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ros2_control_node-1] [INFO] [1645611268.256499486] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2_control_node-1] [ERROR] [1645611268.265065187] [controller_manager]: Can't activate controller 'scaled_joint_trajectory_controller': State interface with key 'speed_scaling/speed_scaling_factor' does not exist
[spawner-8] [INFO] [1645611268.269128132] [spawner_scaled_joint_trajectory_controller]: Configured and started scaled_joint_trajectory_controller
[INFO] [spawner-6]: process has finished cleanly [pid 17306]
[INFO] [spawner-5]: process has finished cleanly [pid 17297]
[INFO] [spawner-4]: process has finished cleanly [pid 17293]
[spawner.py-7] /opt/ros/galactic/lib/controller_manager/spawner.py:204: DeprecationWarning: 'spawner.py' is deprecated, please use 'spawner' (without .py extension)
[spawner.py-7]   warnings.warn(
[INFO] [spawner-3]: process has finished cleanly [pid 17291]
[INFO] [spawner.py-7]: process has finished cleanly [pid 17315]`

destogl, 2022-02-23:
It seems that the interface "speed_scaling/speed_scaling_factor" is missing. I guess that you are missing [this backported commit](https://github.com/ros-controls/ros2_control/commit/89bc3ac5777d1e4de074185bf9000a7c590c3264).  

This is released though, but it is not yet synced in Galactic repositories. Please compile ros2_control manually from `galactic` branch.

fmauch, 2022-02-23:
You should be able to get it running with the non-scaled `joint_trajectory_controller`:

```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy \ 
    initial_joint_controller:=joint_trajectory_controller use_fake_hardware:=true launch_rviz:=false
```

lmendyk, 2022-02-23:
Thanks the "initial_joint_controller:=joint_trajectory_controller" has helped for starting the controller, but for moveit  (I have added blindly the initial_joint_controller:=joint_trajectory_controller to the launch commet for moveit as well ) I get the error.

`[move_group-1] [INFO] [1645616265.629773163] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-1] [INFO] [1645616265.629932645] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [ERROR] [1645616265.630272040] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: scaled_joint_trajectory_controller/follow_joint_trajectory
[move_group-1] [ERROR] [1645616265.630300838] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller scaled_joint_trajectory_controller
[move_group-1] [INFO] [1645616265.630331799] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1645616265.630561953] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
`

So it looks that it still tries to use "scaled_joint_trajectory_controller" for the plan execution. Can I change somehow to "joint_trajectory_controller" ?

destogl, 2022-02-23:
Check the MoveIt configuration under `ur_moveit_config/config/controllers.yaml` or something similar. There, a default controller for Moveit is defined.

> blindly the initial_joint_controller:=joint_trajectory_controller to the launch command for moveit as well ) I get the error.

This is terrible approach / life strategy. Do you also close your eyes before you cross a street? Please read next time first. Otherwise, you owe us a beer for each trivial issue we solve :smile:

prasuchit, 2022-06-02:
Hi,

I'm facing the same issue with moveit2. I have the ros2_control galactic branch built from source. I changed ur_moveit_config/config/controllers.yaml file to make joint_trajectory_controller the default controller. In fact I even tried commenting out all instances of scaled_joint_trajectory_controller from the yaml file. On the bringup side I pass initial_joint_controller:=joint_trajectory_controller this argument correctly.

Even after all this, I see:

[move_group-1] [ERROR] [1654202611.524652366] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: scaled_joint_trajectory_controller/follow_joint_trajectory
[move_group-1] [ERROR] [1654202611.524668362] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller scaled_joint_trajectory_controller
[move_group-1] [INFO] [1654202611.524688844] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1654202611.524806353] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED

Kindly help. Thanks.

prasuchit, 2022-06-04:
**UPDATE:**

I used the following commands:

`ros2 launch ur_bringup ur3e.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false`

and

`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true`

With the joint_trajectory_controller set as default controller in the controller.yaml and I see this now:

`[move_group-1] [ERROR] [1654378632.634635680] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: joint_trajectory_controller/follow_joint_trajectory
[move_group-1] [ERROR] [1654378632.634908001] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller joint_trajectory_controller
[move_group-1] [INFO] [1654378632.635163987] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-1] [INFO] [1654378632.640666648] [moveit_move_group_default_capabilities.move_action_capability]: Solution found but controller failed during execution
[rviz2-3] [INFO] [1654378632.997586567] [move_group_interface]: Plan and Execute request accepted
[rviz2-3] [INFO] [1654378633.097936641] [move_group_interface]: Plan and Execute request aborted
[rviz2-3] [ERROR] [1654378633.198100808] [move_group_interface]: MoveGroupInterface::move() failed or timeout reached`

I'm more confused now because it can't even load the unscaled controller.

fmauch, 2022-06-09:
I'll properly answer this later today, for now I'd like to link the [cross-post](https://answers.ros.org/question/401943/ros2-galactic-universal-robots-driver-controller-error/) on ROS-answers.

prasuchit, 2022-06-09:
Hi @fmauch, that post on ROS answers is mine too and they closed it saying someone from UR would answer it. Thanks.

gavanderhoorn, 2022-06-09:
> Hi @fmauch, that post on ROS answers is mine too and they closed it saying someone from UR would answer it. Thanks.

Note quite: I closed your cross-post because you'd already opened an issue here and given the fact it was a long weekend, it would be nice to give the maintainers of `Universal_Robots_ROS2_Driver` (ie: this repository) some time to respond to your issue on the tracker first.

If @fmauch prefers to answer your ROS Answers post, he should be able to re-open it.

fmauch, 2022-06-09:
@prasuchit when using 

```
ros2 launch ur_bringup ur3e.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false
```

the `scaled_joint_trajectory_controller` will be active. This can be verified by using `ros2 control list_controllers` (when the `ros-galactic-ros2controlcli` package is installed).

Now, without further modifications, it should be possible to run

```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true
```

However, I've just tested this on my local galactic setup and it seems to throw an `std::underflow_error` when executing the motion, which seems to be a bug. This works with a real robot, but apparently not with fake_hardware atm. This will be further investigated...

However, when launching the driver with
```
ros2 launch ur_bringup ur3e.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false initial_joint_controller:=joint_trajectory_controller
```

and switching the default moveit_controller in the `ur_moveit_config/config/controllers.yaml` file as stated above, things work when starting 

```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true
```

It looks as if you've missed to change the `initial_controller` when starting the robot after you've modified your moveit_default_controller.

prasuchit, 2022-06-09:
@fmauch I'll test that out and let you know ASAP. However, I also noticed that it fails to start the force_torque_sensor_broadcaster. The following is the exact error:

```
[ERROR] [spawner-6]: process has died [pid 27529, exit code 1, cmd '/opt/ros/galactic/lib/controller_manager/spawner force_torque_sensor_broadcaster --controller-manager /controller_manager --ros-args'].
```

I saw this while trying it with both real and fake hardware. Would this adversely affect the performance?

fmauch, 2022-06-09:
> 

Yes, this is known but unrelated to that problem.

prasuchit, 2022-06-10:
Hi @fmauch, the fix you mentioned worked. You can close this issue. But I'd appreciate it if you could release bug fixes for the other existing issues in the driver like the one I mentioned above. Thanks.

danzimmerman, 2022-06-13:
> However, I also noticed that it fails to start the force_torque_sensor_broadcaster. 

> > 
> 
> Yes, this is known but unrelated to that problem.

@fmauch Is there an existing issue that's tracking this? 

I'm having the same problem (on URSim if that's relevant)

fmauch, 2022-06-27:
#361 should have been fixed by now in #391, #392 and #411 

Closing, as the fake_hw scaled pos controller is tracked in #390

closed 2022-06-27 by fmauch