UniversalRobots/Universal_Robots_ROS2_Driver #305: Galactic: initial_positions.yaml has no effect on robot position presented in rviz

lmendyk, 2022-02-24:
when following the examples, the initial robot position presented in rviz is always “flat” (all joints has 0 value set).   That looks as if the initial_positions.yam has not effect as this files defines the shoulder_lift_join and wrist_1_joint to -1,57  (which should make the robot arm “stand up”).

Am I missing something or there is a bug in the config files processing?

fmauch, 2022-02-24:
I assume you  are starting it with `use_fake_hardware:=true`? The initial value is not yet supported for galactic. Also see https://github.com/ros-controls/ros2_control/pull/635

lmendyk, 2022-02-24:
Thanks for info. 

The issue refers also just for viewing the robot description - from the example:

`ros2 launch ur_description view_ur.launch.py ur_type:=ur5e`

fmauch, 2022-02-25:
> The issue refers also just for viewing the robot description - from the example:
>
> ros2 launch ur_description view_ur.launch.py ur_type:=ur5e

The `view_ur.launch.oy` file is meant for viewing the description only, which helps when your are actually changing the description. Therefore, any "starring" position doesn't necessarily make much sense.

fmauch, 2023-05-09:
I guess, this can be closed by now.

closed 2023-05-09 by fmauch