UniversalRobots/Universal_Robots_ROS2_Driver #315: unable to setup using foxy

assmaha, 2022-03-21:
Hello, 

I am a beginner and I need help whenever I do the 

colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

I get this error 
---
Failed   <<< ros2_control_demo_hardware [17.4s, exited with code 2]
Aborted  <<< controller_interface [21.6s]               
Aborted  <<< transmission_interface [30.1s] 

Summary: 13 packages finished [2min 37s]
  1 package failed: ros2_control_demo_hardware
  2 packages aborted: controller_interface transmission_interface
  3 packages had stderr output: controller_interface ros2_control_demo_hardware transmission_interface
  20 packages not processed

I made sure I am on the foxy branch. I don't know what is the issue here. 
I appreciate any help. Thanks

fmauch, 2022-03-21:
To further help you, it would be good if you could tell us exactly, what you did. What were the commands that you invoked?

altairve, 2022-04-12:
@assmaha Faced similar issue, I had to checkout the foxy branch in `ros2_control_demos` and `ros2_control` repos.

fmauch, 2022-04-12:
@altairve Same as above, a complete list of what you did (from creating your workspace until invoking the build command) would be helpful in order to help you fix your problems.

fmauch, 2022-05-06:
@altairve any updates on this? Otherwise I will close this issue.

altairve, 2022-05-06:
@fmauch  I think there is a misunderstanding , I do not have any issues, I just faced same problem as @assmaha , and what worked for me was  checking out the foxy branch in ros2_control_demos and ros2_control repos. Here is what I did that caused the build to fail:
1. `mkdir -p arm_control_ros2/src` (colcon ws)
2. `cd arm_control_ros2`
3. `git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver -b foxy`
4. `vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.${ROS_DISTRO}.repos`
5. `rosdep update`
6. `rosdep install --ignore-src --from-paths src -y -r`
7. `colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`

Resulting in 
Failed <<< ros2_control_demo_hardware [17.4s, exited with code 2]
Specifically the compilation error was with regards to some cpp file in `ros2_control_demos` package.

What I did to resolve the issue is to checkout the following branches 
`ros2_control` -> `foxy`
`ros2_control_demos` -> `foxy`
and repeating step 7

Please note this error and error resolution was based on 4 months old (01/2021) commits. I have not try to reproduce with latest changes.

fmauch, 2022-05-06:
@altairve thanks for the update. In that case I think i can happily close this, as this has been addressed by now in #200, #180 and #153

closed 2022-05-06 by fmauch