UniversalRobots/Universal_Robots_ROS2_Driver #317: Joint 'shoulder_pan_joint' has 9765923333140350858 command interfaces found. 2 expected.

adamleon, 2022-03-22:
Hello,

I've been trying to control a UR3 robot using ROS2. Currently I'm using ROS2 Galactic with Ubuntu 20.04.

When running 
`ros2 launch ur_bringup ur3.launch.py`
The ur_robot_driver dies
```
[ur_ros2_control_node-1] [FATAL] [1647936513.117391715] [URPositionHardwareInterface]: Joint 'shoulder_pan_joint' has 9765923333140350858 command interfaces found. 2 expected.
[ur_ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-1]   what():  Failed to initialize 'ur3'
[ERROR] [ur_ros2_control_node-1]: process has died [pid 2776, exit code -6, cmd '/home/adamleon/workspaces/ur_ws/install/ur_robot_driver/lib/ur_robot_driver/ur_ros2_control_node --ros-args --params-file /tmp/launch_params_ng0ohz4w --params-file /home/adamleon/workspaces/ur_ws/install/ur_bringup/share/ur_bringup/config/ur3_update_rate.yaml --params-file /home/adamleon/workspaces/ur_ws/install/ur_bringup/share/ur_bringup/config/ur_controllers.yaml'].
```
I would expect that it's a bug somewhere, but I'm unable to figure out where.

Thanks in advance.

fmauch, 2022-03-24:
When launching `ros2 launch ur_bringup ur3.launch.py` I get 

```
RuntimeError: Included launch description missing required argument 'robot_ip' (description: 'IP address by which the robot can be reached.'), given: []
```

which is what I would expect. When I launch it with a correct IP address everything seems fine. So, without a little context it might be difficult to track that down. Could you please tell us a bit more about your setup (workspace contents, branch names, make sure, everything is up-to-date, exact commands you executed).

adamleon, 2022-03-25:
I'm running galactic on Ubuntu 20.04 as told before.
I have freshly installed everything on a laptop, and the only thing I've installed is ros2-desktop, ros2-control and the UR library.

The UR library is installed in ~/workspaces/ur_ws, which is different from what is stated in the installation:
```
export COLCON_WS=~/workspace/ros_ur_driver
mkdir -p $COLCON_WS/src
```
but I reconned that it wouldn't be a major problem.

I have the UR3 directly connected to the laptop, and I'm able to ping it and use SSH to connect to it (and also ping my laptop).

The only issue that I've come across was when running:
`colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release`
After downloading the Moveit packages.

I got the error
`c++: fatal error: Killed signal terminated program cc1plus`
Which I figured out was because I ran out of memory.
I then ran
`colcon build --parallel-workers 1 --cmake-args -DCMAKE_CXX_FLAGS="--param ggc-min-expand-20"`
as explained in
[https://answers.ros.org/question/368106/c-fatal-moveit-2-error-killed-signal-terminated-program-cc1plus-compilation/?answer=373357#post-id-373357](https://answers.ros.org/question/368106/c-fatal-moveit-2-error-killed-signal-terminated-program-cc1plus-compilation/?answer=373357#post-id-373357)

I noticed that when running 
`ros2 launch ur_bringup ur5.launch.py robot_ip:=*any ip*`
I get the same error. I'm of course unable to connect to the UR robot, but this error is somewhere when loading the ros control command interfaces, which is before the driver connects to the robot.

fmauch, 2022-03-25:
How exactly did you install ros2-control? Did you build it from source or did you use binary packages? Using the wrong branches or combining binary and source installations might cause such problems for example.

fmauch, 2022-03-28:
That should not be an issue w.r.t. the issue shown. Again, could you please share the your workspace's contents with us?

Inside your colcon workspace's src folder, please execute

```
for repo in `find . -name ".git" -type d | sed 's/\/.git//' | sort`; do echo "=== $repo ==="; git rev-parse --abbrev-ref HEAD; git rev-parse --short HEAD; done
```

T2FF, 2022-03-31:
I have the same problem, when I execute

```
for repo in `find . -name ".git" -type d | sed 's/\/.git//' | sort`; do echo "=== $repo ==="; pushd -q $repo; git rev-parse --abbrev-ref HEAD; git rev-parse --short HEAD; popd -q; done
```
 in my src folder I get the following output:

```
=== ./geometric_shapes ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./moveit2 ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./moveit_msgs ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./moveit_resources ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./ompl ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./ros2_control ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./srdfdom ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./Universal_Robots_ROS2_Description ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./Universal_Robots_ROS2_Driver ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./ur_msgs ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
=== ./warehouse_ros_mongo ===
bash: pushd: -q: invalid number
pushd: usage: pushd [-n] [+N | -N | dir]
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
bash: popd: -q: invalid number
popd: usage: popd [-n] [+N | -N]
```

fmauch, 2022-03-31:
ok, that seems odd. Anyway, you could also go into the folders manually and execute `git rev-parse --abbrev-ref HEAD` and `git rev-parse --short HEAD` in there. Especially inside the the ur repos and ros2_control.

gavanderhoorn, 2022-03-31:
@T2FF: how did you download the source code?

T2FF, 2022-03-31:
> @T2FF: how did you download the source code?

I just followed the instructions from the section [getting-started](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#getting-started)

gavanderhoorn, 2022-03-31:
@fmauch: you could also use `vcs` for this:

```
vcs custom --git /path/to/workspace/src --args rev-parse --abbrev-ref HEAD
```

T2FF, 2022-03-31:
I get this as output:
```
vcs custom --git /home/workstation/workspace/ros_ur_driver/src --args rev-parse --abbrev-ref HEAD
...........
=== /home/workstation/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Description (git) ===
ros2
=== /home/workstation/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver (git) ===
main
=== /home/workstation/workspace/ros_ur_driver/src/geometric_shapes (git) ===
ros2
=== /home/workstation/workspace/ros_ur_driver/src/moveit2 (git) ===
main
=== /home/workstation/workspace/ros_ur_driver/src/moveit_msgs (git) ===
HEAD
=== /home/workstation/workspace/ros_ur_driver/src/moveit_resources (git) ===
ros2
=== /home/workstation/workspace/ros_ur_driver/src/ompl (git) ===
main
=== /home/workstation/workspace/ros_ur_driver/src/ros2_control (git) ===
HEAD
=== /home/workstation/workspace/ros_ur_driver/src/srdfdom (git) ===
ros2
=== /home/workstation/workspace/ros_ur_driver/src/ur_msgs (git) ===
ros2
=== /home/workstation/workspace/ros_ur_driver/src/warehouse_ros_mongo (git) ===
ros2
```

fmauch, 2022-03-31:
@T2FF from what I see, you seem not to be on the `galactic` branch of `ros2_control` that might be causing the issue.

The required version in the `.repos` file has been changed in the commit that you have currently checked out in the driver. So, if you setup your workspace with an earlier version, you might have pulled in the incorrect versions unintentionally.

Going to your src folder and executing

```
vcs import . --input Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.galactic.repos
```

might fix your problem. I am not sure about the MoveIt repos, as I am currently using the released versions rather than building from source. (You could do the same for ros2_control, though). My devel workspace for example only contains:

```
Universal_Robots_Client_Library
Universal_Robots_ROS2_Description
Universal_Robots_ROS2_Driver
ur_msgs
```

This works with ros2_control and MoveIt out of the box.

T2FF, 2022-03-31:
Ok i ran the command in the src folder and get this as output:

```
vcs import . --input Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.galactic.repos
......
=== ./Universal_Robots_Client_Library (git) ===
Cloning into '.'...
=== ./Universal_Robots_ROS2_Description (git) ===
From https://github.com/UniversalRobots/Universal_Robots_ROS2_Description
   e73980d..a584902  ros2       -> origin/ros2
 * [new tag]         1.2.7      -> 1.2.7
 * [new tag]         2.0.0      -> 2.0.0
Already on 'ros2'
Your branch is behind 'origin/ros2' by 2 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
=== ./control_msgs (git) ===
Cloning into '.'...
=== ./ros2_control (git) ===
From https://github.com/ros-controls/ros2_control
   ea26563..a11638d  master     -> origin/master
 * [new tag]         2.5.0      -> 2.5.0
Previous HEAD position was 407bbd1 2.1.0
Switched to a new branch 'galactic'
Branch 'galactic' set up to track remote branch 'galactic' from 'origin'.
=== ./ros2_controllers (git) ===
Cloning into '.'...
=== ./ur_msgs (git) ===

Already on 'ros2'
Your branch is up to date with 'origin/ros2'.

```

i guess i have to rebuild now ?

could the issue being generated because I just ran the command from Step 4 in [Getting Started](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#getting-started) without specifying the Galactic distribution? 

```diff
cd $COLCON_WS
git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver
-vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.${ROS_DISTRO}.repos
rosdep install --ignore-src --from-paths src -y -r
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
source install/setup.bash
```

Sorry that i have to ask everything in detail for each step but I am completely new to ROS and the concept of workspace builders

fmauch, 2022-03-31:
@T2FF yes, you should rebuild your workspace now. I hope, this will resolve your problems.

fmauch, 2022-04-07:
No, the `not-released` should be fine for galactic, as well (hence the `.${ROS_DISTRO}` part). If there is a problem with that, we should have a look, but this does get [built in CI](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/actions/workflows/galactic-binary-build.yml?branch=main) every night.

fmauch, 2022-05-06:
As this looks to be resolved, I'll go ahead and close this issue. Feel free to comment or reopen if my interpretation if this issue's state was wrong.

closed 2022-05-06 by fmauch