UniversalRobots/Universal_Robots_ROS2_Driver #318: URSim Docker Issue

T2FF, 2022-03-24:
Hello together,

I am having problems installing URsim as Docker. I followed the UR ROS2 Driver tutorial and everything works fine.

Now I want to simulate a UniversalRobot via URsim and control it via ROS2.

For this I followed the [linked tutorial](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_robot_driver/resources/ursim_driver/README.md). The installation of Docker Engine and Docker Compose also worked. However, what did not work is installing URsim via the command:
`sudo ROS2_DISTRO=galactic BUILD_MOVEIT2=0 docker-compose up --build.`

I ran the command in the following folder:
`/home/workstation/workspace/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/ursim_driver`

and get the following output:
```
Building ursim
unable to prepare context: unable to evaluate symlinks in Dockerfile path: lstat /home/workstation/workspace/ros_ur_driver/install/ur_robot_driver/share/ur_robot_driver/resources/ursim_driver/ursim/g5: no such file or directory
ERROR: Service 'ursim' failed to build : Build failed
```

I also tried the command in the src folder: 
`/home/workstation/workspace/ros_ur_driver/src/Universal_Robots_ROS2_Driver/ur_robot_driver/resources/ursim_driver`

and getting this error message:
```
Step 10/19 : RUN vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos
 ---> Running in 901a47bed5c1
usage: vcs import [-h] [--input FILE_OR_URL] [--force] [--shallow]
                  [--recursive] [--retry N] [--skip-existing] [--debug] [-w N]
                  [--repos]
                  [path]
vcs import: error: argument --input: can't open 'src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos': [Errno 2] No such file or directory: 'src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos'
The command '/bin/sh -c vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.repos' returned a non-zero code: 2
ERROR: Service 'driver' failed to build : Build failed
```
is there something I am doing wrong or did an installation not work properly after all?

I am using Ubuntu 20.04 in combination with ROS2 Galactic.

fmauch, 2022-03-25:
For using URSim inside docker, best use the Docker images from Universal Robots: https://hub.docker.com/u/universalrobots

fmauch, 2022-03-25:
We should update the documentation accordingly

T2FF, 2022-03-25:
> For using URSim inside docker, best use the Docker images from Universal Robots: https://hub.docker.com/u/universalrobots

Many thanks for your reply, that worked for me!

fmauch, 2022-03-25:
@T2FF If it is ok with you, I'd like to keep this open due to the documentation task connected with this.

closed 2022-04-25 by fmauch