UniversalRobots/Universal_Robots_ROS2_Driver #322: Unable to launch ur_moveit.launch.py

JokerYan, 2022-04-01:
I am recently migrating my project from ur ros1 driver to ros2 driver. I have executed the following commands:
```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5 robot_ip:=192.168.1.99 launch_rviz:=false

// on another terminal
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5 launch_rviz:=true
```

However, the rviz window for moveit will show up and close very quickly. The terminal output is the following:
```
[INFO] [launch]: All log files can be found below /home/zhiwen/.ros/log/2022-04-01-15-37-02-302355-zhiwen-Z170X-UD3-205486
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [205494]
[INFO] [mongo_wrapper_ros.py-2]: process started with pid [205496]
[INFO] [rviz2-3]: process started with pid [205498]
[INFO] [servo_node_main-4]: process started with pid [205500]
[servo_node_main-4] [WARN] [1648798623.277355726] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-4] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-4] to the Servo composable node in the launch file
[move_group-1] [WARN] [1648798623.277990981] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[move_group-1] [WARN] [1648798623.278038880] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] Stack trace (most recent call last):
[servo_node_main-4] Stack trace (most recent call last):
[move_group-1] #14   Object "", at 0xffffffffffffffff, in 
[servo_node_main-4] #13   Object "", at 0xffffffffffffffff, in 
[servo_node_main-4] #12   Object "/opt/ros/galactic/lib/moveit_servo/servo_node_main", at 0x555f925b6ced, in 
[move_group-1] #13   Object "/opt/ros/galactic/lib/moveit_ros_move_group/move_group", at 0x55ddb53940ad, in 
[move_group-1] #12   Source "../csu/libc-start.c", line 308, in __libc_start_main [0x7fb001afd0b2]
[move_group-1] #11   Object "/opt/ros/galactic/lib/moveit_ros_move_group/move_group", at 0x55ddb53935e7, in 
[move_group-1] #10   Object "/opt/ros/galactic/lib/libmoveit_cpp.so.2.3.4", at 0x7fb002422665, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&)
[move_group-1] #9    Object "/opt/ros/galactic/lib/libmoveit_cpp.so.2.3.4", at 0x7fb00242013f, in moveit_cpp::MoveItCpp::loadPlanningSceneMonitor(moveit_cpp::MoveItCpp::PlanningSceneMonitorOptions const&)
[servo_node_main-4] #11   Source "../csu/libc-start.c", line 308, in __libc_start_main [0x7ff04a8be0b2]
[servo_node_main-4] #10   Object "/opt/ros/galactic/lib/moveit_servo/servo_node_main", at 0x555f925b6bb4, in 
[servo_node_main-4] #9    Object "/opt/ros/galactic/lib/libservo_node.so", at 0x7ff04b08c97c, in moveit_servo::ServoNode::ServoNode(rclcpp::NodeOptions const&)
[move_group-1] #8    Object "/opt/ros/galactic/lib/libmoveit_planning_scene_monitor.so.2.3.4", at 0x7fb00232881a, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #7    Object "/opt/ros/galactic/lib/libmoveit_planning_scene_monitor.so.2.3.4", at 0x7fb00232876e, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::shared_ptr<planning_scene::PlanningScene> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #6    Object "/opt/ros/galactic/lib/libmoveit_robot_model_loader.so.2.3.4", at 0x7fb0014e29dd, in robot_model_loader::RobotModelLoader::RobotModelLoader(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
[move_group-1] #5    Object "/opt/ros/galactic/lib/libmoveit_robot_model_loader.so.2.3.4", at 0x7fb0014e016a, in robot_model_loader::RobotModelLoader::configure(robot_model_loader::RobotModelLoader::Options const&)
[servo_node_main-4] #8    Object "/opt/ros/galactic/lib/libmoveit_planning_scene_monitor.so.2.3.4", at 0x7ff04a6da81a, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[servo_node_main-4] #7    Object "/opt/ros/galactic/lib/libmoveit_planning_scene_monitor.so.2.3.4", at 0x7ff04a6da76e, in planning_scene_monitor::PlanningSceneMonitor::PlanningSceneMonitor(std::shared_ptr<rclcpp::Node> const&, std::shared_ptr<planning_scene::PlanningScene> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #4    Object "/opt/ros/galactic/lib/libmoveit_rdf_loader.so.2.3.4", at 0x7fb000ef36fd, in rdf_loader::RDFLoader::RDFLoader(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, double)
[move_group-1] #3    Object "/opt/ros/galactic/lib/libmoveit_rdf_loader.so.2.3.4", at 0x7fb000ef1da8, in rdf_loader::RDFLoader::loadFromStrings()
[servo_node_main-4] #6    Object "/opt/ros/galactic/lib/libmoveit_robot_model_loader.so.2.3.4", at 0x7ff04a0ae9dd, in robot_model_loader::RobotModelLoader::RobotModelLoader(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
[servo_node_main-4] #5    Object "/opt/ros/galactic/lib/libmoveit_robot_model_loader.so.2.3.4", at 0x7ff04a0ac16a, in robot_model_loader::RobotModelLoader::configure(robot_model_loader::RobotModelLoader::Options const&)
[move_group-1] #2    Object "/home/zhiwen/ur_ros2/install/srdfdom/lib/libsrdfdom.so.2.0.3", at 0x7fb000fc7c02, in srdf::Model::initString(urdf::ModelInterface const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[move_group-1] #1    Object "/home/zhiwen/ur_ros2/install/srdfdom/lib/libsrdfdom.so.2.0.3", at 0x7fb000fc792c, in srdf::Model::initXml(urdf::ModelInterface const&, tinyxml2::XMLElement*)
[move_group-1] #0    Object "/home/zhiwen/ur_ros2/install/srdfdom/lib/libsrdfdom.so.2.0.3", at 0x7fb000fbe0b8, in srdf::Model::clear()
[move_group-1] Segmentation fault (Address not mapped to object [0x6e69676972af])
[servo_node_main-4] #4    Object "/opt/ros/galactic/lib/libmoveit_rdf_loader.so.2.3.4", at 0x7ff049c416fd, in rdf_loader::RDFLoader::RDFLoader(std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool, double)
[servo_node_main-4] #3    Object "/opt/ros/galactic/lib/libmoveit_rdf_loader.so.2.3.4", at 0x7ff049c3fda8, in rdf_loader::RDFLoader::loadFromStrings()
[servo_node_main-4] #2    Object "/home/zhiwen/ur_ros2/install/srdfdom/lib/libsrdfdom.so.2.0.3", at 0x7ff049e1cc02, in srdf::Model::initString(urdf::ModelInterface const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[servo_node_main-4] #1    Object "/home/zhiwen/ur_ros2/install/srdfdom/lib/libsrdfdom.so.2.0.3", at 0x7ff049e1c92c, in srdf::Model::initXml(urdf::ModelInterface const&, tinyxml2::XMLElement*)
[servo_node_main-4] #0    Object "/home/zhiwen/ur_ros2/install/srdfdom/lib/libsrdfdom.so.2.0.3", at 0x7ff049e130b8, in srdf::Model::clear()
[servo_node_main-4] Segmentation fault (Address not mapped to object [0x6e69676972af])
[mongo_wrapper_ros.py-2] [INFO] [1648798623.387420736] [mongodb]: Starting mongodb with db location /tmp/db listening on localhost:33829
[mongo_wrapper_ros.py-2] 2022-04-01T15:37:03.420+0800 I CONTROL  [initandlisten] MongoDB starting : pid=205536 port=33829 dbpath=/tmp/db 64-bit host=zhiwen-Z170X-UD3
[mongo_wrapper_ros.py-2] 2022-04-01T15:37:03.420+0800 I CONTROL  [initandlisten] db version v3.6.8
[mongo_wrapper_ros.py-2] 2022-04-01T15:37:03.420+0800 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
[... cut ...] -params-file /tmp/launch_params_6h0r1bof --params-file /tmp/launch_params_tz4u_mr1 --params-file /tmp/launch_params_9ugbw29w --params-file /home/zhiwen/ur_ros2/install/ur_moveit_config/share/ur_moveit_config/config/kinematics.yaml'].
```

If I run the bringup with rviz, the pose of my robot is correctly reflected on it. Also, the same error arises if I use fake_hardware.

ROS version: ROS2 Galactic
Ubuntu: 20.04

Please let me know if there is any other information that I can provide. Thank you!

JokerYan, 2022-04-01:
Another important information is that I could not install Moveit2 following the instruction in the repo. When I execute:
```
cd $COLCON_WS
vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/MoveIt_Support.repos
vcs import src --skip-existing --input src/moveit2/moveit2.repos
rosdep install --ignore-src --from-paths src -y -r
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
source install/setup.bash
```
I get some error like
```
/ur_ros2/src/moveit2/moveit_ros/benchmarks/src/BenchmarkExecutor.cpp:1114:31: error: ‘MOVEIT_VERSION_STR’ was not declared in this scope; did you mean ‘MOVEIT_VERSION_EXTRA’?
 1114 |   out << "MoveIt version " << MOVEIT_VERSION_STR << '\n';
      |                               ^~~~~~~~~~~~~~~~~~
      |                               MOVEIT_VERSION_EXTRA
make[2]: *** [CMakeFiles/moveit_ros_benchmarks.dir/build.make:76: CMakeFiles/moveit_ros_benchmarks.dir/src/BenchmarkExecutor.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:109: CMakeFiles/moveit_ros_benchmarks.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

What I did instead, is to install the moveit2 separately following their instructions. Is it because I did not properly import the moveit2 for the ur driver? Thanks a lot

fmauch, 2022-04-01:
We should update the README, as for galactic you can indeed install MoveIt from the binary packages and you should probably not build it from the branches noted in the `MoveIt_Support.repos` repos.

How did you end up installing MoveIt?

JokerYan, 2022-04-01:
Thank you for the reply. 

I did the following steps:
1. use vcs import all the repos
2. remove the moveit2 folder in src
3. colcon build
4. install moveit2 binary

However, do I need to import the moveit2 from the driver workspace somehow?

Thank you!

fmauch, 2022-04-01:
If you installed MoveIt from the binary package, things should be working out of the box.

However, as you tried to build MoveIt from source first, there might be some MoveIt targets left around in your workspace. E.g. in your output, it seems it is using srdfdom built from source, that might be a problem. Try removing all MoveIt releated projects from your workspace (A minimal workspace should only contain `ur_msgs`, `Universal_Robots_ROS2_Description`, `Universal_Robots_ROS2_Driver`) and then rebuild your workspace.

JokerYan, 2022-04-01:
That is indeed the problem. Thank you very much!

I remove the moveit2 and srdfdom from the repos file. It seems that the mongo server needs to be built for the moveit script to work.

ludo2811, 2022-04-20:
Hey everyone,
I had the same issue and was able to solve it with the short instruction of JokerYan with removing the moveit2 and srdfdom folders from src. Then colcon build worked out pretty well. Then I had the problem with launching moveit. Everytime I executed the command "ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5 launch_rviz:=true", rviz opens but without showing the real robot. There is just an empty space. Do you have any suggestions what could be the problem here? The connection between the pc and the robot was successful and earlier tests with "test_scaled_joint_trajectory_controller" worked out too. So I think it is not a connection problem...I hope you can help:)

fmauch, 2022-05-06:
I'm going to close this, as the initial questions seems to be resolved. @ludo2811 please open a new issue if it still exists explaining what exactly you did and what your are experiencing (maybe an RViz screenshot?)

closed 2022-05-06 by fmauch