UniversalRobots/Universal_Robots_ROS2_Driver #327: Unable to launch ur_bringup test_joint_trajectory_controller.launch.py

maffettone, 2022-04-06:
Issue
====
I am trying to start using ROS for work with a UR3e and running into difficulties in the initial set-up. My end goal here is to do most of my development in a simulated environment without the actual UR3e or teach pendant. 
I can successfully launch the fake hardware in rviz; however, receive warning when I attempt to test the joint trajectory controllers, and no movement in rviz. 

`[publisher_joint_trajectory_controller-1] [WARN] [1649259061.755820186] [publisher_scaled_joint_trajectory_controller]: Start configuration is not within configured limits!`

There are also some errors appearing in the initial bringup logging. 

Thank you for any help here!

System
======
ROS: Melodic
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

Traceback
========
Initial bringup
```
$ ros2 launch ur_bringup ur3e.launch.py robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true
[INFO] [launch]: All log files can be found below 
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_stopper_node-2]: process started with pid [46555]
[INFO] [ros2_control_node-1]: process started with pid [46553]
[INFO] [robot_state_publisher-3]: process started with pid [46557]
[INFO] [rviz2-4]: process started with pid [46559]
[INFO] [spawner-5]: process started with pid [46561]
[INFO] [spawner-6]: process started with pid [46563]
[INFO] [spawner-7]: process started with pid [46565]
[INFO] [spawner-8]: process started with pid [46567]
[INFO] [spawner-9]: process started with pid [46569]
[INFO] [spawner-10]: process started with pid [46599]
[ros2_control_node-1] [INFO] [1649258887.015716145] [controller_manager]: update rate is 500 Hz
[robot_state_publisher-3] Link base_link had 2 children
[robot_state_publisher-3] Link base had 0 children
[robot_state_publisher-3] Link base_link_inertia had 1 children
[robot_state_publisher-3] Link shoulder_link had 1 children
[robot_state_publisher-3] Link upper_arm_link had 1 children
[robot_state_publisher-3] Link forearm_link had 1 children
[robot_state_publisher-3] Link wrist_1_link had 1 children
[robot_state_publisher-3] Link wrist_2_link had 1 children
[robot_state_publisher-3] Link wrist_3_link had 2 children
[robot_state_publisher-3] Link flange had 1 children
[robot_state_publisher-3] Link tool0 had 0 children
[robot_state_publisher-3] Link ft_frame had 0 children
[robot_state_publisher-3] [INFO] [1649258887.011347097] [robot_state_publisher]: got segment base
[robot_state_publisher-3] [INFO] [1649258887.011476578] [robot_state_publisher]: got segment base_link
[robot_state_publisher-3] [INFO] [1649258887.011493380] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-3] [INFO] [1649258887.011504660] [robot_state_publisher]: got segment flange
[robot_state_publisher-3] [INFO] [1649258887.011515457] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-3] [INFO] [1649258887.011525952] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-3] [INFO] [1649258887.011536636] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-3] [INFO] [1649258887.011547622] [robot_state_publisher]: got segment tool0
[robot_state_publisher-3] [INFO] [1649258887.011557997] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-3] [INFO] [1649258887.011569186] [robot_state_publisher]: got segment world
[robot_state_publisher-3] [INFO] [1649258887.011579601] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-3] [INFO] [1649258887.011590258] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-3] [INFO] [1649258887.011600483] [robot_state_publisher]: got segment wrist_3_link
[controller_stopper_node-2] [INFO] [1649258887.002884336] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[controller_stopper_node-2] [INFO] [1649258887.022690319] [Controller stopper]: Service available
[controller_stopper_node-2] [INFO] [1649258887.022722883] [Controller stopper]: Waiting for list controllers service to come up on controller_manager/list_controllers
[controller_stopper_node-2] [INFO] [1649258887.022741685] [Controller stopper]: Service available
[ros2_control_node-1] [INFO] [1649258887.199332130] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1649258887.213178053] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[spawner-5] [INFO] [1649258887.222247026] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-1] [INFO] [1649258887.226668299] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1649258887.226751305] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[... 75 lines cut ...]
- wrist_2_joint
- wrist_3_joint
position:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
velocity:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
effort:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
---
```

adnan-saood, 2022-04-06:
Hello, you can edit the `=Universal_Robots_ROS2_Driver/ur_bringup/config/test_goal_publishers_config.yaml` file and make the starting pose check false.
I assume you are not using melodic as the ros_info indicates ros2.

maffettone, 2022-04-06:
I'll take a look there, and have fixed the ros version to galactic, my mistake!

maffettone, 2022-04-06:
Setting `check_starting_point: false` gets rid of the warning; however, the robot still does not move in RViz. No other info is printed  to either log, so I presume it has something to do with the errors in the first call. 

```
[ros2_control_node-1] [ERROR] [1649258887.298834812] [controller_manager]: Can't activate controller 'scaled_joint_trajectory_controller': State interface with key 'speed_scaling/speed_scaling_factor' does not exist
```
```[ros2_control_node-1] [ERROR] [1649258887.354752839] [controller_manager]: Can't activate controller 'io_and_status_controller': Command interface with 'gpio/standard_digital_output_cmd_0' does not exist
```
```
[ros2_control_node-1] [ERROR] [1649258887.383622846] [controller_manager]: Can't activate controller 'speed_scaling_state_broadcaster': State interface with key 'speed_scaling/speed_scaling_factor' does not exist
```

adnan-saood, 2022-04-06:
In the **xacro** file `Universal_Robots_ROS2_Description/urdf/ur.ros2_control.xacro` you will see that these will not be present in the description if `use_fake_hardware:=true`, you can verify that by running `ros2 control list_hardware_interfaces`, the result will only show position and velocity interface for the joints.

you can replace the scaled_joint_trajectory_controller with joint_trajectory_controller in the launch file and it should work (i guess).

maffettone, 2022-04-06:
Unfortunately that didn't work. I tried both passing it as a parameter `initial_joint_controller:=joint_trajectory_controller` and changing the launch file defaults directly then rebuilding with colcon. 
The same 3 errors persist in the logs.

It's worth noting I also can't execute plans made with MoveIt, but expect it's probably worth starting here.

fmauch, 2022-04-07:
I'll have another look at fake_hardware in combination with the test motions.

Meanwhile: another valid approach would be to use URSim for simulating the robot. This would give you the benefits of having the same interfaces available (not only regarding control) as the real robot. To switch to a real robot all you would have to do is changing the robot's IP address.

I usually launch my URSim using
```
docker run --rm -it -p 5900:5900 -p 6080:6080 -v /home/mauch/.ursim/urcaps:/urcaps -v /home/mauch/.ursim/programs:/ursim/programs --name ursim universalrobots/ursim_e-series
```
and then connect the driver to that using
```
ros2 launch ur_bringup ur5e.launch.py robot_ip:=172.17.0.2
```
See the [Dockerfile's documentation](https://hub.docker.com/r/universalrobots/ursim_e-series) for why I chose the parameters like that.

maffettone, 2022-04-07:
This was very helpful! Following the setup of the External Control URCap using VNC viewer, I was able to get a mirrored visualization between URSim and RViz, and get Moveit to plan and execute movement.

However, neither the `test_joint_trajectory_controller.launch.py` nor the `test_scaled_joint_trajectory_controller.launch.py` seemed to have any effect. They reported the same warnings posted in the original issue. I'm happy to consider this issue resolved, but am still confused by the controller tests.

fmauch, 2022-04-08:
I assume that you are referring to the `Start configuration is not within configured limits!` errors.

The test files expect the robot to be close to a hardcoded starting pose. This is rather important, as moving to the first pose of the test motion might otherwise lead to catastrophic failures depending on the robot's mounting pose and obstacles within its workspace.

As mentioned [above](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/327#issuecomment-1090424074) you can disable the starting pose check inside the [test_goal_publishers_config.yaml file](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_bringup/config/test_goal_publishers_config.yaml#L21). Alternatively, you can move the robot to the configured starting pose `[0, -90°, 0, -90°, 0, 0]` manually, then start the external_control on the teach pendant and then execute the `test_scaled_joint_trajectory_controller.launch.py` file.

maffettone, 2022-04-08:
Thank you! Setting the initial position and ensuring that the external control was running on the teach pendant solved it.

closed 2022-04-08 by maffettone