UniversalRobots/Universal_Robots_ROS2_Driver #328: Incorrect state interface name in error message

stephanie-eng, 2022-04-06:
The error messages for having an incorrect state interface for all state interfaces say that position is expected for the state interfaces. The velocity and effort state interfaces should probably say that velocity/effort is expected instead of position.

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/ff3517dff7bd69fafd755b07b6fecc31b0a3ebc5/ur_robot_driver/src/hardware_interface.cpp#L109-L120

fmauch, 2022-04-07:
`URPositionHardwareInterface` is the name of the class and therefore, the reference in the logger is not wrong. I agree that this class naming is not ideal, though.

stephanie-eng, 2022-04-07:
No, I mean for this line:

https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/ff3517dff7bd69fafd755b07b6fecc31b0a3ebc5/ur_robot_driver/src/hardware_interface.cpp#L112

To be

`joint.state_interfaces[1].name.c_str(), hardware_interface::HW_IF_VELOCITY);`

And similar for effort

fmauch, 2022-04-07:
Ah, thanks for clarifying. You're absolutely right!

closed 2022-04-07 by fmauch