UniversalRobots/Universal_Robots_ROS2_Driver #330: Support for cartesian commands

diogoalmeida, 2022-04-07:
Hi!

In the docs, under `features.md`, "Cartesian position-based control" and "Cartesian twist-based control" are marked as not part of the ROS2 driver.

1. Are there plans to change this? If yes, are they public/is there a timeline?
2. If not, why not? Is that decision documented?

Thank you for your time :)

fmauch, 2022-04-08:
This is currently not very high on our priority list, as the interfaces aren't available in ros2_control, at all.

diogoalmeida, 2022-04-08:
Hi @fmauch, thanks for your reply.

> the interfaces aren't available in ros2_control, at all.

Does this mean that ros2_control does not support cartesian commands _at all_? As in it is not possible to define something like a `System` hardware component that offers the possibility to set a 6DOF twist, for example?

Or is it specific to the UR and interfaces such as `SpeedL` aren't available to the ros2_control driver?

fmauch, 2022-04-08:
> Does this mean that ros2_control does not support cartesian commands _at all_? As in it is not possible to define something like a `System` hardware component that offers the possibility to set a 6DOF twist, for example?

As far as I know, yes.

ametrix, 2022-04-11:
Hello @fmauch  

I see that you are listed as maintainer of https://github.com/UniversalRobots/Universal_Robots_ROS_controllers_cartesian.
Are you plan to port this to ROS 2 ?

destogl, 2022-04-24:
> > Does this mean that ros2_control does not support cartesian commands _at all_? As in it is not possible to define something like a `System` hardware component that offers the possibility to set a 6DOF twist, for example?
> 
> As far as I know, yes.

As maintainer of `ros2_control` I would not agree on this. @diogoalmeida if your hardware driver (UR hardware driver) supports/exports those interfaces, you can configure JointTrajectoryController to use those interfaces instead of joints. There could be minor tweaking necessary, but is should work without much effort. 

If you want to tackle this, I'll be glad to support you with my knowlege.

diogoalmeida, 2022-04-25:
Hi @destogl thanks for your comment. 

>  you can configure JointTrajectoryController to use those interfaces instead of joints.

Interesting! I am not sure how that would work for trajectories. I managed to adjust one of the ros2 control demos to forward "cartesian" commands (linear and angular velocity for the robot base) to the diffbot example, which required some modifications to the robot hardware driver. 

I'd expect that the ur driver would require something like that. It is already detecting whether it needs to send `SpeedJ` or `ServoJ` commands when it does a command mode switch (https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/082dd2a90e0b2ac8b69d69bacf662806a7a572b9/ur_robot_driver/src/hardware_interface.cpp#L774). 

If we add a cartesian command mode, I'd expect the driver to be able to detect it and then use, for example, the `SpeedL` command for a cartesian twist command mode.

Was something like this that you had in mind?

destogl, 2022-04-25:
> Was something like this that you had in mind?

Yes, exactly! You should add other interfaces for Cartesian commands, and in `prepare_` and `perform_command_mode_switch` you can detect mode you are in.

> I managed to adjust one of the ros2 control demos to forward "cartesian" commands (linear and angular velocity for the robot base) to the diffbot example, which required some modifications to the robot hardware driver.

This sounds fascinating! Would you be so kind to share this code with us in form of a new example for ros2_control_demos repository? This is very valuable for many users.

JaredRosenbaum, 2025-06-23:
Hi @fmauch - any news on the addition of the twist (and other cartesian) controllers to the ROS2 driver? Does this remain a low priority?

urfeex, 2025-06-25:
> Hi [@fmauch](https://github.com/fmauch) - any news on the addition of the twist (and other cartesian) controllers to the ROS2 driver? Does this remain a low priority?

At the moment, yes.

urfeex, 2026-11-16:
As a short update here:

- The twist_controller has been ported to ROS 2 (See https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/rolling/doc/ur_robot_driver/ur_robot_driver/doc/usage/position_velocity_control.html#twist-controller)
- There is the "Motion Primitive Controller" that exposes motion functions from the robot controller (such as moveJ and moveL) to ROS. With this you can execute Cartesian trajectories (moveL). See https://docs.universal-robots.com/Universal_Robots_ROS_Documentation/rolling/doc/ur_robot_driver/ur_robot_driver/doc/usage/position_velocity_control.html#motion-primitives