UniversalRobots/Universal_Robots_ROS2_Driver #334: error ur_client_library

ludo2811, 2022-04-07:
I was trying to install the ur_driver on ROS2 Galactic for controlling a ur3e. 

I followed the given instructions on github:

 
ludo@ludolaptop:~$ sudo apt install python3-colcon-common-extensions python3-vcstool
[sudo] password for ludolaptop:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-colcon-common-extensions is already the newest version (0.2.1-1).
python3-vcstool is already the newest version (0.3.0-1).
The following packages were automatically installed and are no longer required:
  linux-headers-5.4.0-104 linux-headers-5.4.0-104-generic
  linux-image-5.4.0-104-generic linux-modules-5.4.0-104-generic
  linux-modules-extra-5.4.0-104-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded

ludo@ludolaptop:~$ mkdir -p ~/workspace/ros_ur_driver/src
ludo@ludolaptop:~$ cd ~/workspace/ros_ur_driver
ludo@ludolaptop:~/workspace/ros_ur_driver$ git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver
Cloning into 'src/Universal_Robots_ROS2_Driver'...
remote: Enumerating objects: 3310, done.
remote: Counting objects: 100% (1565/1565), done.
remote: Compressing objects: 100% (764/764), done.
remote: Total 3310 (delta 1110), reused 956 (delta 779), pack-reused 1745
Receiving objects: 100% (3310/3310), 71.79 MiB | 10.85 MiB/s, done.
Resolving deltas: 100% (1975/1975), done.

ludo@ludolaptop:~/workspace/ros_ur_driver$ vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.${ROS_DISTRO}.repos
..
=== src/Universal_Robots_ROS2_Description (git) ===
Cloning into '.'...
=== src/ur_msgs (git) ===
Cloning into '.'...

ludo@ludolaptop:~/workspace/ros_ur_driver$ rosdep install --ignore-src --from-paths src -y -r
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ur_bringup: Cannot locate rosdep definition for [ros2_controllers_test_nodes]
ur_calibration: Cannot locate rosdep definition for [ur_client_library]
ur_robot_driver: Cannot locate rosdep definition for [ur_client_library]
ur_controllers: Cannot locate rosdep definition for [ur_client_library]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

Everytime i try to build the dependencies the given error occurs.

In the instructions i read, the ROS2_ur_driver is build on the ur_client_library. Therefore, I thought it would be necessary to install this from github too (build in the same workspace), but nothing has changed with this error.
I know there is already a closed issue about the ur_client on github, but this actually did not help me to find the mistake in my doings.

I also tried to build ros2 foxy from source, then the error with the ur_client_library is gone, but still have the error with "ur_bringup: Cannot locate rosdep definition for [ros2_controllers_test_nodes]". Additionally i get a vcs error which says 
vcs import: error: argument --input: can't open 'src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.foxy.repos': ...

Any suggestion on solving my problem would be helpful.

fmauch, 2022-04-08:
Propably you are missing a `rosdep update` beforehand. Note, this is currently being updated in #333.

fmauch, 2022-04-12:
I am happy, this is resolved for you.

Then I will go ahead and close this.

closed 2022-04-12 by fmauch