UniversalRobots/Universal_Robots_ROS2_Driver #349: Connecting to robot during boot

urmahp, 2022-04-22:
This problem is related to the changes just merged in the client library with the bootup, see the [merged pull request](https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/98). The problem only occurs when the robot is actually booting or if you connect to the robot just after it has booted. 

The problem is that the hardware interface is waiting for the robot to come up and the controller manager is not available during this. We are still trying to spawn the controllers, but since the controller manager is unavailable this fails and essential times out (I believe) and the controllers will therefore not be spawned (see terminal output). The hardware interface comes up as expected once the robot is booted. See below terminal output. 

To reproduce just launch the ROS2 driver while the robot is booting. 

```bash
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5 robot_ip:=127.0.0.1
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-04-22-13-54-53-837437-ur-li-95qgtn2.unirobotts.local-3053
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [3059]
[INFO] [controller_stopper_node-3]: process started with pid [3061]
[INFO] [ur_ros2_control_node-1]: process started with pid [3057]
[INFO] [robot_state_publisher-4]: process started with pid [3063]
[INFO] [rviz2-5]: process started with pid [3065]
[INFO] [spawner-6]: process started with pid [3070]
[INFO] [spawner-7]: process started with pid [3072]
[INFO] [spawner-8]: process started with pid [3081]
[INFO] [spawner-9]: process started with pid [3092]
[INFO] [spawner-10]: process started with pid [3103]
[INFO] [spawner-11]: process started with pid [3115]
[ur_ros2_control_node-1] 1650635695.829920 [0] ur_ros2_co: using network interface enx482ae3ba050a (udp/10.53.253.27) selected arbitrarily from: enx482ae3ba050a, wlp2s0, docker0, br-3362adecda1e
[ur_ros2_control_node-1] [INFO] [1650635695.867884505] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1650635695.868007683] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1650635695.885376801] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[robot_state_publisher-4] 1650635695.816748 [0] robot_stat: using network interface enx482ae3ba050a (udp/10.53.253.27) selected arbitrarily from: enx482ae3ba050a, wlp2s0, docker0, br-3362adecda1e
[robot_state_publisher-4] Link base_link had 2 children
[robot_state_publisher-4] Link base had 0 children
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 2 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] Link ft_frame had 0 children
[robot_state_publisher-4] [INFO] [1650635695.844032175] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1650635695.844197970] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1650635695.845600512] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1650635695.845638551] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1650635695.845652881] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1650635695.845664592] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1650635695.845683872] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1650635695.845695094] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1650635695.845706381] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1650635695.845717236] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1650635695.845728150] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1650635695.845739622] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1650635695.845759032] [robot_state_publisher]: got segment wrist_3_link
[rviz2-5] qt.qpa.xcb: could not connect to display 
[rviz2-5] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-5] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-5] 
[rviz2-5] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-5] 
[dashboard_client-2] 1650635695.808438 [0] dashboard_: using network interface enx482ae3ba050a (udp/10.53.253.27) selected arbitrarily from: enx482ae3ba050a, wlp2s0, docker0, br-3362adecda1e
[dashboard_client-2] [ERROR] [1650635695.826066429] [UR_Client_Library]: Failed to connect to robot on IP 127.0.0.1. Please check that the robot is booted and reachable on 127.0.0.1. Retrying in 10 seconds
[controller_stopper_node-3] 1650635695.890780 [0] controller: using network interface enx482ae3ba050a (udp/10.53.253.27) selected arbitrarily from: enx482ae3ba050a, wlp2s0, docker0, br-3362adecda1e
[controller_stopper_node-3] [INFO] [1650635695.899571418] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[spawner-7] 1650635696.104941 [0]    python3: using network interface enx482ae3ba050a (udp/10.53.253.27) selected arbitrarily from: enx482ae3ba050a, wlp2s0, docker0, br-3362adecda1e
[ur_ros2_control_node-1] [WARN] [1650635696.125363099] [UR_Client_Library]: Did not receive protocol negotiation answer from robot. Message received instead: 
[ur_ros2_control_node-1] message: SafetySetup has not been confirmed yet, therefore no data will be send yet
[ur_ros2_control_node-1] source: 
[ur_ros2_control_node-1] warning level: 0. Retrying...
[ur_ros2_control_node-1] [INFO] [1650635696.125411611] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1650635696.130790204] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[spawner-6] 1650635696.184467 [0]    python3: using network interface enx482ae3ba050a (udp/10.53.253.27) selected arbitrarily from: enx482ae3ba050a, wlp2s0, docker0, br-3362adecda1e
[spawner-7] [INFO] [1650635696.310689028] [spawner_io_and_status_controller]: Waiting for '/controller_manager' services to be available
[... 42 lines cut ...]
[ERROR] [spawner-7]: process has died [pid 3072, exit code 1, cmd '/~/workspace/ros_ws_ur_driver/install/controller_manager/lib/controller_manager/spawner io_and_status_controller -c /controller_manager --ros-args'].
[ERROR] [spawner-6]: process has died [pid 3070, exit code 1, cmd '/~/workspace/ros_ws_ur_driver/install/controller_manager/lib/controller_manager/spawner joint_state_broadcaster --controller-manager /controller_manager --ros-args'].
[ERROR] [spawner-9]: process has died [pid 3092, exit code 1, cmd '/~/workspace/ros_ws_ur_driver/install/controller_manager/lib/controller_manager/spawner force_torque_sensor_broadcaster --controller-manager /controller_manager --ros-args'].
[ERROR] [spawner-10]: process has died [pid 3103, exit code 1, cmd '/~/workspace/ros_ws_ur_driver/install/controller_manager/lib/controller_manager/spawner forward_position_controller -c /controller_manager --stopped --ros-args'].
[ERROR] [spawner-8]: process has died [pid 3081, exit code 1, cmd '/~/workspace/ros_ws_ur_driver/install/controller_manager/lib/controller_manager/spawner speed_scaling_state_broadcaster --controller-manager /controller_manager --ros-args'].
[ERROR] [spawner-11]: process has died [pid 3115, exit code 1, cmd '/~/workspace/ros_ws_ur_driver/install/controller_manager/lib/controller_manager/spawner scaled_joint_trajectory_controller -c /controller_manager --ros-args'].
[ur_ros2_control_node-1] [WARN] [1650635708.167080349] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[... cut ...]
[controller_stopper_node-3] [INFO] [1650635713.141319039] [Controller stopper]: Service available
[controller_stopper_node-3] [INFO] [1650635713.141348490] [Controller stopper]: Waiting for list controllers service to come up on controller_manager/list_controllers
[controller_stopper_node-3] [INFO] [1650635713.141418319] [Controller stopper]: Service available
[dashboard_client-2] [INFO] [1650635715.828479092] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 

```

destogl, 2022-04-24:
A way to “solve” this would be to make `timeout` for spawners longer. But on the other hand, I am not sure if it is OK that controller just start some time in the future. 

 As a solution, I would rather document this behavior and add information for users to start controllers manually when hardware starts to run.

fmauch, 2022-04-27:
We could also add a small workaround node.

Imagine we don't activate controllers at all and have a separate node that is waiting for the robot to become alive which then starts the controllers. Of course, this could be controlled by yet another flag, but I would suggest to always use that scheme unless fake_hardware is used.

gavanderhoorn, 2022-04-27:
> We could also add a small workaround node.

like an inverse "controller stopper".

Making this state-based rather than time-based would be a good idea indeed.

destogl, 2022-04-27:
> We were discussing making the controller_manager a lifecycle node. In this case, spawner could simply depend on controller_manager in the launch file so it only tries to spawn controllers once there's someone to receive them

This is a possible solution, but it could fail in multi-robot systems. For example, CM is running with a hardware and UR hardware  is started later, i.e., becomes available later. So this is rather that one would need sync with the hardware lifecycle and not only CM