UniversalRobots/Universal_Robots_ROS2_Driver #350: Example Moveit-setup fails...

ludo2811, 2022-04-26:
Hey everyone,
lately i was trying to use the example moveit-setup to test the driver, but unfortunately it does not work. I followed the given instructions in "Getting started" without any problems. Then connected the robot with my laptop successfully. Then I started the driver with the following command:
$ ros2 launch ur_bringup ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.1.102 launch_rviz:=false

Then in another terminal:
fams-projekt-mosim2@FAMS-PROJEKT-MOSIM2:~/workspace_0/ros_ur_driver_0$ ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur3e launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/fams-projekt-mosim2/.ros/log/2022-04-26-10-18-43-806004-FAMS-PROJEKT-MOSIM2-11488
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=PackageNotFoundError("package 'warehouse_ros_mongo' not found, searching: ['/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_calibration', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_robot_driver', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_controllers', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_msgs', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_moveit_config', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_bringup', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_description', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_dashboard_msgs', '/opt/ros/galactic']")>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/ament_index_python/packages.py", line 50, in get_package_prefix
    content, package_prefix = get_resource('packages', package_name)
  File "/opt/ros/galactic/lib/python3.8/site-packages/ament_index_python/resources.py", line 48, in get_resource
    raise LookupError(
LookupError: Could not find the resource 'warehouse_ros_mongo' of type 'packages'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 2 more times]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/actions/node.py", line 484, in execute
    ret = super().execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 837, in execute
    self.__expand_substitutions(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 681, in __expand_substitutions
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/execute_process.py", line 681, in <listcomp>
    cmd = [perform_substitutions(context, x) for x in self.__cmd]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_context.py", line 197, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/substitutions/executable_in_package.py", line 76, in perform
    package_prefix = super().perform(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/substitutions/find_package.py", line 79, in perform
    result = self.find(package)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch_ros/substitutions/find_package.py", line 96, in find
    return get_package_prefix(package_name)
  File "/opt/ros/galactic/lib/python3.8/site-packages/ament_index_python/packages.py", line 52, in get_package_prefix
    raise PackageNotFoundError(
ament_index_python.packages.PackageNotFoundError: "package 'warehouse_ros_mongo' not found, searching: ['/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_calibration', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_robot_driver', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_controllers', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_msgs', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_moveit_config', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_bringup', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_description', '/home/fams-projekt-mosim2/workspace_0/ros_ur_driver_0/install/ur_dashboard_msgs', '/opt/ros/galactic']"

I tried apt updating and building the workspace new. When i am right, there is a problem with the warehouse_mongo package?

Any help on solving my problem is appreciated!

destogl, 2022-04-27:
have you execute `rosdep install` inside your workspace? It is very obvious that the package `warehouse_ros_mongo` is missing.

ludo2811, 2022-04-28:
Yes, I did. As stated in the getting started part. 
$rosdep install --ignore-src --from-paths src -y -r

Any suggestions what else could be the reason why the warehouse_ros_mongo package is missing?

fmauch, 2022-04-28:
A quick grep in my workspace shows that indeed `warehouse_ros_mongo` is needed, but not defined as a dependency

```
grep -r warehouse_ros_mongo
Universal_Robots_ROS2_Driver/ur_moveit_config/launch/ur_moveit.launch.py:        package="warehouse_ros_mongo",
Universal_Robots_ROS2_Driver/ur_moveit_config/launch/ur_moveit.launch.py:            {"warehouse_plugin": "warehouse_ros_mongo::MongoDatabaseConnection"},
``

PR #352 (2022-04-28): Add missing dep to warehouse_ros_mongo
This fixes #350
closed 2022-04-28 by fmauch