UniversalRobots/Universal_Robots_ROS2_Driver #361: Controller manager spawner fails to load forward_position_controller and force_torque_sensor_broadcaster

AmetistDrake, 2022-05-03:
I've been trying to use the ros2 driver with fake_hardware for ros2 galactic, but I can't get the controller manager running. I've also tried to use the real robot, followed the documentation, but same issue, process dies when starting the controller manager.

```
[INFO] [launch]: All log files can be found below /home/dani/.ros/log/2022-05-03-11-11-07-740440-dragonfly-25127
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [25131]
[INFO] [robot_state_publisher-2]: process started with pid [25133]
[INFO] [spawner-3]: process started with pid [25135]
[INFO] [spawner-4]: process started with pid [25137]
[INFO] [spawner-5]: process started with pid [25139]
[INFO] [spawner-6]: process started with pid [25141]
[INFO] [spawner-7]: process started with pid [25143]
[INFO] [spawner-8]: process started with pid [25145]
[robot_state_publisher-2] Link base_link had 2 children
[robot_state_publisher-2] Link base had 0 children
[robot_state_publisher-2] Link base_link_inertia had 1 children
[robot_state_publisher-2] Link shoulder_link had 1 children
[robot_state_publisher-2] Link upper_arm_link had 1 children
[robot_state_publisher-2] Link forearm_link had 1 children
[robot_state_publisher-2] Link wrist_1_link had 1 children
[robot_state_publisher-2] Link wrist_2_link had 1 children
[robot_state_publisher-2] Link wrist_3_link had 2 children
[robot_state_publisher-2] Link flange had 1 children
[robot_state_publisher-2] Link tool0 had 0 children
[robot_state_publisher-2] Link ft_frame had 0 children
[robot_state_publisher-2] [INFO] [1651569068.910017030] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1651569068.910176610] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1651569068.910203723] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1651569068.910220119] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1651569068.910236263] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1651569068.910252561] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-2] [INFO] [1651569068.910268382] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1651569068.910283253] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1651569068.910298504] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1651569068.910314183] [robot_state_publisher]: got segment world
[robot_state_publisher-2] [INFO] [1651569068.910329444] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1651569068.910345135] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1651569068.910360675] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1651569068.926586432] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [INFO] [1651569069.172612361] [controller_manager]: Loading controller 'io_and_status_controller'
[ros2_control_node-1] [INFO] [1651569069.191581757] [controller_manager]: Loading controller 'forward_position_controller'
[ros2_control_node-1] [ERROR] [1651569069.191641519] [controller_manager]: Loader for controller 'forward_position_controller' not found.
[ros2_control_node-1] [INFO] [1651569069.191658397] [controller_manager]: Available classes:
[ros2_control_node-1] [INFO] [1651569069.191676862] [controller_manager]:   controller_manager/test_controller
[ros2_control_node-1] [INFO] [1651569069.191691002] [controller_manager]:   controller_manager/test_controller_failed_init
[ros2_control_node-1] [INFO] [1651569069.191705620] [controller_manager]:   controller_manager/test_controller_with_interfaces
[ros2_control_node-1] [INFO] [1651569069.191720559] [controller_manager]:   diff_drive_controller/DiffDriveController
[ros2_control_node-1] [INFO] [1651569069.191740045] [controller_manager]:   effort_controllers/GripperActionController
[ros2_control_node-1] [INFO] [1651569069.191748849] [controller_manager]:   force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster
[ros2_control_node-1] [INFO] [1651569069.191756940] [controller_manager]:   joint_state_broadcaster/JointStateBroadcaster
[ros2_control_node-1] [INFO] [1651569069.191768944] [controller_manager]:   joint_trajectory_controller/JointTrajectoryController
[ros2_control_node-1] [INFO] [1651569069.191783103] [controller_manager]:   position_controllers/GripperActionController
[ros2_control_node-1] [INFO] [1651569069.191793280] [controller_manager]:   ur_controllers/GPIOController
[ros2_control_node-1] [INFO] [1651569069.191801475] [controller_manager]:   ur_controllers/ScaledJointTrajectoryController
[ros2_control_node-1] [INFO] [1651569069.191809481] [controller_manager]:   ur_controllers/SpeedScalingStateBroadcaster
[spawner-4] [INFO] [1651569069.200237780] [spawner_io_and_status_controller]: Loaded io_and_status_controller
[ros2_control_node-1] [INFO] [1651569069.202076032] [controller_manager]: Configuring controller 'io_and_status_controller'
[spawner-4] [INFO] [1651569069.219620379] [spawner_io_and_status_controller]: Configured and started io_and_status_controller
[ros2_control_node-1] [INFO] [1651569069.274293102] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[ros2_control_node-1] [INFO] [1651569069.282070664] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[ros2_control_node-1] [ERROR] [1651569069.282118593] [controller_manager]: Loader for controller 'force_torque_sensor_broadcaster' not found.
[ros2_control_node-1] [INFO] [1651569069.282136502] [controller_manager]: Available classes:
[ros2_control_node-1] [INFO] [1651569069.282153736] [controller_manager]:   controller_manager/test_controller
[ros2_control_node-1] [INFO] [1651569069.282168484] [controller_manager]:   controller_manager/test_controller_failed_init
[ros2_control_node-1] [INFO] [1651569069.282181933] [controller_manager]:   controller_manager/test_controller_with_interfaces
[ros2_control_node-1] [INFO] [1651569069.282191886] [controller_manager]:   diff_drive_controller/DiffDriveController
[ros2_control_node-1] [INFO] [1651569069.282199243] [controller_manager]:   effort_controllers/GripperActionController
[ros2_control_node-1] [INFO] [1651569069.282205783] [controller_manager]:   force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster
[ros2_control_node-1] [INFO] [1651569069.282212378] [controller_manager]:   joint_state_broadcaster/JointStateBroadcaster
[ros2_control_node-1] [INFO] [1651569069.282220904] [controller_manager]:   joint_trajectory_controller/JointTrajectoryController
[... 2 lines cut ...]
[ros2_control_node-1] [INFO] [1651569069.282242058] [controller_manager]:   ur_controllers/ScaledJointTrajectoryController
[ros2_control_node-1] [INFO] [1651569069.282249823] [controller_manager]:   ur_controllers/SpeedScalingStateBroadcaster
[ros2_control_node-1] [INFO] [1651569069.283829299] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-5] [INFO] [1651569069.300269049] [spawner_speed_scaling_state_broadcaster]: Loaded speed_scaling_state_broadcaster
[ros2_control_node-1] [INFO] [1651569069.301828597] [controller_manager]: Loading controller 'joint_trajectory_controller'
[ros2_control_node-1] [INFO] [1651569069.307876577] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[ros2_control_node-1] [INFO] [1651569069.307944628] [speed_scaling_state_broadcaster]: Publisher rate set to : 100.0 Hz
[spawner-3] [INFO] [1651569069.308734252] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-1] [INFO] [1651569069.310037867] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ros2_control_node-1] [INFO] [1651569069.310110075] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-5] [INFO] [1651569069.316289294] [spawner_speed_scaling_state_broadcaster]: Configured and started speed_scaling_state_broadcaster
[spawner-3] [INFO] [1651569069.320252423] [spawner_joint_state_broadcaster]: Configured and started joint_state_broadcaster
[spawner-8] [INFO] [1651569069.320408770] [spawner_joint_trajectory_controller]: Loaded joint_trajectory_controller
[... cut ...]
[ERROR] [spawner-7]: process has died [pid 25143, exit code 1, cmd '/opt/ros/galactic/lib/controller_manager/spawner forward_position_controller -c /controller_manager --stopped --ros-args'].
[ERROR] [spawner-6]: process has died [pid 25141, exit code 1, cmd '/opt/ros/galactic/lib/controller_manager/spawner force_torque_sensor_broadcaster --controller-manager /controller_manager --ros-args'].
[INFO] [spawner-4]: process has finished cleanly [pid 25137]
[INFO] [spawner-5]: process has finished cleanly [pid 25139]
[INFO] [spawner-3]: process has finished cleanly [pid 25135]
[INFO] [spawner-8]: process has finished cleanly [pid 25145]
```

destogl, 2022-05-04:
Are you sure you have all dependencies installed? Have you run `rosdep`? Or do you have source installation?

AmetistDrake, 2022-05-24:
I have source installation, and run rosdep. All dependencies have been successfully installed.

fmauch, 2022-06-13:
@AmetistDrake 

That issue pointed out a missing dependency on position_controllers. That should be "fixable" for you by running `sudo apt install ros-galactic-position-controllers`, but that should definitively be fixed. Will prepare a PR for that later today.

danzimmerman, 2022-06-13:
I pulled in #392 (and used `vcs pull` to update any source dependencies originally installed in the workspace `src` from the `.repos` file), updated via `rosdep`, cleaned my workspace, and rebuilt, and still getting a spawn error on `force_torque_sensor_broadcaster` when trying to use this with URSim.

<details>

```
[ur_ros2_control_node-1] [INFO] [1655158418.350478783] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[ur_ros2_control_node-1] [ERROR] [1655158418.350541373] [controller_manager]: Loader for controller 'force_torque_sensor_broadcaster' not found.
[ur_ros2_control_node-1] [INFO] [1655158418.350568735] [controller_manager]: Available classes:
[ur_ros2_control_node-1] [INFO] [1655158418.350589072] [controller_manager]:   controller_manager/test_controller
[ur_ros2_control_node-1] [INFO] [1655158418.350602217] [controller_manager]:   controller_manager/test_controller_failed_init
[ur_ros2_control_node-1] [INFO] [1655158418.350614127] [controller_manager]:   controller_manager/test_controller_with_interfaces
[ur_ros2_control_node-1] [INFO] [1655158418.350626261] [controller_manager]:   diff_drive_controller/DiffDriveController[ur_ros2_control_node-1] [INFO] [1655158418.350637913] [controller_manager]:   effort_controllers/GripperActionController
[ur_ros2_control_node-1] [INFO] [1655158418.350649474] [controller_manager]:   effort_controllers/JointGroupEffortController
[ur_ros2_control_node-1] [INFO] [1655158418.350666915] [controller_manager]:   force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster
[ur_ros2_control_node-1] [INFO] [1655158418.350679310] [controller_manager]:   forward_command_controller/ForwardCommandController
[ur_ros2_control_node-1] [INFO] [1655158418.350690519] [controller_manager]:   imu_sensor_broadcaster/IMUSensorBroadcaster
[ur_ros2_control_node-1] [INFO] [1655158418.350701613] [controller_manager]:   joint_state_broadcaster/JointStateBroadcaster
[ur_ros2_control_node-1] [INFO] [1655158418.350712796] [controller_manager]:   joint_trajectory_controller/JointTrajectoryController
[ur_ros2_control_node-1] [INFO] [1655158418.350724238] [controller_manager]:   position_controllers/GripperActionController
[ur_ros2_control_node-1] [INFO] [1655158418.350735298] [controller_manager]:   position_controllers/JointGroupPositionController
[ur_ros2_control_node-1] [INFO] [1655158418.350746875] [controller_manager]:   ur_controllers/GPIOController
[ur_ros2_control_node-1] [INFO] [1655158418.350758287] [controller_manager]:   ur_controllers/ScaledJointTrajectoryController
[ur_ros2_control_node-1] [INFO] [1655158418.350778078] [controller_manager]:   ur_controllers/SpeedScalingStateBroadcaster
[ur_ros2_control_node-1] [INFO] [1655158418.350794026] [controller_manager]:   velocity_controllers/JointGroupVelocityController
```
</details>

Should #392 solve the `force_torque_sensor_broadcaster` issue? 

Is this possibly an issue with simulated hardware not actually having F/T sensor functionality? 

I'm not using the Docker install, I'm running URSim in a VM on windows. I can connect URSim to the driver:

```
[ur_ros2_control_node-1] [INFO] [1655158899.049087778] [UR_Client_Library]: Robot requested program
[ur_ros2_control_node-1] [INFO] [1655158899.049245551] [UR_Client_Library]: Sent program to robot
[ur_ros2_control_node-1] [INFO] [1655158899.111604531] [UR_Client_Library]: Robot connected to reverse interface. Ready to receive control commands.`
```

The other controllers are present and active as necessary, `/joint_states` and `/speed_scaling_state_broadcaster/speed_scaling` topics are publishing the robot joint state and speed slider scaling as I change those in Polyscope, but there's no `force_torque_sensor_broadcaster`:

```
dan@<computername>:~$ ros2 control list_controllers
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
io_and_status_controller[ur_controllers/GPIOController] active
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] active
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
forward_position_controller[position_controllers/JointGroupPositionController] inactive
```

or relevant topic

```
dan@<computername>:~$ ros2 topic list
/dynamic_joint_states
/forward_position_controller/commands
/io_and_status_controller/io_states
/io_and_status_controller/robot_mode
/io_and_status_controller/robot_program_running
/io_and_status_controller/safety_mode
/io_and_status_controller/tool_data
/joint_states
/parameter_events
/robot_description
/rosout
/scaled_joint_trajectory_controller/joint_trajectory
/scaled_joint_trajectory_controller/state
/speed_scaling_state_broadcaster/speed_scaling
/tf
/tf_static
```

danzimmerman, 2022-06-15:
Looks from local tests like #405 is going to fix the `force_torque_sensor_broadcaster` issue... works with fake hardware at least. Having trouble with URSim but I think that's a different discussion.