UniversalRobots/Universal_Robots_ROS2_Driver #365: Installation fails due to ros-foxy-ur-msgs

robertokcanale, 2022-05-04:
I am trying to install the repo, following the turtorial steps.
But when I run:

_rosdep install --ignore-src --from-paths src -y -r_

I get the following error

executing command [sudo -H apt-get install -y ros-foxy-ur-msgs]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-foxy-ur-msgs
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y ros-foxy-ur-msgs] failed
  apt: Failed to detect successful installation of [ros-foxy-ur-msgs]

How can I go forward?   I tired to go on with ros-foxy-ur-msgs but unfortunately there is no guide on how to install them as well

destogl, 2022-05-04:
I think that `ur_msgs` package is released recently to foxy, so you need to call `rosdep update` to get the newest definitions. Or maybe the sync is still not out.

fmauch, 2022-05-05:
Indeed there has not been a sync since we released the message package.

That's why the repo is still in the [Universal_Robots_ROS2_Driver-not-released.foxy.repos](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/foxy/Universal_Robots_ROS2_Driver-not-released.foxy.repos) file.

When building the driver as [explained in the README](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#getting-started) things should work as expected.

@robertokcanale Did you follow any other instruction that has been missing the step of importingnon-released upstream repositories? If yes, which one? If not, please try following the instructions ;-)

fmauch, 2022-07-11:
Since the driver ist released for Foxy by now I think this can be closed.

closed 2022-07-11 by fmauch