UniversalRobots/Universal_Robots_ROS2_Driver #374: [Foxy] Cannot Read Robot State Information in RViz

TheFalcoGuy, 2022-05-11:
I am currently running a UR5e and am so very close to getting ROS2 Foxy up and running on the system.

However, I am receiving a string of warnings about the "***_link" passed to canTransform argument source_frame - frame does not exist when I run the ur_bringup command as below...

d-lab@dlab-Alienware-m15-R6:~/workspace/ros_ur_driver$`` ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.0.99 use_fake_hardware:=false launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/d-lab/.ros/log/2022-05-10-21-53-20-404791-dlab-Alienware-m15-R6-56819
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [56825]
[INFO] [controller_stopper_node-3]: process started with pid [56827]
[INFO] [ros2_control_node-1]: process started with pid [56823]
[INFO] [robot_state_publisher-4]: process started with pid [56829]
[INFO] [rviz2-5]: process started with pid [56831]
[INFO] [spawner.py-6]: process started with pid [56833]
[INFO] [spawner.py-7]: process started with pid [56835]
[INFO] [spawner.py-8]: process started with pid [56837]
[INFO] [spawner.py-9]: process started with pid [56839]
[INFO] [spawner.py-10]: process started with pid [56841]
[controller_stopper_node-3] [INFO] [1652234000.697704638] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[dashboard_client-2] [INFO] [1652234000.698640861] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link base had 0 children
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 1 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] [INFO] [1652234000.706016214] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1652234000.706075563] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1652234000.706080188] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1652234000.706083373] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1652234000.706086401] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1652234000.706089118] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1652234000.706091947] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1652234000.706094681] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1652234000.706097542] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1652234000.706100289] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1652234000.706102932] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1652234000.706266756] [URPositionHardwareInterface]: Starting ...please wait...
[ros2_control_node-1] [INFO] [1652234000.706300971] [URPositionHardwareInterface]: Initializing driver...
[ros2_control_node-1] [WARN] [1652234000.707368571] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[spawner.py-7] [INFO] [1652234000.818231585] [spawner_io_and_status_controller]: Waiting for /controller_manager services
[rviz2-5] [INFO] [1652234000.904283485] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1652234000.904400513] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-5] [INFO] [1652234000.920151968] [rviz2]: Stereo is NOT SUPPORTED
[ros2_control_node-1] [INFO] [1652234000.937735470] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ros2_control_node-1] [INFO] [1652234000.938230036] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[rviz2-5] Parsing robot urdf xml string.
[rviz2-5] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[... 27 lines cut ...]
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist

This results in the robot not properly publishing the state as seen in Rviz.

When I try to run the joint_test_trajectory_controller.launch.py script, I yield the following...

d-lab@dlab-Alienware-m15-R6:~/workspace/ros_ur_driver$ ros2 launch ur_bringup test_joint_trajectory_controller.launch.py
[INFO] [launch]: All log files can be found below /home/d-lab/.ros/log/2022-05-10-21-43-26-852869-dlab-Alienware-m15-R6-55450
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [publisher_joint_trajectory_controller-1]: process started with pid [55452]
[publisher_joint_trajectory_controller-1] [INFO] [1652233407.180296846] [publisher_joint_trajectory_controller]: Publishing 4 goals on topic "/joint_trajectory_controller/joint_trajectory" every 6 s
[publisher_joint_trajectory_controller-1] [WARN] [1652233413.182542816] [publisher_joint_trajectory_controller]: Start configuration could not be checked! Check "joint_state" topic!

So I went to check on the controllers, I tried manually activating the controllers, which yields...
d-lab@dlab-Alienware-m15-R6:~/workspace/ros_ur_driver$ ros2 control load_controller --set-state start joint_trajectory_controller
Could not contact service /controller_manager/load_controller

So my suspicion is that the controller_manager is not properly initiating, I tried searching for possible fixes, but ended up coming empty on this. I was able to validate that the robot state is properly publishing when I utilize fake hardware, only when I run to the actual hardware does it fail.

I am not sure how to properly remedy this particular error

jackdario96, 2022-05-11:
Hi, I'm having the same issue but with `use_fake_hardware:=true` on Galactic, if this can be helpful for debugging. I'm investigating and hope to come back with more clear ideas

fmauch, 2022-05-11:
The errors come from RViz and are "normal" when starting things up (also see #259). To debug this further, best use the argument `launch_rviz:=false`. The output you posted seems not to contain anything going wrong (as your shell gets flooded from RViz...)

Please use [code blocks](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks) when posting shell output, this helps reading it a lot.

jackdario96, 2022-05-11:
I don't know if the original issue is the same so I'm posting here but let me know if I need to open a new issue. After following @fmauch suggestion to disable RViz, this is the output I get, than it loops the same exceptions again and again:

```
root@7fab5ba7b37f:/# ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e description_package:=ur_description robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false

[INFO] [launch]: All log files can be found below /root/.ros/log/2022-05-11-16-22-39-887615-7fab5ba7b37f-30964
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [31000]
[INFO] [robot_state_publisher-2]: process started with pid [31002]
[INFO] [spawner-3]: process started with pid [31004]
[INFO] [spawner-4]: process started with pid [31006]
[INFO] [spawner-5]: process started with pid [31008]
[INFO] [spawner-6]: process started with pid [31010]
[INFO] [spawner-7]: process started with pid [31012]
[INFO] [spawner-8]: process started with pid [31014]
[robot_state_publisher-2] Link base_link had 2 children
[robot_state_publisher-2] Link base had 0 children
[robot_state_publisher-2] Link base_link_inertia had 1 children
[robot_state_publisher-2] Link shoulder_link had 1 children
[robot_state_publisher-2] Link upper_arm_link had 1 children
[robot_state_publisher-2] Link forearm_link had 1 children
[robot_state_publisher-2] Link wrist_1_link had 1 children
[robot_state_publisher-2] Link wrist_2_link had 1 children
[robot_state_publisher-2] Link wrist_3_link had 2 children
[robot_state_publisher-2] Link flange had 1 children
[robot_state_publisher-2] Link tool0 had 0 children
[robot_state_publisher-2] Link ft_frame had 0 children
[robot_state_publisher-2] [INFO] [1652286161.407461200] [robot_state_publisher]: got segment base
[robot_state_publisher-2] [INFO] [1652286161.407609900] [robot_state_publisher]: got segment base_link
[robot_state_publisher-2] [INFO] [1652286161.407636800] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-2] [INFO] [1652286161.407651700] [robot_state_publisher]: got segment flange
[robot_state_publisher-2] [INFO] [1652286161.407664700] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-2] [INFO] [1652286161.408592800] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-2] [INFO] [1652286161.408626100] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-2] [INFO] [1652286161.408640400] [robot_state_publisher]: got segment tool0
[robot_state_publisher-2] [INFO] [1652286161.408653000] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-2] [INFO] [1652286161.408665900] [robot_state_publisher]: got segment world
[robot_state_publisher-2] [INFO] [1652286161.408682300] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-2] [INFO] [1652286161.408704200] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-2] [INFO] [1652286161.408718200] [robot_state_publisher]: got segment wrist_3_link
[ros2_control_node-1] [INFO] [1652286161.414764500] [controller_manager]: update rate is 500 Hz
[spawner-4] Traceback (most recent call last):
[spawner-4]   File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_generator_py/import_type_support_impl.py", line 46, in import_type_support
[spawner-4]     return importlib.import_module(module_name, package=pkg_name)
[spawner-4]   File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
[spawner-4]     return _bootstrap._gcd_import(name[level:], package, level)
[spawner-4]   File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
[spawner-4]   File "<frozen importlib._bootstrap>", line 991, in _find_and_load
[spawner-4]   File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
[spawner-4]   File "<frozen importlib._bootstrap>", line 657, in _load_unlocked
[spawner-4]   File "<frozen importlib._bootstrap>", line 556, in module_from_spec
[spawner-4]   File "<frozen importlib._bootstrap_external>", line 1166, in create_module
[spawner-4]   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
[spawner-4] ImportError: /opt/ros/galactic/lib/libcontroller_manager_msgs__rosidl_generator_c.so: undefined symbol: builtin_interfaces__msg__Duration__are_equal
[spawner-4] 
[spawner-4] During handling of the above exception, another exception occurred:
[spawner-4] 
[spawner-4] Traceback (most recent call last):
[spawner-4]   File "/opt/ros/galactic/lib/controller_manager/spawner", line 20, in <module>
[spawner-4]     ret = main()
[spawner-4]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 167, in main
[spawner-4]     if is_controller_loaded(node, controller_manager_name, controller_name):
[spawner-4]   File "/opt/ros/galactic/lib/controller_manager/spawner.py", line 110, in is_controller_loaded
[spawner-4]     controllers = list_controllers(node, controller_manager).controller
[spawner-4]   File "/opt/ros/galactic/lib/python3.8/site-packages/controller_manager/controller_manager_services.py", line 49, in list_controllers
[spawner-4]     return service_caller(node, f'{controller_manager_name}/list_controllers',
[spawner-4]   File "/opt/ros/galactic/lib/python3.8/site-packages/controller_manager/controller_manager_services.py", line 23, in service_caller
[spawner-4]     cli = node.create_client(service_type, service_name)
[... cut ...] galactic branch of the driver there is no robot description, I cloned the robot description from the [Universal_Rbots_ROS2_drescription](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description) and the launch file to simply visualize the robot in RViz works perfectly. So the RViz problem is caused by some other nodes that are broken when launched, which probably breaks the robot_state_publisher.
Probably it is something dummy, so any clue giving nothing as granted is highly welcome!

fmauch, 2022-05-17:
@jackdario96 That output seems as if there is a type mismatch regarding controller_manager_msgs. Did you build anything from source or did you use the binary installation? I'd recommend using the latter.

TheFalcoGuy, 2022-06-13:
So I built from source, and it turns out it had to do specifically with how I was sourcing the UR drivers. Basically had to do both the ROS2 Foxy AND UR driver sources for each terminal (source /workspace/ros_ur_driver/install/setup.bash) and that fixed the error.

closed 2022-06-13 by TheFalcoGuy