UniversalRobots/Universal_Robots_ROS2_Driver #378: Installing driver using `sudo apt-get install ros-galactic-ur-robot-driver` does not install ur_moveit_config

haavarpb, 2022-05-16:
`apt show ros-galactic-ur-robot-driver`
shows:
`Package: ros-galactic-ur-robot-driver
Version: 2.1.0-1focal.20220503.203410
Priority: optional
Section: misc
Maintainer: Denis Stogl <denis@stoglrobotics.de>
Installed-Size: 3 166 kB
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.0), libstdc++6 (>= 9), ros-galactic-console-bridge-vendor, ros-galactic-controller-manager, ros-galactic-controller-manager-msgs, ros-galactic-geometry-msgs, ros-galactic-hardware-interface, ros-galactic-launch-testing-ament-cmake, ros-galactic-pluginlib, ros-galactic-rclcpp, ros-galactic-rclpy, ros-galactic-std-msgs, ros-galactic-std-srvs, ros-galactic-tf2-geometry-msgs, ros-galactic-ur-bringup, ros-galactic-ur-client-library, ros-galactic-ur-controllers, ros-galactic-ur-dashboard-msgs, ros-galactic-ur-description, ros-galactic-ur-msgs, ros-galactic-ros-workspace
Download-Size: 1 561 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros2/ubuntu focal/main amd64 Packages
Description: The new driver for Universal Robots UR3, UR5 and UR10 robots with CB3 controllers and the e-series.`

Have I done something wrong, or is this intended?

fmauch, 2022-06-02:
The `ur_moveit_config` package was not built initially and we didn't manage to get in the fix before the sync freeze. The issue is addressed and fixed in fd66700db47aefe7fa8f64e4068d9d5213d367d5 and will be included into the next galactic sync. The sync should be happen [any day now](https://discourse.ros.org/t/preparing-for-galactic-sync-2022-06-01/25824).

Sorry for the inconvenience.

haavarpb, 2022-06-02:
I see, I moved on by installing from source anyways. Thanks!

fmauch, 2022-06-20:
closing as the sync has happened.

closed 2022-06-20 by fmauch