UniversalRobots/Universal_Robots_ROS2_Driver #380: Occasional crash with SJTC

Schnilz, 2022-05-18:
Hi, 
first of all thanks for writing this, it is awesome :)
The driver works realy well and setting it up was pretty straight forward because of the nice documentation, but i think i encountered a bug. Most of the time this issue doesn't occur but sometimes the SJTC crashes with following log messages:
```
[ur_ros2_control_node-1] [INFO] [1652881749.579905572] [scaled_joint_trajectory_controller]: Received new action goal
[ur_ros2_control_node-1] [INFO] [1652881749.580010468] [scaled_joint_trajectory_controller]: Accepted new action goal
[ur_ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ur_ros2_control_node-1]   what():  Executing action server but nothing is ready
[ERROR] [ur_ros2_control_node-1]: process has died [pid 57695, exit code -6, cmd ...
```
the error comes from rclcpp_action/src/server.cpp line 306 but im not realy sure what it actualy means :/
Like i said, reproducing it is not so easy as it works most of the time. I have a feeling it happens more often when the trajectory is short, but this could be wrong....

fmauch, 2022-06-02:
Thanks for posting this. This is definitively something to look into. Could you tell us a little about your setup for us to reproduce this?

- ROS version
- This repo version (commit hash ideally)

fmauch, 2022-06-13:
This issue is not forgotten, but will need some more time to reproduce.

RoyCLR, 2022-08-23:
I seems to hit the same issue, but with `joint_trajectory_controller`:

```
[ros2_control_node-1] [1661228175.046757625INFO] [joint_trajectory_controller::goal_callback]: Received new action goal
[ros2_control_node-1] [1661228175.046794670INFO] [joint_trajectory_controller::goal_callback]: Accepted new action goal
[ros2_control_node-1] terminate called after throwing an instance of 'std::runtime_error'
[ros2_control_node-1]   what():  Executing action server but nothing is ready
[ERROR] [ros2_control_node-1]: process has died [pid 1708770, exit code -6, cmd
```

Setup:
- Using ROS2 Foxy
- Universal_Robots_ROS2_Driver at 69c2030fe88861039826f838aeca0ffc03b5a303
- With close to, but non-real time kernel "Liquorix" https://liquorix.net/

fmauch, 2023-11-07:
Coming back to this: https://github.com/ros2/rclcpp/pull/2250

tommy-erko, 2023-11-07:
Detailed explanation of the issue is here: https://github.com/ros2/rclcpp/issues/2242

firesurfer, 2024-04-25:
@fmauch Could you figure out if that fix is also backported to iron or humble ? Apparently it is finally merged into rolling but I couldn't figure out if there are plans to backport it

fmauch, 2024-04-25:
https://github.com/ros2/rclcpp/issues/2242#issuecomment-2072845701 not necessarily :-/

urfeex, 2025-01-15:
Seems like this got merged to [humble](https://github.com/ros2/rclcpp/pull/2635) by now.