UniversalRobots/Universal_Robots_ROS2_Driver #383: Galactic Repository Missing Items

FieldOfSnow, 2022-05-26:
Hi all,

I'm new to Github and ROS, but trying to get everything installed and it looks like there are some things missing from the Galactic install. First, on the main page it says Galactic Binary is passing, but when you go to the tree it says it's failing.(https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/tree/galactic)

Then when I clone the files, I don't have a galactic version of the "-not-released" repos.

Then when I run the install line (colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release) I get these errors:
1) --- stderr: ur_controllers  
2) [file] error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
   65 |   controller_interface::CallbackReturn on_configure(const rclcpp_lifecycle::State& previous_state) override;
[other file] error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
   67 |   controller_interface::CallbackReturn on_activate(const rclcpp_lifecycle::State& previous_state) override;
69,71,61,86,107,113,44,154,238,283,57,55
3) Ending with this: make[2]: *** [CMakeFiles/ur_controllers.dir/build.make:76: CMakeFiles/ur_controllers.dir/src/speed_scaling_state_broadcaster.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/ur_controllers.dir/build.make:63: CMakeFiles/ur_controllers.dir/src/scaled_joint_trajectory_controller.cpp.o] Error 1
make[2]: *** [CMakeFiles/ur_controllers.dir/build.make:89: CMakeFiles/ur_controllers.dir/src/gpio_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/ur_controllers.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< ur_controllers [8.26s, exited with code 2]

Summary: 27 packages finished [14.9s]
  1 package failed: ur_controllers
  1 package had stderr output: ur_controllers
  2 packages not processed

I'm going to try and get the Foxy version running, but if someone could help me out that would be greatly appreciated!

T2FF, 2022-05-30:
Hi I faced some similar problems at the beginning. Maybe these steps will help you:

1. make sure ROS2 is correctly installed and sourced in your terminal (for me the [debian package installation](https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html) worked best) 
2. try to specify the branch:
`git clone -b galactic https://github.com/ros-controls/ros2_control_demos src/ros2_control_demos`
and install repositories with the ony .repos file there is in the galactic branch:
`vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.galactic.repos
`

fmauch, 2022-06-13:
Note: For galactic it is not necessary to build any upstream packages from source, therefore, there is no not-released repos file. This could and should be updated in the README for galactic, hence I'll tag this issue as documentation problem.

fmauch, 2022-06-28:
Closed via #423

PR #423 (2022-06-28): Remove vcstool instruction for galactic
Closes #383
closed 2022-06-28 by fmauch