UniversalRobots/Universal_Robots_ROS2_Driver #390: galactic: scaled JTC does not work on fake_hw

fmauch, 2022-06-09:
When trying to use the scaled JTC on Fake HW I get 

```
[ros2_control_node-1] terminate called after throwing an instance of 'std::underflow_error'
[ros2_control_node-1]   what():  addition leads to int64_t underflow
```

Obviously this needs to be further debugged, but I wanted to post it here as a first step.

fmauch, 2022-06-09:
Update: It seems that all floating point io states generate a `nan` value when using fake_hw. @destogl do you know where this could come from? For example, the output of the `/io_and_status_controller/tool_data` is

```
...
tool_output_voltage: 0
tool_current: .nan
tool_temperature: .nan
tool_mode: 0
```

ferrolho, 2022-08-15:
I think we can remove the "galactic:" prefix from the title ─ I just tested this on ROS 2 Humble, and I am getting the same error message:
```
[ros2_control_node-1] terminate called after throwing an instance of 'std::underflow_error'
[ros2_control_node-1]   what():  addition leads to int64_t underflow
```

fmauch, 2022-08-15:
@ferrolho thanks for adding this information

haider8645, 2023-05-02:
This seems to be still open. I had to switch the SJTC to JTC on humble.

fmauch, 2024-05-22:
the scaled jtc should work on mock hardware (fake hardware on humble) by now.

closed 2024-05-22 by fmauch

gautz, 2024-06-07:
@fmauch 
Trying to [use the mock hardware on Rolling](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#start-hardware-simulator-or-mockup)
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_mock_hardware:=true launch_rviz:=true`
I'm still getting a similar error:
```
[INFO] [launch]: All log files can be found below /home/etudiant/.ros/log/2024-06-07-11-51-40-145410-ubuntu22ros2-3870
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [ros2_control_node-1]: process started with pid [3876]
[INFO] [robot_state_publisher-2]: process started with pid [3878]
[INFO] [rviz2-3]: process started with pid [3880]
[INFO] [spawner-4]: process started with pid [3882]
[INFO] [spawner-5]: process started with pid [3884]
[INFO] [spawner-6]: process started with pid [3886]
[INFO] [spawner-7]: process started with pid [3888]
[INFO] [spawner-8]: process started with pid [3890]
[INFO] [spawner-9]: process started with pid [3892]
[ros2_control_node-1] [WARN] [1717753902.029562273] [controller_manager]: [Deprecated] Passing the robot description parameter directly to the control_manager node is deprecated. Use '~/robot_description' topic from 'robot_state_publisher' instead.
[robot_state_publisher-2] [INFO] [1717753902.029905659] [robot_state_publisher]: Robot initialized
[ros2_control_node-1] [INFO] [1717753902.032391742] [resource_manager]: Loading hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1717753902.041191299] [resource_manager]: Initialize hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1717753902.041700137] [resource_manager]: Successful initialization of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1717753902.042041105] [resource_manager]: 'configure' hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1717753902.042058241] [resource_manager]: Successful 'configure' of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1717753902.042098039] [resource_manager]: 'activate' hardware 'ur5e' 
[ros2_control_node-1] [INFO] [1717753902.042107553] [resource_manager]: Successful 'activate' of hardware 'ur5e'
[ros2_control_node-1] [INFO] [1717753902.305384643] [controller_manager]: update rate is 500 Hz
[ros2_control_node-1] [WARN] [1717753902.305775689] [controller_manager]: Could not enable FIFO RT scheduling policy. Consider setting up your user to do FIFO RT scheduling. See [https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html] for details.
[ros2_control_node-1] [INFO] [1717753904.053198962] [controller_manager]: Loading controller 'scaled_joint_trajectory_controller'
[ros2_control_node-1] [INFO] [1717753904.541281115] [controller_manager]: Loading controller 'forward_position_controller'
[spawner-4] [INFO] [1717753904.551334558] [spawner_scaled_joint_trajectory_controller]: Loaded scaled_joint_trajectory_controller
[ros2_control_node-1] [INFO] [1717753904.672102543] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[spawner-9] [INFO] [1717753904.715156818] [spawner_forward_position_controller]: Loaded forward_position_controller
[ros2_control_node-1] [INFO] [1717753904.922560977] [controller_manager]: Loading controller 'io_and_status_controller'
[spawner-8] [INFO] [1717753904.970341360] [spawner_force_torque_sensor_broadcaster]: Loaded force_torque_sensor_broadcaster
[ros2_control_node-1] [INFO] [1717753905.065300266] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[spawner-6] [INFO] [1717753905.136619431] [spawner_io_and_status_controller]: Loaded io_and_status_controller
[ros2_control_node-1] [INFO] [1717753905.211807422] [speed_scaling_state_broadcaster]: Loading UR SpeedScalingStateBroadcaster with tf_prefix: 
[ros2_control_node-1] [INFO] [1717753905.214072880] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-7] [INFO] [1717753905.240668864] [spawner_speed_scaling_state_broadcaster]: Loaded speed_scaling_state_broadcaster
[ros2_control_node-1] [INFO] [1717753905.443239332] [controller_manager]: Configuring controller 'scaled_joint_trajectory_controller'
[ros2_control_node-1] [INFO] [1717753905.443528215] [scaled_joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ros2_control_node-1] [INFO] [1717753905.443648548] [scaled_joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ros2_control_node-1] [INFO] [1717753905.443672279] [scaled_joint_trajectory_controller]: Using 'splines' interpolation method.
[spawner-5] [INFO] [1717753905.468039187] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ros2_control_node-1] [INFO] [1717753905.488513355] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2_control_node-1] [INFO] [1717753905.562940376] [controller_manager]: Configuring controller 'forward_position_controller'
[ros2_control_node-1] [INFO] [1717753905.587465647] [forward_position_controller]: configure successful
[ros2_control_node-1] [INFO] [1717753905.590070898] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'
[ros2_control_node-1] [INFO] [1717753905.608281348] [controller_manager]: Configuring controller 'io_and_status_controller'
[ros2_control_node-1] [INFO] [1717753905.666355321] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[... cut ...]
[rviz2-3] [INFO] [1717753906.757209800] [rviz2]: OpenGl version: 3.3 (GLSL 3.3)
[rviz2-3] [INFO] [1717753906.874182894] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [ros2_control_node-1]: process has died [pid 3876, exit code -6, cmd '/opt/ros/rolling/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_aty_4uee --params-file /opt/ros/rolling/share/ur_robot_driver/config/ur5e_update_rate.yaml --params-file /tmp/launch_params_sl0eu_nn']
```

Is the fix already released or should I build UR packages ?