UniversalRobots/Universal_Robots_ROS2_Driver #417: BUG with planning

yoavRobee, 2022-06-22:
i mannged to get it to plan in rviz with moveit
but when i try to execute the plan i get the following error :

[moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: scaled_joint_trajectory_controller/follow_joint_trajectory

fmauch, 2022-06-22:
A bit more context how you started everything might be helpful. Real robot? URSim? Fake hardware?

Please provide the exact startup commands that you executed and the complete output of those.

closed 2022-06-22 by yoavRobee

fmauch, 2022-06-23:
I'm glad to hear this has been resolved.

Could you please post your solution so others with a similar problem can get a hint how to solve this?