UniversalRobots/Universal_Robots_ROS2_Driver #428: Add setPayload service

fmauch, 2022-06-28:
In the ROS1 driver we could set the payload through ROS. This is still missing.

sillkjc, 2023-04-19:
This could be fixed easily by providing a mechanism to send ur_scripts just like the ROS1 driver.

fmauch, 2023-06-20:
Note: There is a set_payload service available in the gpio controller that should be working. Also, with #721 a direct URScript interface will be available soon.

closed 2023-06-20 by fmauch

wafts95, 2023-11-29:
Hi @fmauch, we are trying to use the setPayload service on a UR10e and latest binary for ROS2 Humble.
What are the units for the Mass and Center of Gravity (x, y, z) in the ROS API?

RobertWilbrandt, 2023-11-30:
They follow the general ROS unit standards ([REP 103](https://www.ros.org/reps/rep-0103.html)): Meters for the CoG displacement from the tool mount and kilogramms for the mass.