UniversalRobots/Universal_Robots_ROS2_Driver #433: get_prefix_name() c++ error, can't build main branch

ahumphries2, 2022-07-12:
I'm having a really hard time just getting the main branch installed, for either ROS2 Humble or Rolling. I keep getting the same error when running the colcon build:

    error: ‘class hardware_interface::LoanedCommandInterface’ has no member named ‘get_prefix_name’

Please assist. Below is the output from terminal:

```
cobot@cobot-Precision-7560:~/workspace/ros_ur_driver$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
[0.323s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'gripper_controllers' is in: /home/cobot/workspace/ros_ur_driver/install/gripper_controllers, /opt/ros/humble
	'joint_state_broadcaster' is in: /home/cobot/workspace/ros_ur_driver/install/joint_state_broadcaster, /opt/ros/humble
	'position_controllers' is in: /home/cobot/workspace/ros_ur_driver/install/position_controllers, /opt/ros/humble
	'joint_trajectory_controller' is in: /home/cobot/workspace/ros_ur_driver/install/joint_trajectory_controller, /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding gripper_controllers joint_state_broadcaster joint_trajectory_controller position_controllers

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> forward_command_controller
Starting >>> joint_trajectory_controller
Starting >>> force_torque_sensor_broadcaster
Starting >>> joint_state_broadcaster
Starting >>> ur_dashboard_msgs
Starting >>> ur_description
Starting >>> ur_msgs
Starting >>> ros2_controllers_test_nodes
Starting >>> ur_client_library
Starting >>> diff_drive_controller
Starting >>> imu_sensor_broadcaster
Starting >>> gripper_controllers
--- stderr: ros2_controllers_test_nodes                                                                                                                                    
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< ros2_controllers_test_nodes [2.60s]
Finished <<< ur_description [3.79s]                                                                                                                                        
Starting >>> ur_moveit_config
Finished <<< ur_moveit_config [2.51s]                                                                                                                                                  
Finished <<< ur_dashboard_msgs [12.2s]                                                                                                                                                     
Finished <<< ur_client_library [13.6s]                                                                                                                                                    
--- stderr: gripper_controllers                                                                                                                                                           
In file included from /home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller.hpp:176,
                 from /home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/src/gripper_action_controller.cpp:18:
/home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller_impl.hpp: In instantiation of ‘controller_interface::CallbackReturn gripper_action_controller::GripperActionController<HardwareInterface>::on_activate(const rclcpp_lifecycle::State&) [with const char* HardwareInterface = (& hardware_interface::HW_IF_POSITION); controller_interface::CallbackReturn = rclcpp_lifecycle::node_interfaces::LifecycleNodeInterface::CallbackReturn]’:
/home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller_impl.hpp:222:38:   required from here
/home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller_impl.hpp:235:38: error: ‘class hardware_interface::LoanedCommandInterface’ has no member named ‘get_prefix_name’
  235 |   if (position_command_interface_it->get_prefix_name() != joint_name_)
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/client.hpp:40,
                 from /opt/ros/humble/include/rclcpp/rclcpp/callback_group.hpp:23,
                 from /opt/ros/humble/include/rclcpp/rclcpp/any_executable.hpp:20,
                 from /opt/ros/humble/include/rclcpp/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/humble/include/rclcpp/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executor_options.hpp:20,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executor.hpp:37,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/humble/include/rclcpp/rclcpp/executors.hpp:21,
                 from /opt/ros/humble/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller.hpp:29,
                 from /home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/src/gripper_action_controller.cpp:18:
/home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller_impl.hpp:239:69: error: ‘class hardware_interface::LoanedCommandInterface’ has no member named ‘get_prefix_name’
  239 |                                   << position_command_interface_it->get_prefix_name() << "` != `"
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
In file included from /home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller.hpp:176,
                 from /home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/src/gripper_action_controller.cpp:18:
/home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller_impl.hpp:253:36: error: ‘class hardware_interface::LoanedStateInterface’ has no member named ‘get_prefix_name’
  253 |   if (position_state_interface_it->get_prefix_name() != joint_name_)
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
[... 92 lines cut ...]
                 from /home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller.hpp:29,
                 from /home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/src/gripper_action_controller.cpp:18:
/home/cobot/workspace/ros_ur_driver/src/ros2_controllers/gripper_controllers/include/gripper_controllers/gripper_action_controller_impl.hpp:275:67: error: ‘class hardware_interface::LoanedStateInterface’ has no member named ‘get_prefix_name’
  275 |                                   << velocity_state_interface_it->get_prefix_name() << "` != `"
      |                                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
gmake[2]: *** [CMakeFiles/gripper_action_controller.dir/build.make:76: CMakeFiles/gripper_action_controller.dir/src/gripper_action_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:176: CMakeFiles/gripper_action_controller.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< gripper_controllers [16.8s, exited with code 2]
[... cut ...]
Aborted  <<< joint_trajectory_controller [39.4s]                                    

Summary: 5 packages finished [39.6s]
  1 package failed: gripper_controllers
  7 packages aborted: diff_drive_controller force_torque_sensor_broadcaster forward_command_controller imu_sensor_broadcaster joint_state_broadcaster joint_trajectory_controller ur_msgs
  4 packages had stderr output: diff_drive_controller gripper_controllers joint_state_broadcaster ros2_controllers_test_nodes
  9 packages not processed
```

danzimmerman, 2022-07-13:
This is related to ROS2 Control API changes like https://github.com/ros-controls/ros2_control/pull/737 

You should be able to fix this for now if you check out slightly older released versions of `ros2_control` and/or `ros2_controllers`, see:

https://github.com/ros-controls/ros2_control/tags
https://github.com/ros-controls/ros2_controllers/tags

`2.6.0` in `ros2_controllers` and `2.10.0` in `ros2_control` work here with the UR ROS2 Driver `main` as of a few days ago. (I'm using [Robostack ROS2 Humble](https://github.com/RoboStack/ros-humble) and installing this driver from source.)

If you don't need the `ros2_controllers` source code, installing it using `apt` (or `rosdep`) should work on native Ubuntu 22.04.

ahumphries2, 2022-07-13:
That solved my problem.  Thanks!!

closed 2022-07-13 by ahumphries2

LukGen, 2022-07-18:
Thanks @ahumphries2 for your help. How did you install the specific versions for ros2-control and ros2-controllers as it seems that there are only packages 2.12.1 for ros2-control / 2.8.0 for ros2-controllers for ubuntu jammy.

fmauch, 2022-07-19:
When building from source, please always use the repos file as explained in the [README](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver#getting-started).

```
vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver-not-released.${ROS_DISTRO}.repos
```

Whenever there are upstream changes necessary to build this package, the respective versions will be noted there.

However, we suggest using the binary-released packages as long as you don't want to actively develop on the driver.