UniversalRobots/Universal_Robots_ROS2_Driver #434: Issue to build Driver on Humble

LukGen, 2022-07-13:
Hello,

I have trouble to build the ur-driver. Here is my console output:

colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
[0.230s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
	'gripper_controllers' is in: /opt/ros/humble
	'joint_trajectory_controller' is in: /opt/ros/humble
	'joint_state_broadcaster' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
	--allow-overriding gripper_controllers joint_state_broadcaster joint_trajectory_controller

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> forward_command_controller
Starting >>> joint_trajectory_controller
Starting >>> force_torque_sensor_broadcaster
Starting >>> joint_state_broadcaster
Starting >>> ur_dashboard_msgs
Starting >>> ur_description
Starting >>> ur_msgs
Starting >>> ros2_controllers_test_nodes
Finished <<< ros2_controllers_test_nodes [2.03s]                                
Starting >>> ur_client_library
Finished <<< ur_description [2.77s]                                             
Starting >>> diff_drive_controller
--- stderr: joint_state_broadcaster                                                   
In file included from /home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:15:
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:74:25: error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
   74 |   controller_interface::CallbackReturn on_init() override;
      |                         ^~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:77:25: error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
   77 |   controller_interface::CallbackReturn on_configure(
      |                         ^~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:81:25: error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
   81 |   controller_interface::CallbackReturn on_activate(
      |                         ^~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/include/joint_state_broadcaster/joint_state_broadcaster.hpp:85:25: error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
   85 |   controller_interface::CallbackReturn on_deactivate(
      |                         ^~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:49:23: error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
   49 | controller_interface::CallbackReturn JointStateBroadcaster::on_init()
      |                       ^~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:103:23: error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
  103 | controller_interface::CallbackReturn JointStateBroadcaster::on_configure(
      |                       ^~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:180:23: error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
  180 | controller_interface::CallbackReturn JointStateBroadcaster::on_activate(
      |                       ^~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:206:23: error: ‘CallbackReturn’ in namespace ‘controller_interface’ does not name a type
  206 | controller_interface::CallbackReturn JointStateBroadcaster::on_deactivate(
      |                       ^~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp: In member function ‘bool joint_state_broadcaster::JointStateBroadcaster::init_joint_data()’:
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:240:42: error: ‘const class hardware_interface::LoanedStateInterface’ has no member named ‘get_prefix_name’
  240 |     if (name_if_value_mapping_.count(si->get_prefix_name()) == 0)
      |                                          ^~~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:242:34: error: ‘const class hardware_interface::LoanedStateInterface’ has no member named ‘get_prefix_name’
  242 |       name_if_value_mapping_[si->get_prefix_name()] = {};
      |                                  ^~~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:250:32: error: ‘const class hardware_interface::LoanedStateInterface’ has no member named ‘get_prefix_name’
  250 |     name_if_value_mapping_[si->get_prefix_name()][interface_name] = kUninitializedValue;
      |                                ^~~~~~~~~~~~~~~
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp: In member function ‘virtual controller_interface::return_type joint_state_broadcaster::JointStateBroadcaster::update(const rclcpp::Time&, const rclcpp::Duration&)’:
/home/developer/ros_ur_driver/src/ros2_controllers/joint_state_broadcaster/src/joint_state_broadcaster.cpp:348:44: error: ‘const class hardware_interface::LoanedStateInterface’ has no member named ‘get_prefix_name’
  348 |     name_if_value_mapping_[state_interface.get_prefix_name()][interface_name] =
      |                                            ^~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp:57,
                 from /opt/ros/humble/include/class_loader/class_loader/class_loader.hpp:55,
[... 14 lines cut ...]
   93 |   virtual LifecycleNodeInterface::CallbackReturn on_init() = 0;
      |                                                  ^~~~~~~
gmake[2]: *** [CMakeFiles/joint_state_broadcaster.dir/build.make:76: CMakeFiles/joint_state_broadcaster.dir/src/joint_state_broadcaster.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:178: CMakeFiles/joint_state_broadcaster.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< joint_state_broadcaster [32.6s, exited with code 2]
Aborted  <<< diff_drive_controller [29.9s]                                                       
Aborted  <<< ur_client_library [30.6s]
Aborted  <<< forward_command_controller [32.7s]
Aborted  <<< force_torque_sensor_broadcaster [32.9s]                                              
Aborted  <<< joint_trajectory_controller [33.5s]                                                  
Aborted  <<< ur_msgs [37.8s]                                                                                     
Aborted  <<< ur_dashboard_msgs [40.6s]                                   

Summary: 2 packages finished [40.8s]
  1 package failed: joint_state_broadcaster
  7 packages aborted: diff_drive_controller force_torque_sensor_broadcaster forward_command_controller joint_trajectory_controller ur_client_library ur_dashboard_msgs ur_msgs
  5 packages had stderr output: diff_drive_controller force_torque_sensor_broadcaster forward_command_controller joint_state_broadcaster joint_trajectory_controller
  12 packages not processed

Can anybody help me?

Thank you

ahumphries2, 2022-07-16:
Looks like you are having the same issue I was. 
[link](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/433)

fmauch, 2022-07-19:
Yes, this seems to ne a duplicate of #433. Please see my [comment](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/433#issuecomment-1188745408) there.