UniversalRobots/Universal_Robots_ROS2_Driver #437: Robot_state not shown

ludo2811, 2022-07-14:
Recently I was trying to connect my PC with Ros2 Foxy to my ur3e. So I installed the foxy branch of this ROS2 ur driver and all necessary software on my ur3e. 
Unfortunately the connection is not working since the robot always states "connection refused". Ping is working and the IPs are correct. Then I want to start the launch file for testing the driver with the following result:

fams@fams-VirtualBox:~$ cd ~/workspace/ros_ur_driver
fams@fams-VirtualBox:~/workspace/ros_ur_driver$ source install/setup.bash
fams@fams-VirtualBox:~/workspace/ros_ur_driver$ ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.1.105 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/fams/.ros/log/2022-07-14-11-32-16-951038-fams-VirtualBox-24732
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [24738]
[INFO] [controller_stopper_node-3]: process started with pid [24740]
[INFO] [ros2_control_node-1]: process started with pid [24736]
[INFO] [robot_state_publisher-4]: process started with pid [24742]
[INFO] [rviz2-5]: process started with pid [24774]
[INFO] [spawner.py-6]: process started with pid [24776]
[INFO] [spawner.py-7]: process started with pid [24780]
[INFO] [spawner.py-8]: process started with pid [24782]
[INFO] [spawner.py-9]: process started with pid [24789]
[INFO] [spawner.py-10]: process started with pid [24796]
[ros2_control_node-1] malloc(): invalid size (unsorted)
[dashboard_client-2] [INFO] [1657791138.265105713] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1657791138.262153851] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[robot_state_publisher-4] Parsing robot urdf xml string.
[robot_state_publisher-4] Link base had 0 children
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 1 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] [INFO] [1657791138.359597260] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1657791138.384010209] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1657791138.384056882] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1657791138.384077200] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1657791138.384096310] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1657791138.384115178] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1657791138.384205983] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1657791138.384225528] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1657791138.384318717] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1657791138.384339677] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1657791138.384358946] [robot_state_publisher]: got segment wrist_3_link
[spawner.py-8] [INFO] [1657791139.108360360] [spawner_speed_scaling_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-6] [INFO] [1657791139.167561915] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-10] [INFO] [1657791139.184530685] [spawner_joint_trajectory_controller]: Waiting for /controller_manager services
[spawner.py-9] [INFO] [1657791139.271175677] [spawner_force_torque_sensor_broadcaster]: Waiting for /controller_manager services
[spawner.py-7] [INFO] [1657791139.273189289] [spawner_io_and_status_controller]: Waiting for /controller_manager services
[ERROR] [ros2_control_node-1]: process has died [pid 24736, exit code -6, cmd '/opt/ros/foxy/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_l_ux686t --params-file /home/fams/workspace/ros_ur_driver/install/ur_bringup/share/ur_bringup/config/ur_controllers.yaml'].
[rviz2-5] [INFO] [1657791140.293706763] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] [INFO] [1657791140.294026651] [rviz2]: OpenGl version: 3.1 (GLSL 1.4)
[rviz2-5] [INFO] [1657791140.502385841] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-5] Parsing robot urdf xml string.
[spawner.py-8] [INFO] [1657791141.122100151] [spawner_speed_scaling_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-6] [INFO] [1657791141.189613158] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-10] [INFO] [1657791141.200006164] [spawner_joint_trajectory_controller]: Waiting for /controller_manager services
[spawner.py-9] [INFO] [1657791141.283716568] [spawner_force_torque_sensor_broadcaster]: Waiting for /controller_manager services
[spawner.py-7] [INFO] [1657791141.291778080] [spawner_io_and_status_controller]: Waiting for /controller_manager services
[spawner.py-8] [INFO] [1657791143.137959814] [spawner_speed_scaling_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-6] [INFO] [1657791143.203515077] [spawner_joint_state_broadcaster]: Waiting for /controller_manager services
[spawner.py-10] [INFO] [1657791143.210703260] [spawner_joint_trajectory_controller]: Waiting for /controller_manager services
[spawner.py-9] [INFO] [1657791143.295685921] [spawner_force_torque_sensor_broadcaster]: Waiting for /controller_manager services
[spawner.py-7] [INFO] [1657791143.303284021] [spawner_io_and_status_controller]: Waiting for /controller_manager services
[rviz2-5] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[... 388 lines cut ...]
[rviz2-5] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_2_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[INFO] [robot_state_publisher-4]: process has finished cleanly [pid 24742]
[INFO] [dashboard_client-2]: process has finished cleanly [pid 24738]
[rviz2-5] Warning: Invalid frame ID "forearm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist
[... cut ...]
[ERROR] [controller_stopper_node-3]: process has died [pid 24740, exit code -6, cmd '/home/fams/workspace/ros_ur_driver/install/ur_robot_driver/lib/ur_robot_driver/controller_stopper_node --ros-args -r __node:=controller_stopper --params-file /tmp/launch_params_a_xdbqam'].

Therefore, RVIZ also does not show the right  position of the robot. I also tried to rebuild my workspace.
Does anyone has a suggestion on how to solve my issues (Connection and launching)? I guess the connection problem could be solved when the launch files work properly.

ludo2811, 2022-07-18:
I have now installed the galactic branch on another pc. The connection problem to the robot is gone now, but I have the same problem with subscribing to topics like /joint_states or /tf. Therefore, the robot_state still is not shown in rviz. In the following code it says that e.g. joint_state_broadcaster not found. I do not know how to fix this problem. Maybe someone can help with my problem. More code in the following:

ros2 launch ur_bringup ur_control.launch.py ur_type:=ur3e robot_ip:=192.168.56.105 launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/fams2/.ros/log/2022-07-18-11-06-58-392450-fams2-VirtualBox-83577
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [83583]
[INFO] [controller_stopper_node-3]: process started with pid [83585]
[INFO] [ur_ros2_control_node-1]: process started with pid [83581]
[INFO] [robot_state_publisher-4]: process started with pid [83587]
[INFO] [rviz2-5]: process started with pid [83617]
[INFO] [spawner-6]: process started with pid [83619]
[INFO] [spawner-7]: process started with pid [83622]
[INFO] [spawner-8]: process started with pid [83624]
[INFO] [spawner-9]: process started with pid [83635]
[INFO] [spawner-10]: process started with pid [83637]
[INFO] [spawner-11]: process started with pid [83639]
[ur_ros2_control_node-1] [INFO] [1658135221.121029396] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1658135221.123586548] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] [WARN] [1658135221.144160160] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[robot_state_publisher-4] Link base_link had 2 children
[robot_state_publisher-4] Link base had 0 children
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 2 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] Link ft_frame had 0 children
[robot_state_publisher-4] [INFO] [1658135221.195036774] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1658135221.195167658] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1658135221.195196455] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1658135221.195220824] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1658135221.195244976] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1658135221.195269115] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1658135221.195293211] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1658135221.195317462] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1658135221.195341380] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1658135221.195365606] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1658135221.195389534] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1658135221.195413699] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1658135221.195437854] [robot_state_publisher]: got segment wrist_3_link
[dashboard_client-2] [INFO] [1658135221.126337808] [UR_Client_Library]: Connected: Universal Robots Dashboard Server
[dashboard_client-2] 
[controller_stopper_node-3] [INFO] [1658135221.123087916] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] [INFO] [1658135221.586266859] [UR_Client_Library]: Negotiated RTDE protocol version to 2.
[ur_ros2_control_node-1] [INFO] [1658135221.586620757] [UR_Client_Library]: Setting up RTDE communication with frequency 500.000000
[spawner-9] [INFO] [1658135221.840497124] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-8] [INFO] [1658135221.979201044] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-6] [INFO] [1658135222.071560098] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-7] [INFO] [1658135222.252300901] [spawner_io_and_status_controller]: Waiting for '/controller_manager' services to be available
[spawner-10] [INFO] [1658135222.266676986] [spawner_forward_position_controller]: Waiting for '/controller_manager' services to be available
[spawner-11] [INFO] [1658135222.332297484] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' services to be available
[ur_ros2_control_node-1] [INFO] [1658135222.605757810] [URPositionHardwareInterface]: Calibration checksum: 'calib_16756443741236045476'.
[ur_ros2_control_node-1] [WARN] [1658135222.606754743] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[... cut ...]
[rviz2-5] Warning: Invalid frame ID "shoulder_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "upper_arm_link" passed to canTransform argument source_frame - frame does not exist
[rviz2-5]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.3/src/buffer_core.cpp
[rviz2-5] Warning: Invalid frame ID "wrist_1_link" passed to canTransform argument source_frame - frame does not exist

fmauch, 2022-07-26:
I've edited your posts such that the console output is inside a [code block](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks). Please do the same in future posts.

To make the outout more readable, I suggest to pass `launch_rviz:=false` to the launchfile that will prevent the wall of text produced by rviz.

Let's debug this bit by bit. On Foxy your hardware interface seems to have died. That should not happen and should further be debugged.

On Galactic there seem to be a couple of dependencies missing.

FlorianPix, 2023-01-16:
I've got the same problem with a ur10e on foxy. How could I debug the hardware interface ?

```
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur10e robot_ip:=172.19.0.10 launch_rviz:=false
[ros2_control_node-1] [WARN] [1673882198.185045165] [UR_Client_Library]: No realtime capabilities found. Consider using a realtime system for better performance
[ros2_control_node-1] [WARN] [1673882198.593959388] [UR_Client_Library]: Did not receive answer to RTDE pause request. Message received instead: 
[ros2_control_node-1] actual_current: [0.0332751, -0.0147321, -0.00627446, -0.00629029, -0.0168215, -0.0139855]
[ros2_control_node-1] safety_status_bits: 1
[ros2_control_node-1] robot_status_bits: 1
[ros2_control_node-1] robot_mode: 5
[ros2_control_node-1] tool_output_current: 0.015767
[ros2_control_node-1] tool_analog_input0: 0
[ros2_control_node-1] tool_mode: 253
[ros2_control_node-1] analog_io_types: 3
[ros2_control_node-1] standard_analog_output1: 0.004
[ros2_control_node-1] tool_analog_input_types: 3
[ros2_control_node-1] actual_q: [3.2191, -1.97498, -1.50792, 0.25015, 1.51214, 0.0671201]
[ros2_control_node-1] actual_qd: [0, 0, -0, 0, 0, 0]
[ros2_control_node-1] safety_mode: 1
[ros2_control_node-1] tool_temperature: 28.3125
[ros2_control_node-1] timestamp: 4493.8
[ros2_control_node-1] speed_scaling: 0
[ros2_control_node-1] tool_output_voltage: 24
[ros2_control_node-1] target_speed_fraction: 1
[ros2_control_node-1] runtime_state: 1
[ros2_control_node-1] actual_TCP_force: [-0.256705, 9.89861, 0.411029, -0.192473, -0.0154221, 0.274553]
[ros2_control_node-1] actual_TCP_pose: [-0.974025, 0.109861, 0.655807, -1.31905, -1.25642, 1.22262]
[ros2_control_node-1] actual_digital_input_bits: 0
[ros2_control_node-1] tool_analog_input1: 0
[ros2_control_node-1] standard_analog_output0: 0.004
[ros2_control_node-1] actual_digital_output_bits: 0
[ros2_control_node-1] standard_analog_input0: -0
[ros2_control_node-1] standard_analog_input1: -0
[ros2_control_node-1] 
[ros2_control_node-1] [FATAL] [1673882204.598856444] [URPositionHardwareInterface]: Variable 'speed_slider_mask' is [ERROR] [ros2_control_node-1]: process has died [pid 8302, exit code -11, cmd '/opt/ros/foxy/lib/controller_manager/ros2_control_node --ros-args --params-file /tmp/launch_params_9mkixa83 --params-file /home/florian/git_repos/ur10_ws/install/ur_bringup/share/ur_bringup/config/ur_controllers.yaml'].
```

VinDp, 2024-06-12:
Since this is pretty old and there were no further replies after first suggestions, I will close the issue.

closed 2024-06-12 by VinDp