UniversalRobots/Universal_Robots_ROS2_Driver #438: Galactic - MoveIt Planning Library doesn't load

ahumphries2, 2022-07-16:
Running ROS2 Galactic and tried installing UR ROS2 drivers for Galactic from binary and source and get the same results.

Per the documentation, I run this command in one terminal without issue:
`ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false initial_joint_controller:=joint_trajectory_controller`

Then I run this command in another terminal to bring up Rviz and MoveIt:
`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true`

Rviz opens up with a UR5e, but motion planning library refuses to load.
[image]

Here is the terminal output from the ur_moveit.launch.py command:
```
cobot@cobot-Precision-7560:~$ ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/cobot/.ros/log/2022-07-15-21-52-22-195823-cobot-Precision-7560-92544
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [92552]
[ERROR] [move_group-1]: process has died [pid 92552, exit code 127, cmd '/opt/ros/galactic/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_6vdqrb6q --params-file /tmp/launch_params_vd7qwrh4 --params-file /home/cobot/workspace/ros_ur_driver/install/ur_moveit_config/share/ur_moveit_config/config/kinematics.yaml --params-file /tmp/launch_params_pf32um75 --params-file /tmp/launch_params_8y6py06t --params-file /tmp/launch_params_crp9a_qc --params-file /tmp/launch_params_766uk6lc --params-file /tmp/launch_params_8qh00f2d'].
[INFO] [mongo_wrapper_ros.py-2]: process started with pid [92554]
[INFO] [rviz2-3]: process started with pid [92556]
[INFO] [servo_node_main-4]: process started with pid [92558]
[ERROR] [servo_node_main-4]: process has died [pid 92558, exit code 127, cmd '/opt/ros/galactic/lib/moveit_servo/servo_node_main --ros-args --params-file /tmp/launch_params_yzlur3di --params-file /tmp/launch_params_jmcpi9xv --params-file /tmp/launch_params_66i3eir9'].
[move_group-1] /opt/ros/galactic/lib/moveit_ros_move_group/move_group: error while loading shared libraries: libbackward.so: cannot open shared object file: No such file or directory
[servo_node_main-4] /opt/ros/galactic/lib/moveit_servo/servo_node_main: error while loading shared libraries: libbackward.so: cannot open shared object file: No such file or directory
[mongo_wrapper_ros.py-2] [INFO] [1657936342.950759074] [mongodb]: Starting mongodb with db location /tmp/db listening on localhost:33829
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten] MongoDB starting : pid=92580 port=33829 dbpath=/tmp/db 64-bit host=cobot-Precision-7560
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten] db version v3.6.8
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten] git version: 8e540c0b6db93ce994cc548f000900bdc740f80a
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten] OpenSSL version: OpenSSL 1.1.1f  31 Mar 2020
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten] allocator: tcmalloc
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten] modules: none
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten] build environment:
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten]     distarch: x86_64
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten]     target_arch: x86_64
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.979-0400 I CONTROL  [initandlisten] options: { net: { port: 33829 }, storage: { dbPath: "/tmp/db" } }
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.980-0400 I -        [initandlisten] Detected data files in /tmp/db created by the 'wiredTiger' storage engine, so setting the active storage engine to 'wiredTiger'.
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.980-0400 I STORAGE  [initandlisten] 
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.980-0400 I STORAGE  [initandlisten] ** WARNING: Using the XFS filesystem is strongly recommended with the WiredTiger storage engine
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.980-0400 I STORAGE  [initandlisten] **          See http://dochub.mongodb.org/core/prodnotes-filesystem
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:22.980-0400 I STORAGE  [initandlisten] wiredtiger_open config: create,cache_size=15142M,session_max=20000,eviction=(threads_min=4,threads_max=4),config_base=false,statistics=(fast),cache_cursors=false,compatibility=(release="3.0",require_max="3.0"),log=(enabled=true,archive=true,path=journal,compressor=snappy),file_manager=(close_idle_time=100000),statistics_log=(wait=0),verbose=(recovery_progress),
[rviz2-3] [INFO] [1657936343.111337742] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] [INFO] [1657936343.111396487] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[rviz2-3] [INFO] [1657936343.120059445] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-3] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occured with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
[rviz2-3]          at line 253 in /opt/ros/galactic/include/class_loader/class_loader_core.hpp
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.209-0400 I STORAGE  [initandlisten] WiredTiger message [1657936343:209372][92580:0x7f4d0140ff40], txn-recover: Main recovery loop: starting at 9/9472
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.267-0400 I STORAGE  [initandlisten] WiredTiger message [1657936343:267854][92580:0x7f4d0140ff40], txn-recover: Recovering log 9 through 10
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.310-0400 I STORAGE  [initandlisten] WiredTiger message [1657936343:310008][92580:0x7f4d0140ff40], txn-recover: Recovering log 10 through 10
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.339-0400 I STORAGE  [initandlisten] WiredTiger message [1657936343:339106][92580:0x7f4d0140ff40], txn-recover: Set global recovery timestamp: 0
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] ** WARNING: Access control is not enabled for the database.
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] **          Read and write access to data and configuration is unrestricted.
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] ** WARNING: This server is bound to localhost.
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] **          Remote systems will be unable to connect to this server. 
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] **          Start the server with --bind_ip <address> to specify which IP 
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] **          addresses it should serve responses from, or with --bind_ip_all to
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] **          bind to all interfaces. If this behavior is desired, start the
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] **          server with --bind_ip 127.0.0.1 to disable this warning.
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.351-0400 I CONTROL  [initandlisten] 
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.373-0400 I FTDC     [initandlisten] Initializing full-time diagnostic data capture with directory '/tmp/db/diagnostic.data'
[mongo_wrapper_ros.py-2] 2022-07-15T21:52:23.373-0400 I NETWORK  [initandlisten] waiting for connections on port 33829
[rviz2-3] [ERROR] [1657936346.188811004] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-3] [INFO] [1657936346.260605419] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00230597 seconds
[rviz2-3] [INFO] [1657936346.260677196] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[... cut ...]  soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[rviz2-3] [INFO] [1657936411.465258918] [move_group_interface]: Ready to take commands for planning group ur_manipulator.
[rviz2-3] [INFO] [1657936411.465295129] [move_group_interface]: Looking around: no
[rviz2-3] [INFO] [1657936411.465304769] [move_group_interface]: Replanning: no
```

Any assistance is greatly appreciated!  Thanks

fmauch, 2022-07-26:
This seems to be a dependency problem. We have recently updated the package's dependencies and have not made a binary release since.

Since you already tried a source install: Did you call the `rosdep install` command after cloning / updating to the latest `galactic` commit?

Could you please post the output of `apt search ros-galactic-moveit`? This should print the state of all moveit packages whether they are installed or not. If there's a package missing, we should find it there.

renzobazan96, 2022-07-28:
I have the same error with the galactic branch. I built from source and called  `rosdep install `. Here is the output of `apt search ros-galactic-moveit`:
```
ros-galactic-moveit/focal 2.3.4-1focal.20220526.154855 amd64
  Meta package that contains all essential packages of MoveIt 2

ros-galactic-moveit-chomp-optimizer-adapter/focal 2.3.4-1focal.20220514.025535 amd64
  MoveIt planning request adapter utilizing chomp for solution optimization

ros-galactic-moveit-chomp-optimizer-adapter-dbgsym/focal 2.3.4-1focal.20220514.025535 amd64
  debug symbols for ros-galactic-moveit-chomp-optimizer-adapter

ros-galactic-moveit-common/focal 2.3.4-1focal.20220430.053314 amd64
  Common support functionality used throughout MoveIt

ros-galactic-moveit-configs-utils/focal,now 2.3.4-1focal.20220526.150119 amd64 [installed,automatic]
  Python library for loading moveit config parameters in launch files

ros-galactic-moveit-core/focal,now 2.3.4-1focal.20220514.022605 amd64 [installed,automatic]
  Core libraries used by MoveIt

ros-galactic-moveit-core-dbgsym/focal 2.3.4-1focal.20220514.022605 amd64
  debug symbols for ros-galactic-moveit-core

ros-galactic-moveit-hybrid-planning/focal 2.3.4-1focal.20220601.135034 amd64
  Hybrid planning components of MoveIt 2

ros-galactic-moveit-hybrid-planning-dbgsym/focal 2.3.4-1focal.20220601.135034 amd64
  debug symbols for ros-galactic-moveit-hybrid-planning

ros-galactic-moveit-kinematics/focal,now 2.3.4-1focal.20220526.150941 amd64 [installed]
  Package for all inverse kinematics solvers in MoveIt

ros-galactic-moveit-kinematics-dbgsym/focal 2.3.4-1focal.20220526.150941 amd64
  debug symbols for ros-galactic-moveit-kinematics

ros-galactic-moveit-msgs/focal,now 2.2.0-1focal.20220430.125544 amd64 [installed,automatic]
  Messages, services and actions used by MoveIt

ros-galactic-moveit-msgs-dbgsym/focal 2.2.0-1focal.20220430.125544 amd64
  debug symbols for ros-galactic-moveit-msgs

ros-galactic-moveit-planners/focal 2.3.4-1focal.20220526.154721 amd64
  Meta package that installs all available planners for MoveIt

ros-galactic-moveit-planners-chomp/focal 2.3.4-1focal.20220514.025537 amd64
  The interface for using CHOMP within MoveIt

ros-galactic-moveit-planners-chomp-dbgsym/focal 2.3.4-1focal.20220514.025537 amd64
  debug symbols for ros-galactic-moveit-planners-chomp

ros-galactic-moveit-planners-ompl/focal,now 2.3.4-1focal.20220514.031330 amd64 [installed]
  MoveIt interface to OMPL

ros-galactic-moveit-planners-ompl-dbgsym/focal 2.3.4-1focal.20220514.031330 amd64
  debug symbols for ros-galactic-moveit-planners-ompl

ros-galactic-moveit-plugins/focal 2.3.4-1focal.20220514.025538 amd64
  Metapackage for MoveIt plugins.

ros-galactic-moveit-resources/focal 2.0.3-1focal.20220430.171124 amd64
  Resources used for MoveIt testing

ros-galactic-moveit-resources-fanuc-description/focal 2.0.3-1focal.20220124.200203 amd64
  Fanuc Resources used for MoveIt testing

ros-galactic-moveit-resources-fanuc-moveit-config/focal 2.0.3-1focal.20220430.143330 amd64
  MoveIt Resources for testing: Fanuc M-10iA.

ros-galactic-moveit-resources-panda-description/focal 2.0.3-1focal.20220124.200204 amd64
  panda Resources used for MoveIt testing
[... 81 lines cut ...]

ros-galactic-moveit-ros-warehouse-dbgsym/focal 2.3.4-1focal.20220514.031328 amd64
  debug symbols for ros-galactic-moveit-ros-warehouse

ros-galactic-moveit-runtime/focal 2.3.4-1focal.20220526.154917 amd64
  moveit_runtime meta package contains MoveIt packages that are essential for its runtime (e.g.

ros-galactic-moveit-servo/focal,now 2.3.4-1focal.20220601.140103 amd64 [installed]
  Provides real-time manipulator Cartesian and joint servoing.

ros-galactic-moveit-servo-dbgsym/focal 2.3.4-1focal.20220601.140103 amd64
  debug symbols for ros-galactic-moveit-servo

ros-galactic-moveit-setup-assistant/focal 2.3.4-1focal.20220526.154731 amd64
  Generates a configuration package that makes it easy to use MoveIt

ros-galactic-moveit-setup-assistant-dbgsym/focal 2.3.4-1focal.20220526.154731 amd64
  debug symbols for ros-galactic-moveit-setup-assistant

ros-galactic-moveit-simple-controller-manager/focal,now 2.3.4-1focal.20220514.025116 amd64 [installed]
  A generic, simple controller manager plugin for MoveIt.

ros-galactic-moveit-simple-controller-manager-dbgsym/focal 2.3.4-1focal.20220514.025116 amd64
  debug symbols for ros-galactic-moveit-simple-controller-manager
```

Topasm, 2022-08-27:
@fmauch Same error here, I checked node list and rqt_graph but cannot find anything similar to move_group node.

I can see my robot when launch
 `ros2 launch ur_bringup ur_control.launch.py ur_type:=ur3 robot_ip:=yyy.yyy.yyy.yyy  launch_rviz:=true initial_joint_controller:=joint_trajectory_controller` same pose as real robot.

but when launch moveit it shows No planning library loaded message and MotionPlanning status ⚠️ also 
planningScene message is Requesting inition scene failed.

I'm using Ros2 Galactic and semi-binary installed UR driver for UR3

Topasm, 2022-08-29:
Answer myself,
'sudo apt install ros-galactic-backward-ros' or 'sudo apt install ros-<ROS DISTRO>-backward-ros'
Can solve this problem.

fmauch, 2022-09-19:
> Answer myself, `sudo apt install ros-galactic-backward-ros` or `sudo apt install ros-<ROS DISTRO>-backward-ros` Can solve this problem.

As far as I can follow the package trace on my testing machine that should be installed with the current version, as the dependency goes like this: `ur_moveit_config`->`moveit_simple_controller_manager`->`moveit_common`->`backward_ros`. So this should be addressed since #392.

niemsoen, 2022-11-29:
I had the same problem, I wrote a workaround in the moveit launch file which lets the rviz launch wait for the move_group being finished (it prints the message "You can start planning now"). It's a bit hacky but it does the job.

this is in my `def launch_setup(context, *args, **kwargs):`

```python
from launch.actions import RegisterEventHandler
from launch.event_handlers import OnProcessIO
from launch.event import Event

# define your nodes here
rviz_node = ...
move_group_node = ...
mongodb_server_node = ...
servo_node = ...

move_group_loaded_message = "You can start planning now"
def on_output(event: Event):
    for line in event.text.decode().splitlines():
        # print('[{}] {}'.format(
        #     cast(ProcessIO, event).process_name, line))
        if move_group_loaded_message in line:
            return [rviz_node, move_group_interface_server]
delay_rviz = RegisterEventHandler(OnProcessIO(
    on_stdout=on_output,
    on_stderr=on_output,
))

nodes_to_start = [
    move_group_node,
    delay_rviz, # starts rviz once move group is ready
    mongodb_server_node,
    servo_node,
]

```
For reference see https://github.com/ros2/launch/blob/b78d87f31bb995c7226df66826e1a57079ae3a3f/launch/examples/launch_counters.py#L51-L62
and https://docs.ros.org/en/galactic/Tutorials/Intermediate/Launch/Using-Event-Handlers.html

fmauch, 2023-01-03:
@niemsoen really, that's the problem? Thanks for sharing this workaround! But I agree that it's a bit hacky and string matching from output is always a bit brittle (Although this output stayed the same as long as I can remember).

VinDp, 2024-06-05:
Since Galactic is not supported anymore and we don't have this issue in for the latest distros, I will close it.

closed 2024-06-05 by VinDp