UniversalRobots/Universal_Robots_ROS2_Driver #445: Humble: Speed scaling interface not found in hardware interface[bug]

luozhijing, 2022-07-25:
hello, recently, i tried to test the driver using URsim,  I follow the step as below:
1. External control-1.0.5.urcap was successfully installed and external_control was created and loaded;
2. Run the command "ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur10 robot_ip:=192.168.65.129 use_fake_hardware:=false launch_rviz:=false"
3. ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur10
Expected behaviour
use moveit to plan and execute the trajectory

Actual behaviour
Planed to a goal successfully but trajecory exectution failed for controller is not running

controller status:
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active    
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active    
io_and_status_controller[ur_controllers/GPIOController] active    
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active    
forward_position_controller[position_controllers/JointGroupPositionController] inactive  
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] inactive  

Node information
[INFO] [launch]: All log files can be found below /home/luo/.ros/log/2022-07-25-09-03-41-382979-luo-virtual-machine-49677
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [49682]
[INFO] [controller_stopper_node-3]: process started with pid [49684]
[INFO] [ur_ros2_control_node-1]: process started with pid [49680]
[INFO] [robot_state_publisher-4]: process started with pid [49686]
[INFO] [spawner-5]: process started with pid [49688]
[INFO] [spawner-6]: process started with pid [49690]
[INFO] [spawner-7]: process started with pid [49692]
[INFO] [spawner-8]: process started with pid [49694]
[INFO] [spawner-9]: process started with pid [49696]
[INFO] [spawner-10]: process started with pid [49738]
[controller_stopper_node-3] [INFO] [1658711021.992582544] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] [INFO] [1658711022.013697486] [resource_manager]: Loading hardware 'ur10' 
[robot_state_publisher-4] Link base_link had 2 children
[ur_ros2_control_node-1] [INFO] [1658711022.017282529] [resource_manager]: Initialize hardware 'ur10' 
[ur_ros2_control_node-1] [INFO] [1658711022.017477401] [resource_manager]: Successful initialization of hardware 'ur10'
[ur_ros2_control_node-1] [INFO] [1658711022.017798714] [resource_manager]: 'configure' hardware 'ur10' 
[ur_ros2_control_node-1] [INFO] [1658711022.017819880] [resource_manager]: Successful 'configure' of hardware 'ur10'
[ur_ros2_control_node-1] [INFO] [1658711022.017864325] [resource_manager]: 'activate' hardware 'ur10' 
[ur_ros2_control_node-1] [INFO] [1658711022.017881527] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1658711022.017934061] [URPositionHardwareInterface]: Initializing driver...
[robot_state_publisher-4] Link base had 0 children
[ur_ros2_control_node-1] Warning: No realtime capabilities found. Consider using a realtime system for better performance
[ur_ros2_control_node-1]          at line 396 in ./include/ur_client_library/comm/pipeline.h
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 2 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] Link ft_frame had 0 children
[robot_state_publisher-4] [INFO] [1658711022.024026949] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1658711022.024407196] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1658711022.024443632] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1658711022.024458150] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1658711022.024471130] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1658711022.024482735] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1658711022.024494310] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1658711022.024505878] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1658711022.024517595] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1658711022.024529181] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1658711022.024540287] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1658711022.024551768] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1658711022.024563718] [robot_state_publisher]: got segment wrist_3_link
[spawner-8] [INFO] [1658711022.303116557] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-7] [INFO] [1658711022.306384320] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' services to be available
[spawner-6] [INFO] [1658711022.321547808] [spawner_io_and_status_controller]: Waiting for '/controller_manager' node to exist
[... 33 lines cut ...]
[ur_ros2_control_node-1] [INFO] [1658711024.744560037] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ur_ros2_control_node-1] [INFO] [1658711024.744644352] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ur_ros2_control_node-1] [INFO] [1658711024.762178447] [controller_manager]: Configuring controller 'forward_position_controller'
[ur_ros2_control_node-1] [INFO] [1658711024.769458657] [forward_position_controller]: configure successful
[spawner-9] [WARN] [1658711024.770356781] [spawner_forward_position_controller]: "--stopped" flag is deprecated use "--inactive" instead
[ur_ros2_control_node-1] [INFO] [1658711024.771621465] [controller_manager]: Configuring controller 'scaled_joint_trajectory_controller'
[ur_ros2_control_node-1] [INFO] [1658711024.772083049] [scaled_joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ur_ros2_control_node-1] [INFO] [1658711024.778366978] [scaled_joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ur_ros2_control_node-1] [INFO] [1658711024.782538325] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-7] [INFO] [1658711024.827694258] [spawner_speed_scaling_state_broadcaster]: Configured and activated speed_scaling_state_broadcaster
[spawner-6] [INFO] [1658711024.890843170] [spawner_io_and_status_controller]: Configured and activated io_and_status_controller
[spawner-5] [INFO] [1658711024.905259648] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[INFO] [spawner-9]: process has finished cleanly [pid 49696]
[ur_ros2_control_node-1] [ERROR] [1658711024.921636548] [scaled_joint_trajectory_controller]: Speed scaling interface not found in hardware interface.
[spawner-10] [INFO] [1658711024.924125919] [spawner_scaled_joint_trajectory_controller]: Configured and activated scaled_joint_trajectory_controller
[ur_ros2_control_node-1] [ERROR] [1658711024.931347775] [scaled_joint_trajectory_controller]: Speed scaling interface not found in hardware interface.
[ur_ros2_control_node-1] [ERROR] [1658711024.939526352] [scaled_joint_trajectory_controller]: Speed scaling interface not found in hardware interface.
[ur_ros2_control_node-1] [WARN] [1658711024.940153035] [controller_manager]: 'stop_controllers' field is deprecated, use 'deactivate_controllers' field instead!
[spawner-8] [INFO] [1658711024.942111912] [spawner_force_torque_sensor_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[ur_ros2_control_node-1] [ERROR] [1658711024.949800424] [scaled_joint_trajectory_controller]: Speed scaling interface not found in hardware interface.
[INFO] [spawner-7]: process has finished cleanly [pid 49692]
[INFO] [spawner-6]: process has finished cleanly [pid 49690]
[INFO] [spawner-5]: process has finished cleanly [pid 49688]
[INFO] [spawner-10]: process has finished cleanly [pid 49738]
[INFO] [spawner-8]: process has finished cleanly [pid 49694]

luozhijing, 2022-07-25:
Can someone help me?

kschwan, 2022-07-25:
Just wanted to say that I experience the same issue. I don't have any solution

fmauch, 2022-08-01:
Please see #448 With the current main branch this should be resolved. I'll go ahead and close this, but feel free to comment or reopen if the problem persists.

closed 2022-08-01 by fmauch

Maxeva, 2023-09-07:
I have the same issue as described above.

scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] inactive

Also need it to run MoveIt, any solution for that?

fmauch, 2023-09-12:
@Maxeva could you please open a new issue with more information of your system and log output? The issue mentioned above should not exist anymore. So, if it does for you we would need some further information.