UniversalRobots/Universal_Robots_ROS2_Driver #452: ros2 foxy moveit2 move_group_interface can't find planning group

jiaqiwu1999, 2022-07-27:
Hi, 
We have a ur5e arm and we want to control it from ros2 scipt. We used 
ros2 launch ur_bringup ur_control.launch.py ur_type:=ur5e robot_ip:=192.168.50.95 use_fake_hardware:=false launch_rviz:=false
and then
ros2 launch ur_bringup ur_moveit.launch.py ur_type:=ur5e robot_ip:="192.168.50.95" use_fake_hardware:=false launch_rviz:=true 
We're able to control the robot arm from Rviz. 

However, when we tried to initiate a move_group object in our ROS cpp file and ros run the node as such:
    using moveit::planning_interface::MoveGroupInterface;
    auto move_group_interface = MoveGroupInterface(node, "ur_manipulator");
We encountered such error:
[INFO] [1658952783.078692872] [moveit_rdf_loader.rdf_loader]: Robot model parameter not found! Did you remap 'robot_description'?

[FATAL] [1658952783.078782619] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.

We knew that there was a similar issue in #145, but it seems that the situation is different from it because the launch files are different. We believe the name of the planning group to be ur_manipulator, and we have no idea why the moveit in ros doesn't register the ur5e. Thank you in advance!

UttshaKhan, 2022-07-28:
I am also stuck here and getting exactly the same error.

destogl, 2022-07-30:
This sounds more like issue with moveIt and not the driver. Is your robot description there?

UttshaKhan, 2022-07-31:
There are  'robot_description's under multiple nodes, when checked with ros2 param list.

fmauch, 2022-08-01:
Is the description available for the node containing your C++ interface? I agree with @destogl that this is most probably not an issue with anything from this repository which is why I added the `invalid` tag.

UttshaKhan, 2022-08-01:
I agree, this is not an issue from this repo. There was no description under the moveit node, I had to write a launch
 file to add all the parameters.

closed 2022-08-02 by fmauch