UniversalRobots/Universal_Robots_ROS2_Driver #457: ur10 cannot find robot_descrition semantic in ros humble

gjane5, 2022-08-02:
Hello,
 I am trying to use the moveit_config files from the ur repository together with the ur driver. I can launch the driver and the moveit package with the commands:

`ros2 launch ur_robot_driver ur10.launch.py robot_ip:=localhost use_fake_hardware:=true launch_rviz:=false initial_joint_controller:=joint_trajectory_controller`
`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur10 launch_rviz:=true `

This launches the ur_driver and also rviz with moveit. I can plan and execute a motion with the motion planning plugin and it all works out well. 
However when I try to publish a motion via script I get the error as below

`[ERROR] [1659434560.178463185] [moveit_ur_test]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.  Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0                                                                                                                      at line 715 in /home/******/ws_moveit2/src/srdfdom/src/model.cpp                                                                                                                                         [ERROR] [1659434560.193027035] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF                                                                                                                                   [FATAL] [1659434560.193493008] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server.                                                           terminate called after throwing an instance of 'std::runtime_error'                                                                                                                                                     what():  Unable to construct robot model. Please make sure all needed information is on the parameter server.                                                                                                       [ros2run]: Aborted`

I have loaded the script after launching the ur_driver and moveit files and also sourcing the workspaces. I am  not sure if I am missing something
Can anyone help me with this error and elp me figure out what I am doing wrong?

vetletj, 2022-08-02:
Hello @gjane5, 

I'm sorry but I'm not able help you out, but i was wondering how you were able to execute trajectory's with RViz MotionPlanning. 

I'm running on Ubuntu 22.04, ROS 2 Humble (Debian desktop install), and followed the instructions from UR Driver main branch. When I'm running the same ur_driver and rviz with moveit as you I am getting MoveGroupInterface error... Have you encountered the same error before or did everything, except what you're writing about above, work perfectly?

gjane5, 2022-08-02:
> Hello @gjane5,
>
> I'm sorry but I'm not able help you out, but i was wondering how you were able to execute trajectory's with RViz MotionPlanning.
>
> I'm running on Ubuntu 22.04, ROS 2 Humble (Debian [...]

@vetletj The scaled_trajectory_controller had an error in as described in this issue #301, hence i used the joint_trajectory_controller (as you can see in the launch parameters). Other than for this if the launch files are launched in the order I mentioned in the post it should work with the motion planning plugin

vuva, 2022-08-03:
> Hello @gjane5,
>
> I'm sorry but I'm not able help you out, but i was wondering how you were able to execute trajectory's with RViz MotionPlanning.
>
> I'm running on Ubuntu 22.04, ROS 2 Humble (Debian [...]

Did rviz even open for you? Or did it open but there's no robot model in the scene?
I have the same setup with UR5e.

vetletj, 2022-08-04:
I tried with the same controller as you did @gjane5 (joint_trajectory_controller), and still got some move_group_interface error `[rviz2-2] [ERROR] [1659616782.563235559] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached`.

Yes @vuva, RViz opens with the correct robot description and i can plan trajectories, but not execute with motion planning plugin. Guess I'm just gonna delete the workspace and try again, hopefully I'll get another result.

fmauch, 2022-08-15:
@gjane5 Since you access the move_group from a custom script: Is it possible, that you'll have to pass the robot_description_semantic to your custom node, explicitly? I am not familiar with MoveIt! on ROS2 very much, but this sounds like a "ROS2 parameters  live only inside the node's namespace" problem.

7ossamamr, 2022-09-14:
> I tried with the same controller as you did @gjane5 (joint_trajectory_controller), and still got some move_group_interface error `[rviz2-2] [ERROR] [1659616782.563235559] [move_group_interface]: [...]

I had the same problem as mentioned here and using the new ros2 controllers solved it for me. The model is not moving though but the controllers and publishers are working.
https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/issues/471#issuecomment-1237268460

enriLoniterp, 2022-10-05:
Hi everyone, got the same issues of all above!

> @vetletj The scaled_trajectory_controller had an error in as described in this issue #301, hence i used the joint_trajectory_controller (as you can see in the launch parameters). Other than for this [...]

This worked also for me, but still i can't understand how to use MoveGroup on this robot.

skaeringur97, 2022-10-07:
I had similar issues as mentioned above, where I could plan but not execute. I was able to solve it like this:

First, I found out that the scaled_trajectory_controller does not work for simulation, and that joint_trajectory_controller should be used. When  'use_fake_hardware:=true' is set, the controllers should default to joint_trajectory_controller. After adding that to my launch arguments, running 'ros2 control list_controllers' still showed the scaled_trajectory_controller instead of the joint_trajectory_controller. I can only get the joint_trajectory_controller running by adding 'initial_joint_controller:=joint_trajectory_controller' to my command when starting the driver. 

When I then ran the moveit launch file, I still got the same error as @doshininad. 
```
Action client not connected to action server: scaled_joint_trajectory_controller/follow_joint_trajectory
```
After looking in the moveit launch file I saw that 'use_sim_time' needs to be set to true when planning in simulation. Not sure why that isn't connected to the use_fake_hardware boolean.

After adding that to the moveit launch arguments I managed to execute my plans.

This is what I use to run the driver and moveit:
```
ros2 launch ur_robot_driver ur10e.launch.py robot_ip:=aaa.bbb.cc.ddd use_fake_hardware:=true initial_joint_controller:=joint_trajectory_controller
ros2 launch ur_moveit_config ur_moveit.launch.py use_fake_hardware:=true ur_type:=ur10e use_sim_time:=true
```

enriLoniterp, 2022-10-07:
> aunch ur_moveit

yes, that's should work for sure, but if want to use `MoveGroupInterface` it will also publish robotdescription in the correct topic?
In particular i'm refering to this error got by first creator of this issue:
`[ERROR] [1659434560.178463185] [moveit_ur_test]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds. Error: Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0 at line 715 in /home/******/ws_moveit2/src/srdfdom/src/model.cpp [ERROR] [1659434560.193027035] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF [FATAL] [1659434560.193493008] [move_group_interface]: Unable to construct robot model. Please make sure all needed information is on the parameter server. terminate called after throwing an instance of 'std::runtime_error' what(): Unable to construct robot model. Please make sure all needed information is on the parameter server. [ros2run]: Aborted`

enriLoniterp, 2022-10-07:
I've just looked in commits ad i think someone understood and fixeed in code:
if use fake hadrare
change_controllers
if true controllers_yaml[joint_trajectory_controller] = true

not written this way, i was just explaining

enriLoniterp, 2022-10-07:
now i try: https://moveit.picknik.ai/humble/doc/tutorials/your_first_project/your_first_project.html this tutorial changing with ur_manipulator as suggested

enriLoniterp, 2022-10-07:
@skaeringur97 I've done as you said with ur_manipulator in MoveGroupInterface but still not working.. do you have something to show me? if you dont mind

skaeringur97, 2022-10-07:
@enriLoniterp sorry I just saw I have the robot details in [my launch file](https://gist.github.com/skaeringur97/ff87ce4c53ea056e45504b2ba37f3e01). It basically tells the node everything it needs to know about ur10e. You can try something similar for your type :)

enriLoniterp, 2022-10-07:
this launch file it is your personal launch file and you have not modified any MoveIt2 preset?

Anyway i thank you so much for help, i lost a lot of time for this problem and i appreciate so much!!

enriLoniterp, 2022-10-07:
@skaeringur97 
there's something i dont'understand still, if i don't have any end-effector in `ur_descripption` how can i simulate a pick and place application with this driver?

skaeringur97, 2022-10-07:
@enriLoniterp For simulation, you can think of the last link as the "end-effector", and it should be able to "grab" an object, kind of like a magnet. However, if you want to use the actual universal robot for pick and place, I believe you would need some separate code to run the gripper part.

enriLoniterp, 2022-10-07:
yes i think you are right, but iwhere i can find urdf of the official universal robot gripper? if you know

LucaBross, 2022-10-19:
I also have the same problem like @enriLoniterp 
I tried to create a launchfile like @skaeringur97 explained. 

```
import launch
import os
import sys

from launch_ros.actions import Node
from launch.substitutions import PathJoinSubstitution, Command, FindExecutable
from launch_ros.substitutions import FindPackageShare

def get_robot_description():
    joint_limit_params = PathJoinSubstitution(
        [FindPackageShare("ur_description"), "config", "ur3", "joint_limits.yaml"]
    )
    kinematics_params = PathJoinSubstitution(
        [FindPackageShare("ur_description"), "config", "ur3", "default_kinematics.yaml"]
    )
    physical_params = PathJoinSubstitution(
        [FindPackageShare("ur_description"), "config", "ur3", "physical_parameters.yaml"]
    )
    visual_params = PathJoinSubstitution(
        [FindPackageShare("ur_description"), "config", "ur3", "visual_parameters.yaml"]
    )
    robot_description_content = Command(
        [
            PathJoinSubstitution([FindExecutable(name="xacro")]),
            " ",
            PathJoinSubstitution([FindPackageShare("ur_description"), "urdf", "ur.urdf.xacro"]),
            " ",
            "robot_ip:=192.168.1.102",
            " ",
            "joint_limit_params:=",
            joint_limit_params,
            " ",
            "kinematics_params:=",
            kinematics_params,
            " ",
            "physical_params:=",
            physical_params,
            " ",
            "visual_params:=",
            visual_params,
            " ",
           "safety_limits:=",
            "true",
            " ",
            "safety_pos_margin:=",
            "0.15",
            " ",
            "safety_k_position:=",
            "20",
            " ",
            "name:=",
            "ur",
            " ",
            "ur_type:=",
            "ur3",
            " ",
            "prefix:=",
            '""',
            " ",
        ]
    )

    robot_description = {"robot_description": robot_description_content}
    return robot_description

def get_robot_description_semantic():
    # MoveIt Configuration
    robot_description_semantic_content = Command(
        [
            PathJoinSubstitution([FindExecutable(name="xacro")]),
            " ",
            PathJoinSubstitution([FindPackageShare("ur_moveit_config"), "srdf", "ur.srdf.xacro"]),
            " ",
            "name:=",
            # Also ur_type parameter could be used but then the planning group names in yaml
            # configs has to be updated!
            "ur",
            " ",
            "prefix:=",
            '""',
            " ",
        ]
    )
    robot_description_semantic = {
        "robot_description_semantic": robot_description_semantic_content
    }
    return robot_description_semantic
    
def generate_launch_description():
    # generate_common_hybrid_launch_description() returns a list of nodes to launch
    robot_description = get_robot_description()
    robot_description_semantic = get_robot_description_semantic()
    demo_node = Node(
        package="hello_moveit_ur",
        executable="hello_moveit_ur",
        name="hello_moveit_ur",
        output="screen",
        parameters=[
            robot_description,
            robot_description_semantic,
        ],
    )

    return launch.LaunchDescription([demo_node])
```
It is located in ~/workspace/ws_moveit2/src/hello_moveit_ur/launch
I changed the robot from ur10e to ur3. Also i changed the ip adress.
When i start it i get following error. 
```
ros2 launch hello_moveit_ur hello_moveit_ur_launch.py 
[INFO] [launch]: All log files can be found below /home/luca/.ros/log/2022-10-19-15-27-56-985161-Lenovo-V15-G2-ALC-3632
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [hello_moveit_ur-1]: process started with pid [3635]
[hello_moveit_ur-1] [INFO] [1666186077.689468980] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00746732 seconds
[hello_moveit_ur-1] [INFO] [1666186077.689722295] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[hello_moveit_ur-1] [INFO] [1666186077.689736543] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[hello_moveit_ur-1] [WARN] [1666186077.763847117] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[hello_moveit_ur-1] [FATAL] [1666186077.763931555] [move_group_interface]: Group 'ur3e' was not found.
[hello_moveit_ur-1] terminate called after throwing an instance of 'std::runtime_error'
[hello_moveit_ur-1]   what():  Group 'ur3e' was not found.
[ERROR] [hello_moveit_ur-1]: process has died [pid 3635, exit code -6, cmd '/home/luca/ws_moveit2/install/hello_moveit_ur/lib/hello_moveit_ur/hello_moveit_ur --ros-args -r __node:=hello_moveit_ur --params-file /tmp/launch_params_q4ymv7_2 --params-file /tmp/launch_params_b4vw6jtn'].
```
Where is the mistake?

skaeringur97, 2022-10-19:
@LucaBross you changed it to ur3 but somewhere you are calling ur3e, I'm guessing in the hello_moveit_ur node. If you want to use ur3e, you should change that in the launch file. Also, it looks like you are trying to move the group ur3e, but ur3e is not the name of the group. The group name is ur_manipulator and the robot name is ur3e.

LucaBross, 2022-10-19:
Thanks for you help! 

My hello_moveit_file looks like this 

```
#include <memory>

#include <rclcpp/rclcpp.hpp>
#include <moveit/move_group_interface/move_group_interface.h>

int main(int argc, char * argv[])
{
  // Initialize ROS and create the Node
  rclcpp::init(argc, argv);
  auto const node = std::make_shared<rclcpp::Node>(
    "hello_moveit",
    rclcpp::NodeOptions().automatically_declare_parameters_from_overrides(true)
  );

  // Create a ROS logger
  auto const logger = rclcpp::get_logger("hello_moveit");

  // Create the MoveIt MoveGroup Interface
using moveit::planning_interface::MoveGroupInterface;
auto move_group_interface = MoveGroupInterface(node, "ur_manipulator"); //change

// Set a target Pose
auto const target_pose = []{
  geometry_msgs::msg::Pose msg;
  msg.orientation.w = 1.0;
  msg.position.x = 0.28;
  msg.position.y = -0.2;
  msg.position.z = 0.5;
  return msg;
}();
move_group_interface.setPoseTarget(target_pose);

// Create a plan to that target pose
auto const [success, plan] = [&move_group_interface]{
  moveit::planning_interface::MoveGroupInterface::Plan msg;
  auto const ok = static_cast<bool>(move_group_interface.plan(msg));
  return std::make_pair(ok, msg);
}();

// Execute the plan
if(success) {
  move_group_interface.execute(plan);
} else {
  RCLCPP_ERROR(logger, "Planing failed!");
}

  // Shutdown ROS
  rclcpp::shutdown();
  return 0;
}
```
I changed all the ur3 to ur3e in the launch file. 
Also i changed the IP adress to the one from my docker 

After building and sourcing i launched my docker simulation
` docker run --rm -it -p 5900:5900 -p 6080:6080 -v ${HOME}/.ursim/urcaps:/urcaps -v ${HOME}/.ursim/programs:/ursim/programs --name ursim universalrobots/ursim_e-series`

then i started the ur driver 'ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur3e robot_ip:=172.17.0.2 launch_rviz:=false initial_joint_controller:=joint_trajectory_controller' 
When i start my node `ros2 launch hello_moveit_ur hello_moveit_ur_launch.py` i still get the same error. 

I can't find the problem.

LucaBross, 2022-10-19:
@skaeringur97 
I forgot to save the change from "ur3e" to "ur_manipulator" an entire day. 
Now i finally figured it out and it works now.

fmauch, 2023-01-02:
Skimming over this it seems that there's nothing to be done from this repository's perspective, is that correct? I'll go ahead and close this. Feel free to comment or reopen if you disagree.

closed 2023-01-02 by fmauch

SebiGoebel, 2023-10-22:
I had the same problem and fixed it with the launchfile. That worked!
However now i have the Problem that a path is planned, but the robot does not execute the plan.
Does anyone know how that can be fixed?

ros2 launch moving_ur3e_ros2 move_ur3e.launch.py 
[INFO] [launch]: All log files can be found below /home/sebi/.ros/log/2023-10-22-17-28-21-484334-sebi-Lenovo-Yoga-C940-15IRH-93108
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [moving_ur3e-1]: process started with pid [93111]
[moving_ur3e-1] [INFO] [1697988502.045885323] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00787052 seconds
[moving_ur3e-1] [INFO] [1697988502.046745847] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[moving_ur3e-1] [INFO] [1697988502.046789538] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[moving_ur3e-1] [WARN] [1697988502.095280582] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[moving_ur3e-1] [INFO] [1697988502.113102855] [move_group_interface]: Ready to take commands for planning group ur_manipulator.
[moving_ur3e-1] [INFO] [1697988502.113303708] [move_group_interface]: MoveGroup action client/server ready
[moving_ur3e-1] [INFO] [1697988503.124325586] [move_group_interface]: Planning request accepted
[moving_ur3e-1] [INFO] [1697988506.369474712] [move_group_interface]: Planning request complete!
[moving_ur3e-1] [INFO] [1697988506.370627303] [move_group_interface]: time taken to generate plan: 0.240219 seconds
[moving_ur3e-1] [INFO] [1697988506.372169621] [move_group_interface]: Execute request accepted
[moving_ur3e-1] [INFO] [1697988506.376978801] [move_group_interface]: Execute request aborted
[moving_ur3e-1] [ERROR] [1697988506.377995400] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
[INFO] [moving_ur3e-1]: process has finished cleanly [pid 93111]

skaeringur97, 2023-10-22:
@SebiGoebel Usually when I had that problem, the movegroup node did not start properly. Check your parameters in your launch file and see if everything is correct. Maybe you need to set use_fake_hardware to true or something like that

ShahhhVihaan, 2023-11-27:
I used the launch file that's written above and I keep getting the Execute request aborted issue repeatedly like @SebiGoebel . I am working on a real robot so I cannot use fake_hardware. I'd appreciate some help.

SebiGoebel, 2023-11-30:
i tried the use_fake_hardware:=true and it worked, however the robot does not move. When I move the interactive marker in rviz a motion is planned, but when i try to execute this plan the robot does not move.

yctros2, 2024-03-05:
> Hello, I am trying to use the moveit_config files from the ur repository together with the ur driver. I can launch the driver and the moveit package with the commands:
>
> `ros2 launch ur_robot_driver [...]

hello，could you solve it?i meet the same issue。

Divyam-uga, 2024-07-02:
I also had the same problem as @enriLoniterp. 
I tried to create a launch file as @skaeringur97 explained.

While I am able to initialize the “MoveitPy” node, I am encountering an error that mentions there is no planning pipeline available for the name ' '. This seems odd to me since I have already specified the pipeline in the launch file. 

I suspect there might be an error in the way I am passing the parameters. Any help on resolving this issue would be greatly appreciated. I am a beginner and currently learning ROS 2 and Moveit 2. 

Error:
[image]

Python_api_launch_file:
[python_ur_control.launch.zip](https://github.com/user-attachments/files/16069163/python_ur_control.launch.zip)

MoveitPy script:
[ur1_moveitpy.zip](https://github.com/user-attachments/files/16069212/ur1_moveitpy.zip)

yctros2, 2024-07-02:
> I also had the same problem as @enriLoniterp. I tried to create a launch file as @skaeringur97 explained.
> 
> While I am able to initialize the “MoveitPy” node, I am encountering an error that mentions there is no planning pipeline available for the name ' '. This seems odd to me since I have already specified the pipeline in the launch file.
> 
> I suspect there might be an error in the way I am passing the parameters. Any help on resolving this issue would be greatly appreciated. I am a beginner and currently learning ROS 2 and Moveit 2.
> 
> Error: ![Screenshot from 2024-07-02 09-31-24](https://private-user-images.githubusercontent.com/167348850/345051565-54a7947e-b4f5-40cc-ad89-b2aee702ee4c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk5MjkzNzEsIm5iZiI6MTcxOTkyOTA3MSwicGF0aCI6Ii8xNjczNDg4NTAvMzQ1MDUxNTY1LTU0YTc5NDdlLWI0ZjUtNDBjYy1hZDg5LWIyYWVlNzAyZWU0Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNzAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDcwMlQxNDA0MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04YjY0YWI4ODllYmU0N2QyNjVmZjUzMDZkNzkzYmIwZTZjOTc1YTNkZGNlZjgwZWM2NGY4ZmI3ODM4ODQ3NGEwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.syt1_cWnTXF3iIWF7GNIaQVISOJK-MrXrk3H_Ov4l-M)
> 
> Python_api_launch_file: [python_ur_control.launch.zip](https://github.com/user-attachments/files/16069163/python_ur_control.launch.zip)
> 
> MoveitPy script: [ur1_moveitpy.zip](https://github.com/user-attachments/files/16069212/ur1_moveitpy.zip)

Please check the working namespace issue.

yctros2, 2024-07-02:
I haven't been able to find a good solution to this problem yet, and I was able to fix it by re-configuring it with the moveit configuration helper!