UniversalRobots/Universal_Robots_ROS2_Driver #462: Inactive forward_position_controller

vuva, 2022-08-03:
Hi all,
I am trying to run URDriver for UR5e in this environment:
OS: Ubuntu Jammy 22.04
ROS2 Humble

After running ur_driver:
`$ ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=201.0.0.101  launch_rviz:=false`
(log output at the end of the post)

the forward_position_controller didn't seem to be activated. 
```
$ ros2 control list_controllers
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active
forward_position_controller[position_controllers/JointGroupPositionController] inactive
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] active
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
io_and_status_controller[ur_controllers/GPIOController] active

```
I tried to manually activate with:
```
$ ros2 control switch_controllers --activate forward_position_controller
Successfully switched controllers
```
also didn't work despite the output.

_scaled_joint_trajectory_controller_ worked ok, since I could run _test_scaled_joint_trajectory_controller.launch.py_ successfully.

Does anyone encounter the same issue?

Cheers,

```
[ur_ros2_control_node-1] [INFO] [1659536731.072210141] [URPositionHardwareInterface]: Calibration checked successfully.
[ur_ros2_control_node-1] [INFO] [1659536731.073060047] [URPositionHardwareInterface]: System successfully started!
[ur_ros2_control_node-1] [INFO] [1659536731.073084140] [resource_manager]: Successful 'activate' of hardware 'ur5e'
[controller_stopper_node-3] [INFO] [1659536731.078791739] [Controller stopper]: Service available
[controller_stopper_node-3] [INFO] [1659536731.078826066] [Controller stopper]: Waiting for list controllers service to come up on controller_manager/list_controllers
[controller_stopper_node-3] [INFO] [1659536731.078845317] [Controller stopper]: Service available
[ur_ros2_control_node-1] [INFO] [1659536731.156641595] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[spawner-7] [INFO] [1659536731.180233000] [spawner_speed_scaling_state_broadcaster]: Loaded speed_scaling_state_broadcaster
[ur_ros2_control_node-1] [INFO] [1659536731.183276906] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[ur_ros2_control_node-1] [INFO] [1659536731.183382671] [speed_scaling_state_broadcaster]: Publisher rate set to : 100.0 Hz
[ur_ros2_control_node-1] [INFO] [1659536731.186001104] [controller_manager]: Loading controller 'io_and_status_controller'
[spawner-6] [INFO] [1659536731.201942921] [spawner_io_and_status_controller]: Loaded io_and_status_controller
[ur_ros2_control_node-1] [INFO] [1659536731.205598452] [controller_manager]: Loading controller 'scaled_joint_trajectory_controller'
[spawner-7] [INFO] [1659536731.213587091] [spawner_speed_scaling_state_broadcaster]: Configured and activated speed_scaling_state_broadcaster
[ur_ros2_control_node-1] [INFO] [1659536731.228731139] [joint_trajectory_controller.interpolation_methods]: No interpolation method parameter was given. Using the default, VARIABLE_DEGREE_SPLINE.
[ur_ros2_control_node-1] [INFO] [1659536731.233990241] [controller_manager]: Loading controller 'forward_position_controller'
[spawner-10] [INFO] [1659536731.237476811] [spawner_scaled_joint_trajectory_controller]: Loaded scaled_joint_trajectory_controller
[ur_ros2_control_node-1] [INFO] [1659536731.252040315] [controller_manager]: Configuring controller 'scaled_joint_trajectory_controller'
[ur_ros2_control_node-1] [INFO] [1659536731.252145782] [scaled_joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[spawner-9] [INFO] [1659536731.252536269] [spawner_forward_position_controller]: Loaded forward_position_controller
[ur_ros2_control_node-1] [INFO] [1659536731.254487136] [scaled_joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ur_ros2_control_node-1] [INFO] [1659536731.256493275] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[ur_ros2_control_node-1] [INFO] [1659536731.260225364] [controller_manager]: Configuring controller 'forward_position_controller'
[ur_ros2_control_node-1] [INFO] [1659536731.260768832] [forward_position_controller]: configure successful
[ur_ros2_control_node-1] [INFO] [1659536731.285673283] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[spawner-10] [INFO] [1659536731.287782085] [spawner_scaled_joint_trajectory_controller]: Configured and activated scaled_joint_trajectory_controller
[ur_ros2_control_node-1] [INFO] [1659536731.304675193] [controller_manager]: Configuring controller 'io_and_status_controller'
[spawner-8] [INFO] [1659536731.305431234] [spawner_force_torque_sensor_broadcaster]: Loaded force_torque_sensor_broadcaster
[ur_ros2_control_node-1] [INFO] [1659536731.323638896] [controller_manager]: Loading controller 'joint_state_broadcaster'
[ur_ros2_control_node-1] [INFO] [1659536731.334189486] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'
[spawner-5] [INFO] [1659536731.335786806] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ur_ros2_control_node-1] [INFO] [1659536731.351484928] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ur_ros2_control_node-1] [INFO] [1659536731.351550249] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-6] [INFO] [1659536731.352195919] [spawner_io_and_status_controller]: Configured and activated io_and_status_controller
[INFO] [spawner-7]: process has finished cleanly [pid 150468]
[spawner-8] [INFO] [1659536731.365945762] [spawner_force_torque_sensor_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[spawner-5] [INFO] [1659536731.373283495] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster

```

fmauch, 2022-09-13:
In your switch command I missed deactivating the scaled JTC, as this would be conflicting with the forwarding controller.

Indeed, when trying the switch command the command says "Successfully switched controllers", but the driver's output clearly states:

"Resource conflict for controller 'forward_position_controller'. Command interface 'shoulder_pan_joint/position' is already claimed." (Which is what I would expect). So, the positive output from the switch command seems to be a controller_manager issue, but if you use the command

```
ros2 control switch_controllers --activate forward_position_controller --deactivate scaled_joint_trajectory_controller
```

The switch should succeed and the forward controller should be running:

```
joint_state_broadcaster[joint_state_broadcaster/JointStateBroadcaster] active
scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] inactive
force_torque_sensor_broadcaster[force_torque_sensor_broadcaster/ForceTorqueSensorBroadcaster] active
speed_scaling_state_broadcaster[ur_controllers/SpeedScalingStateBroadcaster] active
io_and_status_controller[ur_controllers/GPIOController] active
forward_position_controller[position_controllers/JointGroupPositionController] active
```

Can you confirm this?

vuva, 2022-09-26:
> In your switch command I missed deactivating the scaled JTC, as this would be conflicting with the forwarding controller.
>
> Indeed, when trying the switch command the command says "Successfully [...]

Thank you very much. I confirm that your suggestion worked. Please feel free to close this issue.

fmauch, 2022-09-26:
Glad I could help :-)

closed 2022-09-26 by fmauch

Daan0103, 2024-01-19:
I got an problem with this:
I can't activate or deactivate it because when I run ros2 control list_controllers this is the line from the scaled joint trajectory controller:

`scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] unconfigured`

Tsuki0129, 2025-05-23:
> 我遇到了一个问题： 我无法激活或停用它，因为当我运行 ros2 control list_controllers 时，这是来自缩放关节轨迹控制器的行：
> 
> `scaled_joint_trajectory_controller[ur_controllers/ScaledJointTrajectoryController] unconfigured`
u could use this command like ros2 control set_state_controller <ur_controller> inactive  
to configure the controller