UniversalRobots/Universal_Robots_ROS2_Driver #465: Humble - Not able to execute trajectories with ur_moveit.launch.py

vetletj, 2022-08-12:
### Description
I'm not able to execute trajectories with MoveIt MotionPlanning. 

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary build (Debian)
* Desktop install

### Steps to reproduce
Build [Universal Robots Driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver) from source (main branch), source workspace, and run ros2 `launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true` in another shell.

### Expected behaviour
Be able to execute trajectories with MoveIt MotionPlanning plugin.  

### Actual behaviour
[image]

Also tried to launch the controller first  `ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false` and then ur_movit `launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true` then I can change to the right **Fixed Frame** (base_link) but RViz just crashes when I'm trying. Didn't include the console output from this to reduce confusion. 

### Console output
```
tht@tht-stationary02:~$ cd workspace/ros_ur_driver/
tht@tht-stationary02:~/workspace/ros_ur_driver$ source install/setup.bash 
tht@tht-stationary02:~/workspace/ros_ur_driver$ ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true
[INFO] [launch]: All log files can be found below /home/tht/.ros/log/2022-08-12-16-15-59-695886-tht-stationary02-533761
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [533768]
[INFO] [rviz2-2]: process started with pid [533770]
[INFO] [servo_node_main-3]: process started with pid [533772]
[servo_node_main-3] [WARN] [1660313760.141345565] [moveit_servo.servo_node]: Intra-process communication is disabled, consider enabling it by adding: 
[servo_node_main-3] extra_arguments=[{'use_intra_process_comms' : True}]
[servo_node_main-3] to the Servo composable node in the launch file
[move_group-1] [WARN] [1660313760.141950569] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[servo_node_main-3] [INFO] [1660313760.143213470] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00114115 seconds
[servo_node_main-3] [INFO] [1660313760.143228347] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[servo_node_main-3] [INFO] [1660313760.143231928] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1660313760.143726073] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.00165654 seconds
[move_group-1] [INFO] [1660313760.143745923] [moveit_robot_model.robot_model]: Loading robot model 'ur'...
[move_group-1] [INFO] [1660313760.143752339] [moveit_robot_model.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[servo_node_main-3] [WARN] [1660313760.148783627] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
[move_group-1] Link base_link had 2 children
[move_group-1] Link base had 0 children
[move_group-1] Link base_link_inertia had 1 children
[move_group-1] Link shoulder_link had 1 children
[move_group-1] Link upper_arm_link had 1 children
[move_group-1] Link forearm_link had 1 children
[move_group-1] Link wrist_1_link had 1 children
[move_group-1] Link wrist_2_link had 1 children
[move_group-1] Link wrist_3_link had 2 children
[move_group-1] Link flange had 1 children
[move_group-1] Link tool0 had 0 children
[move_group-1] Link ft_frame had 0 children
[servo_node_main-3] [INFO] [1660313760.165635592] [moveit_ros.current_state_monitor]: Listening to joint states on topic '/joint_states'
[servo_node_main-3] [INFO] [1660313760.167580546] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[servo_node_main-3] [INFO] [1660313760.167593619] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[servo_node_main-3] [INFO] [1660313760.168129237] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[servo_node_main-3] [INFO] [1660313760.168239767] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on '/servo_node/publish_planning_scene'
[move_group-1] [INFO] [1660313760.168902593] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1660313760.168965841] [moveit.ros_planning_interface.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-1] [INFO] [1660313760.169177025] [moveit_ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-1] [INFO] [1660313760.169293676] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1660313760.169299925] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1660313760.169371171] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1660313760.169375069] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1660313760.169460699] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1660313760.1
![Screenshot from 2022-08-12 16-36-24](https://user-images.githubusercontent.com/67962681/184378199-e06a81fa-e29f-4ef9-b801-47ca58f471a4.png)
69517317] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1660313760.169632046] [moveit.ros.occupancy_map_monitor.middleware_handle]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[... 115 lines cut ...]
[rviz2-2] [INFO] [1660314962.660560934] [move_group_interface]: Planning request accepted
[move_group-1] [INFO] [1660314963.360521024] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[move_group-1] Check clock synchronization if your are running ROS across multiple machines!
[move_group-1] [WARN] [1660314963.360604046] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to fetch current robot state.
[move_group-1] [INFO] [1660314963.360697511] [moveit_move_group_default_capabilities.move_action_capability]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[move_group-1] [INFO] [1660314963.360716795] [moveit_move_group_capabilities_base.move_group_capability]: Using planning pipeline 'move_group'
[move_group-1] [INFO] [1660314963.361568990] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[move_group-1] [INFO] [1660314963.409344368] [moveit_move_group_default_capabilities.move_action_capability]: Motion plan was computed successfully.
[rviz2-2] [INFO] [1660314963.663253245] [move_group_interface]: Planning request complete!
[rviz2-2] [INFO] [1660314963.763512921] [move_group_interface]: time taken to generate plan: 0.0394852 seconds
[move_group-1] [INFO] [1660314965.771082528] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-1] [INFO] [1660314965.771326162] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-1] [INFO] [1660314965.771394554] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1660314965.771440276] [moveit.plugins.moveit_simple_controller_manager]: Returned 2 controllers in list
[move_group-1] [INFO] [1660314965.771697337] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[rviz2-2] [INFO] [1660314966.071657103] [move_group_interface]: Execute request accepted
[move_group-1] [INFO] [1660314966.771921132] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[... cut ...]
[move_group-1] [WARN] [1660314966.772000531] [moveit_ros.trajectory_execution_manager]: Failed to validate trajectory: couldn't receive full current joint state within 1s
[move_group-1] [INFO] [1660314966.772072271] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[rviz2-2] [INFO] [1660314967.074628342] [move_group_interface]: Execute request aborted
[rviz2-2] [ERROR] [1660314967.174900879] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
```

Thanks for taking the time to consider this post.

adverley, 2022-08-13:
If you are using the main branch of this repo and testing on Humble (possibly also on other ROS 2 versions), then the default moveit launch file does not work with fake hardware. The controller that is used by default is the `scaled_joint_trajectory_controller `which does not work with `fake_hardware`. Check whether your code works by checking out #464. As alternative, you can quickly change the `default` fields in the controllers config yaml file.

vetletj, 2022-08-15:
Thanks for the input @adverley ! 

Tried both approaches, with same result as above.

fmauch, 2022-08-15:
You definitively have to run a robot driver instance, as well as the moveit part, as otherwise there will not be anything to execute. With #464 getting merged as soon as the pipeline finished successfully, you can pass `use_fake_hardware:=true` to the moveit launchfile, as well, which should get you up and running. If not, I suggest you post the output of both shells (driver and moveit launch, including the actual launched commands) and we continue from there.

vetletj, 2022-08-16:
Thanks you for clearing that up @fmauch, for me that wasn't clear from the [Universal Robots ROS2 Driver (README)](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/README.md), but that's might also be cause I'm still pretty new to ROS2 in general.

Seems like everything work fine now, many thank to both of you!

closed 2022-08-16 by vetletj