UniversalRobots/Universal_Robots_ROS2_Driver #468: Robot state publisher problem reserving CacheChange in reader error.

urmahp, 2022-08-22:
When using this Driver with humble or rolling the error message `[robot_state_publisher-4] 2022-08-22 07:38:08.041 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg` is spammed to the log. It happens both when the driver is installed as a binary package and when it is cloned. 

I am not sure that it actually has something to do with this driver since the error comes from the robot state publisher, but it makes it difficult to use the terminal output when launching the driver since you will only see that specific message. It is still possible to use the driver to control the robot, so a workaround could be to suppress the output of the robot state publisher in the launch file. 

The command used to launch the driver can be seen below, together with the terminal output. 

```bash
root@ur-li-95qgtn2:/workspace/ros_ws_ur_driver# ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5e robot_ip:=127.0.0.1
[INFO] [launch]: All log files can be found below /root/.ros/log/2022-08-22-07-38-06-654457-ur-li-95qgtn2.unirobotts.local-167
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [dashboard_client-2]: process started with pid [172]
[INFO] [controller_stopper_node-3]: process started with pid [174]
[INFO] [ur_ros2_control_node-1]: process started with pid [170]
[INFO] [robot_state_publisher-4]: process started with pid [176]
[INFO] [rviz2-5]: process started with pid [178]
[INFO] [spawner-6]: process started with pid [180]
[INFO] [spawner-7]: process started with pid [182]
[INFO] [spawner-8]: process started with pid [200]
[INFO] [spawner-9]: process started with pid [213]
[INFO] [spawner-10]: process started with pid [225]
[INFO] [spawner-11]: process started with pid [231]
[robot_state_publisher-4] Link base_link had 2 children
[robot_state_publisher-4] Link base had 0 children
[robot_state_publisher-4] Link base_link_inertia had 1 children
[robot_state_publisher-4] Link shoulder_link had 1 children
[robot_state_publisher-4] Link upper_arm_link had 1 children
[robot_state_publisher-4] Link forearm_link had 1 children
[robot_state_publisher-4] Link wrist_1_link had 1 children
[robot_state_publisher-4] Link wrist_2_link had 1 children
[robot_state_publisher-4] Link wrist_3_link had 2 children
[robot_state_publisher-4] Link flange had 1 children
[robot_state_publisher-4] Link tool0 had 0 children
[robot_state_publisher-4] Link ft_frame had 0 children
[robot_state_publisher-4] [INFO] [1661153887.823766921] [robot_state_publisher]: got segment base
[robot_state_publisher-4] [INFO] [1661153887.825563350] [robot_state_publisher]: got segment base_link
[robot_state_publisher-4] [INFO] [1661153887.825603703] [robot_state_publisher]: got segment base_link_inertia
[robot_state_publisher-4] [INFO] [1661153887.825613798] [robot_state_publisher]: got segment flange
[robot_state_publisher-4] [INFO] [1661153887.825647117] [robot_state_publisher]: got segment forearm_link
[robot_state_publisher-4] [INFO] [1661153887.825658007] [robot_state_publisher]: got segment ft_frame
[robot_state_publisher-4] [INFO] [1661153887.825666780] [robot_state_publisher]: got segment shoulder_link
[robot_state_publisher-4] [INFO] [1661153887.825674998] [robot_state_publisher]: got segment tool0
[robot_state_publisher-4] [INFO] [1661153887.825683111] [robot_state_publisher]: got segment upper_arm_link
[robot_state_publisher-4] [INFO] [1661153887.825691317] [robot_state_publisher]: got segment world
[robot_state_publisher-4] [INFO] [1661153887.825699285] [robot_state_publisher]: got segment wrist_1_link
[robot_state_publisher-4] [INFO] [1661153887.825707351] [robot_state_publisher]: got segment wrist_2_link
[robot_state_publisher-4] [INFO] [1661153887.825715482] [robot_state_publisher]: got segment wrist_3_link
[rviz2-5] qt.qpa.xcb: could not connect to display 
[rviz2-5] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-5] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-5] 
[rviz2-5] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-5] 
[controller_stopper_node-3] [INFO] [1661153887.819316286] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
[ur_ros2_control_node-1] [INFO] [1661153887.865541670] [resource_manager]: Loading hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1661153887.872533455] [resource_manager]: Initialize hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1661153887.872627822] [resource_manager]: Successful initialization of hardware 'ur5e'
[ur_ros2_control_node-1] [INFO] [1661153887.872837133] [resource_manager]: 'configure' hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1661153887.872850662] [resource_manager]: Successful 'configure' of hardware 'ur5e'
[ur_ros2_control_node-1] [INFO] [1661153887.872890803] [resource_manager]: 'activate' hardware 'ur5e' 
[ur_ros2_control_node-1] [INFO] [1661153887.872903838] [URPositionHardwareInterface]: Starting ...please wait...
[ur_ros2_control_node-1] [INFO] [1661153887.872924300] [URPositionHardwareInterface]: Initializing driver...
[ur_ros2_control_node-1] Warning: No realtime capabilities found. Consider using a realtime system for better performance
[ur_ros2_control_node-1]          at line 396 in ./include/ur_client_library/comm/pipeline.h
[ur_ros2_control_node-1] [INFO] [1661153887.897889933] [URPositionHardwareInterface]: Calibration checksum: 'calib_12788084448423163542'.
[ur_ros2_control_node-1] Warning: No realtime capabilities found. Consider using a realtime system for better performance
[ur_ros2_control_node-1]          at line 396 in ./include/ur_client_library/comm/pipeline.h
[robot_state_publisher-4] 2022-08-22 07:38:08.041 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.041 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.045 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.045 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[... 4 lines cut ...]
[robot_state_publisher-4] 2022-08-22 07:38:08.046 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.048 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.049 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.049 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.049 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.075 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.075 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.083 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[... cut ...]
[robot_state_publisher-4] 2022-08-22 07:38:08.086 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.086 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg
[robot_state_publisher-4] 2022-08-22 07:38:08.086 [RTPS_MSG_IN Error] (ID:139735538259520) Problem reserving CacheChange in reader: 01.0f.ad.4e.b0.00.38.a5.01.00.00.00|0.0.15.4 -> Function processDataMsg

```

gavanderhoorn, 2022-08-22:
Could be https://github.com/eProsima/Fast-DDS/issues/2799.

fmauch, 2022-09-16:
@urmahp According to [this post](https://github.com/eProsima/Fast-DDS/issues/2799#issuecomment-1238433376) this should be fixed by now. Can you confirm that?

urmahp, 2022-09-19:
Yes when building humble from source the error has been fixed. However, when not building from source the error still seems to present, but I guess that this will only be until the next humble sync?

fmauch, 2022-09-21:
I would not expect that building the driver from source should have an effect on that.

closed 2022-09-26 by urmahp

fmauch, 2022-09-26:
As addition: What @urmahp meant that he was building **all** humble packages from source, not only our driver. Therefore, we can assume that this has been fixed and hence close this.