UniversalRobots/Universal_Robots_ROS2_Driver #471: Warning when starting up a controller and model not moving

7ossamamr, 2022-09-02:
I am currently using ROS2 Humble and I am trying to use this package. Everything seems to be working just fine except for the controllers.

Whenever I startup a controller, whether it is test_joint_trajectory_controller or test_scaled_joint_trajectory_controller I end up with the following warnings and the robot not moving.

$ ros2 launch ur_robot_driver test_joint_trajectory_controller.launch.py
[INFO] [launch]: All log files can be found below /home/hossam/.ros/log/2022-09-02-16-06-44-571462-hossam-83472
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [publisher_joint_trajectory_controller-1]: process started with pid [83474]
[publisher_joint_trajectory_controller-1] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'joints', declaring a parameter only providing its name is deprecated. You have to either:
[publisher_joint_trajectory_controller-1] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[publisher_joint_trajectory_controller-1] 	- Pass a name and a parameter type.
[publisher_joint_trajectory_controller-1] 	- Pass a name and a descriptor with `dynamic_typing=True
[publisher_joint_trajectory_controller-1]   warnings.warn(
[publisher_joint_trajectory_controller-1] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'starting_point_limits', declaring a parameter only providing its name is deprecated. You have to either:
[publisher_joint_trajectory_controller-1] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[publisher_joint_trajectory_controller-1] 	- Pass a name and a parameter type.
[publisher_joint_trajectory_controller-1] 	- Pass a name and a descriptor with `dynamic_typing=True
[publisher_joint_trajectory_controller-1]   warnings.warn(
[publisher_joint_trajectory_controller-1] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'pos1', declaring a parameter only providing its name is deprecated. You have to either:
[publisher_joint_trajectory_controller-1] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[publisher_joint_trajectory_controller-1] 	- Pass a name and a parameter type.
[publisher_joint_trajectory_controller-1] 	- Pass a name and a descriptor with `dynamic_typing=True
[publisher_joint_trajectory_controller-1]   warnings.warn(
[publisher_joint_trajectory_controller-1] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'pos2', declaring a parameter only providing its name is deprecated. You have to either:
[publisher_joint_trajectory_controller-1] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[publisher_joint_trajectory_controller-1] 	- Pass a name and a parameter type.
[publisher_joint_trajectory_controller-1] 	- Pass a name and a descriptor with `dynamic_typing=True
[publisher_joint_trajectory_controller-1]   warnings.warn(
[publisher_joint_trajectory_controller-1] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'pos3', declaring a parameter only providing its name is deprecated. You have to either:
[publisher_joint_trajectory_controller-1] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[publisher_joint_trajectory_controller-1] 	- Pass a name and a parameter type.
[publisher_joint_trajectory_controller-1] 	- Pass a name and a descriptor with `dynamic_typing=True
[publisher_joint_trajectory_controller-1]   warnings.warn(
[publisher_joint_trajectory_controller-1] /opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/node.py:441: UserWarning: when declaring parameter named 'pos4', declaring a parameter only providing its name is deprecated. You have to either:
[publisher_joint_trajectory_controller-1] 	- Pass a name and a default value different to "PARAMETER NOT SET" (and optionally a descriptor).
[publisher_joint_trajectory_controller-1] 	- Pass a name and a parameter type.
[publisher_joint_trajectory_controller-1] 	- Pass a name and a descriptor with `dynamic_typing=True
[publisher_joint_trajectory_controller-1]   warnings.warn(
[publisher_joint_trajectory_controller-1] [INFO] [1662127605.094305894] [publisher_joint_trajectory_controller]: Publishing 4 goals on topic "/joint_trajectory_controller/joint_trajectory" every 6 s

I have been reading about the rclpy API but it seems like I am missing something.
I don't think it has to do with the YAML file and I tried adding allow_undeclared_parameters=True, automatically_declare_parameters_from_overrides=True to the Node in the launch file but I ended up with a different error so I guess this was not a potential solution.

destogl, 2022-09-05:
Hi @7ossamamr, can you please provide more information about your setup? This warning should not change anything in the behavior. If you want, please check a new version of ros2_controllers, which has a [new PR with updates on test nodes](https://github.com/ros-controls/ros2_controllers/pull/389).

Are you running the real robot, ursim, Gazebo simulation or mock hardware simulation?

7ossamamr, 2022-09-06:
Hello @destogl, thank you for suggesting checking the new ros2_controllers. Now I am using the new version along with ros2_controls as well and the controller seems to be working fine. It is publishing the target joint states which was not the case before, it used to be stuck after the last info message as I posted earlier. However, the model is not moving in any of the cases, neither in the mock hardware nor in the ursim and of course I still haven't tried running the real robot before getting a running simulation.

Maybe I will state some points that might be useful.

- Model is respawning properly in RViz
- Model is connected properly to the ursim, as any joint configuration in the ursim is readable in ROS and reflected in the RViz model
- Controllers are working fine. I checked the published topics and now they are normally updated with the new joint states rather than being blank as before.
- Controllers are able to detect the robot's current state. As they would return an error if they were started when the robot is in a different position than what is defined in the joint limits yaml, in this case it is the default homing position.

But still nothing is moving. I will try to clone the UR repo and rebuild it again maybe it works, I will update you with whatever turns to be.

closed 2022-09-15 by 7ossamamr