UniversalRobots/Universal_Robots_ROS2_Driver #475: Debugging jerky `ur_moveit_config` visualization?

danzimmerman, 2022-09-13:
I'm getting very choppy robot visualization in RViz if I launch the visualization with `ur_moveit_config` 

```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e robot_ip:=192.168.56.101
```

Here's a video (I think this should be visible to anyone with the link, let me know if it isn't)

https://vimeo.com/749030924/877bca2373

In the video, I'm jogging the robot from the URSim GUI.

### Details

* If I launch RViz along with `ur_robot_driver` using the `launch_rviz:=true` parameter, the motion when jogging from URSim is nice and smooth. 

* There's nothing wrong with the basic RViz framerate, it's only the robot updates that are jerky. Window manipulation and smooth visualization of other topics is working fine, including dragging the orange MoveIt goal state around. (See https://vimeo.com/749037979/b1026b094e)
* I'm getting `/joint_states` at about 482Hz average according to `ros2 topic hz`
* I'm running <10% load on all 8 cores, using about 6.5GB of 32GB RAM.
* I don't have motion problems on a real robot or URSim as far as I can tell, just this choppy visualization.
* I've observed choppy visualization connected to the real robot as well.
* URSim is running in Docker. It's been working well in general and I don't think it's lagging so badly. The VNC visualization is not as high-quality as RViz typically is, but it doesn't seem as bad as these videos show. I'll put it on my to-do list to capture `/joint_states` to a rosbag and make sure it's smooth there. 

### Package Info
* UR ROS2 Driver: Forked, but up-to-date with `main` at `#f580ae7`. I've got a few changes to get the driver to build locally against Robostack dependencies, comparison can be seen [here](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/compare/main...Polarworks:Universal_Robots_ROS2_Driver:dz/rs-humble-dev-latest).
* MoveIt `2.5.3` 
* `ros2_control` `2.13`

I'm using a mix of from-source and binary [Robostack](https://robostack.github.io/) packages and running Humble on Ubuntu Focal, so I don't really expect a neat solution, but any ideas for debugging would be appreciated.

Anyone else having this issue with recent releases?

fmauch, 2022-09-21:
To my understanding this is rather normal in MoveIt's visualization, at least I haven't seen this differently in other MoveIt! projects. It probably is a parameter somewhere in the MoveIt! config, but I'd like to keep the MoveIt! config as close as possible to what would be generated by the setup assistant without many modifications.

If this is the case, I'd consider this as an invalid issue, as it is merely a configuration preference different from the default behavior.

danzimmerman, 2022-09-21:
> To my understanding this is rather normal in MoveIt's visualization, at least I haven't seen this differently in other MoveIt! projects. It probably is a parameter somewhere in the MoveIt! config [...]

Thanks, appreciate the response!

I will look at using a RobotModel for visualization instead for what I'm doing at the moment. 

It seems likely that I'm misremembering that the MoveIt update rate was better at some point. Might be conflating it with some other visualization, probably `launch_rviz:=true` from the driver.

closed 2022-09-21 by danzimmerman