UniversalRobots/Universal_Robots_ROS2_Driver #477: access robot_description from outside the launch file

meikse, 2022-09-20:
Hello,

I do some simulations in rviz2 with moveit2 and want to use the provided UR models in this repo.
I try to build my own launch file and thus want to use the provided launch `ur_moveit_config/launch/ur_moveit.launch.py` via **IncludeLaunchDescription** suggested in [here](http://docs.ros.org/en/galactic/Tutorials/Intermediate/Launch/Using-ROS2-Launch-For-Large-Projects.html)

Unfortunately, the robot description is not accessible. 

It would be nice if one could easily extract the robot_description information from the files launch provided in this repo, but since they are nested within the `launch_setup` function and are not returned from it, one needs to configure and build all single nodes by themself, at least how I understood.

fmauch, 2022-09-21:
Are you aware, that the robot description is not part of this repository?

Currently, there exists no launchfile to simply load the description, but you could have a look at a more simple launchfile such as https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/ros2/launch/view_ur.launch.py.

meikse, 2022-09-22:
True. I forgot that the description are coming with the debian packages (at least for galactic, that I use).
Nevertheless, the description files are instantiated with more than 2 lines of code, which seems to me a bit overwhelming if I just want to use the standard launch setup.

My solution at the moment is to copy/paste the necessary launch_setup lines for the respective descriptions into my own launch files. This works but feels odd.
Is this the way to go or do I miss something important? :thinking: 

(besides, if this is the wrong place for such questions I will redirect my questions to answers.ros.org)

fmauch, 2022-09-22:
Since you've created the question here already, we can also discuss it here, as it is definitively affecting the whole ecosystem.

> My solution at the moment is to copy/paste the necessary launch_setup lines for the respective descriptions into my own launch files. This works but feels odd.

Yes, I agree. I think the cleanest solution would be to load the description in a separate launchfile to `ur_description` that can be easily included.  I've played around with that in the past, but I am still not that intuitively familiar with the ROS2 launch system that I managed to finish that and then I forgot about that a little... If you want to give that a shot, a PR on the description repo would be very much appreciated :-)

VinDp, 2024-05-29:
Hey,
nowadays the robot description is propagated via topic and that solves this issue :)

closed 2024-05-29 by VinDp