UniversalRobots/Universal_Robots_ROS2_Driver #481: urdf

ddliu365, 2022-09-22:
Hi, folks,

is there a way to parse the robot description xacro format in current package to urdf?

I would like to use urdf to do some computations with other software.

Thanks.

Best,

fmauch, 2022-09-23:
As with any xacro file `ros2 run xacro xacro <file.xacro> > file.urdf` will do the trick. As the description from `ur_description` has a couple of required parameters, we'll have to define them, as well:

```
ros2 run xacro xacro $(ros2 pkg prefix ur_description)/share/ur_description/urdf/ur.urdf.xacro name:=ur ur_type:=ur5e > ur5e.urdf
```

I hope, this helps.

ddliu365, 2022-09-23:
Thanks.

I tried under workspace and found this error:

```
dongdong@thinkstation:~/ros2_ws$ ros2 run xacro xacro $(ros2 pkg prefix ur_description)/share/ur_description/urdf/ur.urdf.xacro name:=ur ur_type:=ur5e > ur5e.urdf
[Errno 21] Is a directory: '/home/dongdong/ros2_ws/install/ur_description/share/ur_description/urdf/' 
when evaluating expression 'load_yaml(joint_limits_parameters_file)' 
when evaluating expression 'config_joint_limit_parameters['joint_limits']' 
when evaluating expression 'sec_limits['shoulder_pan']['min_position']'
when instantiating macro: read_model_data (/home/dongdong/ros2_ws/install/ur_description/share/ur_description/urdf/inc/ur_common.xacro)
instantiated from: ur_robot (/home/dongdong/ros2_ws/install/ur_description/share/ur_description/urdf/ur_macro.xacro)
in file: /home/dongdong/ros2_ws/install/ur_description/share/ur_description/urdf/ur.urdf.xacro
```
Did I miss some parameters? I am not familiar with the xacro structure defined here.

fmauch, 2022-09-24:
Which ROS version are we talking about?

ddliu365, 2022-09-27:
foxy ros 2

fmauch, 2022-09-30:
Sorry, I should have asked that earlier. The Foxy version did not receive that loop of refactoring. It seems to need more arguments.

```
ros2 run xacro xacro $(ros2 pkg prefix ur_description)/share/ur_description/urdf/ur.urdf.xacro \
name:=ur \
ur_type:=ur5e \
joint_limit_params:=$(ros2 pkg prefix ur_description)/share/ur_description/config/ur5e/joint_limits.yaml \
kinematics_params:=$(ros2 pkg prefix ur_description)/share/ur_description/config/ur5e/default_kinematics.yaml \
physical_params:=$(ros2 pkg prefix ur_description)/share/ur_description/config/ur5e/physical_parameters.yaml \
visual_params:=$(ros2 pkg prefix ur_description)/share/ur_description/config/ur5e/visual_parameters.yaml > ur5e.urdf
```

ddliu365, 2022-10-05:
thanks. It works now

closed 2022-10-05 by ddliu365