UniversalRobots/Universal_Robots_ROS2_Driver #485: move_it config setup, can i use for simulation purpose?

enriLoniterp, 2022-09-27:
Hi everyone,
i am currently discovering this repository and in particular i'm interested in simulate _ur5_ in my rviz or gazebo simulation.
It would be fantastic if i would be able to run this this type of code on a universal robots subject: https://moveit.picknik.ai/humble/doc/tutorials/your_first_project/your_first_project.html

I thought it was only copy and paste some files of that directory but it seems something it's not working with the moveit plugin.

Enrico

fmauch, 2022-09-27:
You have a multitude of options to simulate a UR.
* URSim - Use Universal Robot's own robot simulator. That will behave almost exactly as the real thing. Setting this up is only a `ros2 run` [call away](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#usage-with-official-ur-simulator).
* ros2_control mock hardware. Start the driver with a mock hardware interface as described [here](https://docs.ros.org/en/ros2_packages/rolling/api/ur_robot_driver/usage.html#start-hardware-simulator-or-mockup). Note that you should (currently) pass `use_fake_hardware` to the moveit launchfile, as well when doing this.
* use a gazebo simulation from [this package](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation). Note that this will at the moment also require some manual change in some launchfiles as described in [this comment](https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation/issues/14#issuecomment-1258999175). This is indeed the least tested and supported variant of the three.

Unless you have a very specific reason not to do so, I'd suggest using the URSim simulator, as this will leave everything similar to the real hardware and is probably the most tested simulation in this context.

enriLoniterp, 2022-09-27:
First of all i highly thank you for your fast response over my question.

It is important i only ROS 2 and remain more open-source as possible, i suppose the correct solution for me is `use_fake_hardware` as suggested and it's how i think i did:
`ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=true`
it starts the ur5, then i want to use MoveGroup of MoveIt2 as in the example:
```
// Create the MoveIt MoveGroup Interface
using moveit::planning_interface::MoveGroupInterface;
auto move_group_interface = MoveGroupInterface(node, "panda_arm");

// Set a target Pose
auto const target_pose = []{
  geometry_msgs::msg::Pose msg;
  msg.orientation.w = 1.0;
  msg.position.x = 0.28;
  msg.position.y = -0.2;
  msg.position.z = 0.5;
  return msg;
}();
move_group_interface.setPoseTarget(target_pose);

// Create a plan to that target pose
auto const [success, plan] = [&move_group_interface]{
  moveit::planning_interface::MoveGroupInterface::Plan msg;
  auto const ok = static_cast<bool>(move_group_interface.plan(msg));
  return std::make_pair(ok, msg);
}();

// Execute the plan
if(success) {
  move_group_interface.execute(plan);
} else {
  RCLCPP_ERROR(logger, "Planing failed!");
}
```
Something is not working :(, because it fails.

I've also tried to launch:
`ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true
`
but it is not working anyway.

If you don't understand my point, please ask me question, i really need to solve this one!
And thank you very much for your help,
Enrico

fmauch, 2022-09-27:
When you run your own move_group_interface node you'll have to pass the description / semantic robot description to that, as well. Please note that I am not a MoveIt expert and cannot give support for using MoveIt, though. 

```
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true
```
is what should give you a working MoveIt move_group (and this will also be necessary, if you write your own interface node). Beyond that my knowledge and the scope of this repository ends, I'm afraid.

padhupradheep, 2022-09-28:
This [comment](https://github.com/ros-planning/moveit2_tutorials/pull/262#issuecomment-1014927379) might help.

closed 2023-05-09 by fmauch