UniversalRobots/Universal_Robots_ROS2_Driver #487: Custom docker build fails

sampreets3, 2022-09-27:
Hi, 
I am trying to build a docker image by following the tutorial given on the readme. 

I initially tried working inside a container with `docker run -it osrf/ros:humble-desktop-full` and following the steps in the installation instructions. Everything builds, and I am able to launch the `ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5e launch_rviz:=true` launch file, but I do not have any graphics support in this case, and cannot verify that it indeed works.

I tried to create a custom docker image with the dockerfile as shown below:

```dockerfile
FROM osrf/ros:humble-desktop-full

RUN mkdir -p /home/ros2_ws/src && cd /home/ros2_ws && colcon build

RUN sudo apt-get update && sudo apt-get install libxml2 libpoco-dev libasio-dev xterm gdb -y

RUN cd /home/ros2_ws && \
    git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver && \
    vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.${ROS_DISTRO}.repos && \
    rosdep update && rosdep install --ignore-src --from-paths src -y -r
    
RUN cd /home/ros2_ws && \
    ls -R src && \
    . install/setup.sh && \
    colcon build --mixin release 
    
COPY startup_ur.sh /startup_ur.sh
RUN chmod +x /startup_ur.sh
ENTRYPOINT ["/startup_ur.sh"]
CMD ["/bin/bash"]
```
But in this case, the `colcon build` fails with the log:
```sh
Starting >>> forward_command_controller
Starting >>> joint_trajectory_controller
Starting >>> force_torque_sensor_broadcaster
Starting >>> joint_state_broadcaster
Starting >>> ur_dashboard_msgs
Starting >>> ur_description
Starting >>> ur_msgs
Starting >>> ros2_controllers_test_nodes
Starting >>> ur_client_library
Starting >>> diff_drive_controller
Starting >>> imu_sensor_broadcaster
Starting >>> tricycle_controller
Starting >>> gripper_controllers
Starting >>> rqt_joint_trajectory_controller
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'script-dir' will not be supported in future versions. Please use the underscore name 'script_dir' instead
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/dist.py:723: UserWarning: Usage of dash-separated 'install-scripts' will not be supported in future versions. Please use the underscore name 'install_scripts' instead
  warnings.warn(
--- stderr: forward_command_controller
CMake Error at CMakeLists.txt:24 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< forward_command_controller [0.81s, exited with code 1]
--- stderr: joint_trajectory_controller
CMake Error at CMakeLists.txt:27 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< joint_trajectory_controller [0.80s, exited with code 1]
--- stderr: force_torque_sensor_broadcaster
CMake Error at CMakeLists.txt:24 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< force_torque_sensor_broadcaster [0.80s, exited with code 1]
--- stderr: joint_state_broadcaster
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< joint_state_broadcaster [0.79s, exited with code 1]
--- stderr: ur_dashboard_msgs
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< ur_dashboard_msgs [0.79s, exited with code 1]
--- stderr: ur_description
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< ur_description [0.78s, exited with code 1]
Aborted  <<< ur_msgs [0.79s]
Aborted  <<< tricycle_controller [0.77s]
Aborted  <<< imu_sensor_broadcaster [0.77s]
Aborted  <<< gripper_controllers [0.76s]
Aborted  <<< diff_drive_controller [0.78s]
Aborted  <<< ur_client_library [0.97s]
Aborted  <<< ros2_controllers_test_nodes [1.04s]
Aborted  <<< rqt_joint_trajectory_controller [1.01s]

Summary: 0 packages finished [1.37s]
  6 packages failed: force_torque_sensor_broadcaster forward_command_controller joint_state_broadcaster joint_trajectory_controller ur_dashboard_msgs ur_description
  8 packages aborted: diff_drive_controller gripper_controllers imu_sensor_broadcaster ros2_controllers_test_nodes rqt_joint_trajectory_controller tricycle_controller ur_client_library ur_msgs
  11 packages had stderr output: diff_drive_controller force_torque_sensor_broadcaster forward_command_controller gripper_controllers imu_sensor_broadcaster joint_state_broadcaster joint_trajectory_controller tricycle_controller ur_dashboard_msgs ur_description ur_msgs
  10 packages not processed
```
Would appreciate some help on how I can build my own image. 

Thanks in advance !

fmauch, 2022-09-30:
This works for me:
```Dockerfile
FROM osrf/ros:humble-desktop-full

RUN mkdir -p /home/ros2_ws/src && cd /home/ros2_ws && colcon build

RUN sudo apt-get update && sudo apt-get install libxml2 libpoco-dev libasio-dev xterm gdb -y

RUN cd /home/ros2_ws && \
    git clone https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver.git src/Universal_Robots_ROS2_Driver && \
    vcs import src --skip-existing --input src/Universal_Robots_ROS2_Driver/Universal_Robots_ROS2_Driver.${ROS_DISTRO}.repos && \
    rosdep update && rosdep install --ignore-src --from-paths src -y -r
    
RUN cd /home/ros2_ws && \
    . /opt/ros/humble/setup.sh \
    colcon build --mixin release 
    
#COPY startup_ur.sh /startup_ur.sh
#RUN chmod +x /startup_ur.sh
#ENTRYPOINT ["/startup_ur.sh"]
CMD ["/bin/bash"]
```

sampreets3, 2022-09-30:
Hi @fmauch, 
Thanks for the help. I was able to build the docker image successfully with your edit. It seems to me that I forgot to source the correct setup file. 

I'm closing this issue for now while I carry on with my testing.

closed 2022-09-30 by sampreets3