UniversalRobots/Universal_Robots_ROS2_Driver #488: Controlling custom gripper with SetIO

BilboFromBitola, 2022-09-28:
I am trying to control a custom gripper using SetIO from ur_msgs.srv. I am using ROS2 Humble and UR10e robotic arm. I have written a client and a service for this function but I am unsure of what to put as a response in the service side, in order to get the robotic arm to close. As defined in the "SetIO" the response should be a "bool", but if that is the case then, where would the actual change of values for function, pin and state be defined? I would also like to ask if there is an easier or less complicated way of opening and closing the gripper trough "ur drivers" and ROS2, since doing this in ROS1 is much less complicated. Thank you in advance.

fmauch, 2022-09-30:
I think I don't quite understand. SetIO is a service definition. The driver offers this service and can be called by any arbitrary service client (e.g. `ros2 service call` on the command line, custom node written in Python or C++). A service client will have to fill out the service's request and will receive a response from the server. The `bool` flag in the service's response will tell the caller whether the service call actually worked or not.

e.g. this command line call will set pin 3 to state 1 (given a running driver, and an initialized robot in remote_control mode):

```
ros2 service call /io_and_status_controller/set_io ur_msgs/srv/SetIO "{fun: 1, pin: 3, state: 1}"
```

As this is basically the very same concept as the one used in ROS1, I don't understand what you mean by

> doing this in ROS1 is much less complicated.

Sidenote the requirement for the robot having to be in remote_control mode will [soon drop](https://github.com/UniversalRobots/Universal_Robots_Client_Library/pull/111).

closed 2022-09-30 by fmauch

BilboFromBitola, 2022-10-01:
SIr, I beg to differ, I am asking a very concise question. How can I call the built in service in the driver with a CUSTOM client and NOT trough the command line? This is the issue that I am struggling with.

Andrija987, 2022-10-02:
Hi @BilboFromBitola !

I had a look at your problem. If you want to control the selected digital output at UR robot controller i.e. to send digital output signal of 0 or 1 at desired digital output pin at UR robot controller try the following code:

```bash
import rclpy
from rclpy.node import Node

from ur_msgs.srv import SetIO

# old code from ROS 1
# rospy.wait_for_service('/ur_hardware_interface/set_io')
# set_io = rospy.ServiceProxy('/ur_hardware_interface/set_io',SetIO)
# set_io(fun = 1,pin = 3,state = 1)
# rospy.sleep(2)
# set_io(fun = 1,pin = 3,state = 0)

rclpy.init(args=None)

node = Node("io_control")

io_control = node.create_client(SetIO, '/io_and_status_controller/set_io')
while not io_control.wait_for_service(timeout_sec=1.0):
    node.get_logger().info('service not available, waiting again...')
# try this one
io_control(fun = 1, pin = 0, state = 1.0)
# or this one
io_control.call_async(fun = 1,pin = 0,state = 1.0)
# or this one
# io_control.call_async(req.fun = 1, req.pin = 0, req.state = 1.0)

# rclpy.spin_until_future_complete(node, resp)
```

The _io_and_status_controller_ is actually a part of ur `robot_controller`that will always start when you launch the ur driver. Read more here: https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver, than look for _io_and_status_controller_.

Please post here if this solved your problem.