UniversalRobots/Universal_Robots_ROS2_Driver #497: colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release Starting >>> ur_dashboard_msgs Starting >>> ur_moveit_config --- stderr: ur_moveit_config                                           CMake Error at CMakeLists.txt:4 (find_package):   By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH

enriLoniterp, 2022-10-05:
Found this problem when launching from scratch everything in thre README.

```
colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> ur_dashboard_msgs
Starting >>> ur_moveit_config
--- stderr: ur_moveit_config                                          
CMake Error at CMakeLists.txt:4 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< ur_moveit_config [0.36s, exited with code 1]
Aborted  <<< ur_dashboard_msgs [0.37s]

Summary: 0 packages finished [0.54s]
  1 package failed: ur_moveit_config
  1 package aborted: ur_dashboard_msgs
  2 packages had stderr output: ur_dashboard_msgs ur_moveit_config
  5 packages not processed

```

fmauch, 2022-10-05:
probably you never sourced ROS? That's not part of the README, as it is part of installing ROS (see https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html#sourcing-the-setup-script). Could you please confirm that this solves your problem? I would be totally fine with including this into the README.

enriLoniterp, 2022-10-07:
i think this is the problem, now is working and i've sourced my environment

closed 2022-11-23 by fmauch