UniversalRobots/Universal_Robots_ROS2_Driver #500: Error while compiling ur_calibration package

marioesteban11, 2022-10-07:
Hi,

This error shows up when I`m trying to compile this workspace following the README Instructions.

This is the error that shows up when im trying to compile this workspace using; 

# colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

ERROR: 

## --- stderr: ur_calibration                                                                                                                                                                       
/usr/bin/ld: /home/marioest/workspace/ros_ur_driver/install/ur_robot_driver/lib/libur_robot_driver_plugin.so: undefined reference to `urcl::UrDriver::UrDriver(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::function<void (bool)>, bool, std::unique_ptr<urcl::ToolCommSetup, std::default_delete<urcl::ToolCommSetup> >, unsigned int, unsigned int, int, double, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)'
collect2: error: ld returned 1 exit status

I`m using:

Ubuntu 22.04
ROS2 Humble

The objetive is do a pick and place planning using UR and ROS2.

Thanks in advance

enriLoniterp, 2022-10-07:
i have same issue and same configuration as the user aboeve

enriLoniterp, 2022-10-07:
now is working and it compiles everything

marioesteban11, 2022-10-07:
Same for me. I just delete log, build and install from my workspace and compile it again and it works properly.

closed 2022-10-07 by marioesteban11