UniversalRobots/Universal_Robots_ROS2_Driver #502: Humble - controller_stopper_node-3 dies

JSchaaf2, 2022-10-07:
# Description
We try to connect and control an ur10e robot, but when starting the driver we the controller_stopper_node-3 stops working. In rviz the robot model does load in the correct configuration.
It look similar to #467 but this is in humble and still happens.

# My setup
* OS: Ubuntu 22.04.1 
* ROS Distro: Humble
* Binary install (moveit freshly build from source today)

# Command
* ros2 launch ur_robot_driver ur10e.launch.py robot_ip:=$ROBOT_IP

# Result
```bash
...
[ur_ros2_control_node-1] [INFO] [1665131344.158516370] [controller_manager]: Loading controller 'joint_state_broadcaster'
[spawner-6] [INFO] [1665131344.167322827] [spawner_joint_state_broadcaster]: Loaded joint_state_broadcaster
[ur_ros2_control_node-1] [INFO] [1665131344.168809553] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ur_ros2_control_node-1] [INFO] [1665131344.168913586] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[spawner-6] [INFO] [1665131344.176996644] [spawner_joint_state_broadcaster]: Configured and activated joint_state_broadcaster
[ur_ros2_control_node-1] [INFO] [1665131344.182764371] [controller_manager]: Loading controller 'io_and_status_controller'
[spawner-7] [INFO] [1665131344.197534407] [spawner_io_and_status_controller]: Loaded io_and_status_controller
[ur_ros2_control_node-1] [INFO] [1665131344.206610635] [controller_manager]: Configuring controller 'io_and_status_controller'
[spawner-7] [INFO] [1665131344.218934938] [spawner_io_and_status_controller]: Configured and activated io_and_status_controller
[ur_ros2_control_node-1] [INFO] [1665131344.225197671] [controller_manager]: Loading controller 'forward_position_controller'
[ur_ros2_control_node-1] [INFO] [1665131344.233012525] [controller_manager]: Loading controller 'speed_scaling_state_broadcaster'
[spawner-10] [INFO] [1665131344.233646284] [spawner_forward_position_controller]: Loaded forward_position_controller
[spawner-8] [INFO] [1665131344.243646925] [spawner_speed_scaling_state_broadcaster]: Loaded speed_scaling_state_broadcaster
[ur_ros2_control_node-1] [INFO] [1665131344.244120852] [controller_manager]: Loading controller 'scaled_joint_trajectory_controller'
[ur_ros2_control_node-1] [INFO] [1665131344.253453691] [joint_trajectory_controller.interpolation_methods]: No interpolation method parameter was given. Using the default, VARIABLE_DEGREE_SPLINE.
[ur_ros2_control_node-1] [INFO] [1665131344.254731337] [controller_manager]: Loading controller 'force_torque_sensor_broadcaster'
[spawner-11] [INFO] [1665131344.255547406] [spawner_scaled_joint_trajectory_controller]: Loaded scaled_joint_trajectory_controller
[ur_ros2_control_node-1] [INFO] [1665131344.270571515] [controller_manager]: Configuring controller 'forward_position_controller'
[ur_ros2_control_node-1] [INFO] [1665131344.271246546] [forward_position_controller]: configure successful
[ur_ros2_control_node-1] [INFO] [1665131344.271350093] [controller_manager]: Configuring controller 'speed_scaling_state_broadcaster'
[ur_ros2_control_node-1] [INFO] [1665131344.271404466] [speed_scaling_state_broadcaster]: Publisher rate set to : 100.0 Hz
[ur_ros2_control_node-1] [INFO] [1665131344.271979381] [controller_manager]: Configuring controller 'scaled_joint_trajectory_controller'
[ur_ros2_control_node-1] [INFO] [1665131344.272065871] [scaled_joint_trajectory_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ur_ros2_control_node-1] [INFO] [1665131344.272098423] [scaled_joint_trajectory_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ur_ros2_control_node-1] [INFO] [1665131344.272133367] [joint_trajectory_controller.interpolation_methods]: No interpolation method parameter was given. Using the default, VARIABLE_DEGREE_SPLINE.
[ur_ros2_control_node-1] [INFO] [1665131344.272139570] [scaled_joint_trajectory_controller]: Using 'splines' interpolation method.
[ur_ros2_control_node-1] [INFO] [1665131344.272893931] [scaled_joint_trajectory_controller]: Controller state will be published at 100.00 Hz.
[ur_ros2_control_node-1] [INFO] [1665131344.273658131] [scaled_joint_trajectory_controller]: Action status changes will be monitored at 20.00 Hz.
[spawner-9] [INFO] [1665131344.276380959] [spawner_force_torque_sensor_broadcaster]: Loaded force_torque_sensor_broadcaster
[ur_ros2_control_node-1] [INFO] [1665131344.280659383] [controller_manager]: Configuring controller 'force_torque_sensor_broadcaster'
[spawner-8] [INFO] [1665131344.281488813] [spawner_speed_scaling_state_broadcaster]: Configured and activated speed_scaling_state_broadcaster
[spawner-11] [INFO] [1665131344.286829273] [spawner_scaled_joint_trajectory_controller]: Configured and activated scaled_joint_trajectory_controller
[spawner-9] [INFO] [1665131344.289816377] [spawner_force_torque_sensor_broadcaster]: Configured and activated force_torque_sensor_broadcaster
[INFO] [spawner-6]: process has finished cleanly [pid 20934]
[INFO] [spawner-7]: process has finished cleanly [pid 20936]
[INFO] [spawner-10]: process has finished cleanly [pid 20979]
[INFO] [spawner-8]: process has finished cleanly [pid 20938]
[INFO] [spawner-9]: process has finished cleanly [pid 20944]
[INFO] [spawner-11]: process has finished cleanly [pid 20985]
[ERROR] [controller_stopper_node-3]: process has died [pid 20928, exit code -11, cmd '/opt/ros/humble/lib/ur_robot_driver/controller_stopper_node --ros-args -r __node:=controller_stopper --params-file /tmp/launch_params_xir_w6lv --params-file /tmp/launch_params_bn2wclh1 --params-file /tmp/launch_params_fmq37h2r'].
```

fmauch, 2022-10-10:
Do you see any previous error output? It is always helpful to post the full launch output, as in earlier log messages there might be relevant information, not only on the actual error line. Without any further error output, we'd need to debug the controller_stopper crash further.

To do so, you'd need to build the driver from source and run it inside gdb. A first step would be to build the driver from source and see whether the error is reproducible with that (because I haven't been able to reproduce that, so it's not an always-present bug). If it is, we can build the ur_robot_driver package in debug mode and run it inside gdb for a further backtrace.

To build the driver with debug info, go to your colcon workspace and run

```
colcon build --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo  --packages-select ur_robot_driver
```

To run the controller_stopper in gdb, add the following line after [this line](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/63b8e14c1ad32516d7ec6db08cb5b7cc64ee09b4/ur_robot_driver/launch/ur_control.launch.py#L241)
```
prefix=['xterm -e gdb -ex run --args'],
```

This will start the controller_stopper node in an extra window. If it crashes there, type `bt` and hit enter. Please then post the output here.

Thank you!

JSchaaf2, 2022-10-11:
I build from source and that solved the issue. In case it is of interest I did clean up the complete output a bit more and included both the binary and build from source logs. 
Sadly I now run into the same issue as #438, I looked at that but wasn't able to solve it.

<details>
  <summary>binary logs</summary>

  ```bash
  ros2 launch ur_robot_driver ur10e.launch.py robot_ip:=$ROBOT_IP
  [INFO] [launch]: Default logging verbosity is set to INFO
  [INFO] [dashboard_client-2]: process started with pid [63443]
  [INFO] [controller_stopper_node-3]: process started with pid [63445]
  [INFO] [ur_ros2_control_node-1]: process started with pid [63441]
  [INFO] [robot_state_publisher-4]: process started with pid [63447]
  [INFO] [rviz2-5]: process started with pid [63449]
  [INFO] [spawner-6]: process started with pid [63451]
  [INFO] [spawner-7]: process started with pid [63453]
  [INFO] [spawner-8]: process started with pid [63455]
  [INFO] [spawner-9]: process started with pid [63457]
  [INFO] [spawner-10]: process started with pid [63459]
  [INFO] [spawner-11]: process started with pid [63461]
  [rviz2-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
  [controller_stopper_node-3] [INFO] [1665141725.214655679] [Controller stopper]: Waiting for switch controller service to come up on controller_manager/switch_controller
  [ur_ros2_control_node-1] [INFO] [1665141725.242495122] [resource_manager]: Loading hardware 'ur10e' 
  [robot_state_publisher-4] Link base_link had 2 children
  [robot_state_publisher-4] Link base had 0 children
  [robot_state_publisher-4] Link base_link_inertia had 1 children
  [robot_state_publisher-4] Link shoulder_link had 1 children
  [robot_state_publisher-4] Link upper_arm_link had 1 children
  [robot_state_publisher-4] Link forearm_link had 1 children
  [robot_state_publisher-4] Link wrist_1_link had 1 children
  [robot_state_publisher-4] Link wrist_2_link had 1 children
  [robot_state_publisher-4] Link wrist_3_link had 2 children
  [robot_state_publisher-4] Link flange had 1 children
  [robot_state_publisher-4] Link tool0 had 0 children
  [robot_state_publisher-4] Link ft_frame had 0 children
  [robot_state_publisher-4] [INFO] [1665141725.244959749] [robot_state_publisher]: got segment base
  [robot_state_publisher-4] [INFO] [1665141725.245085030] [robot_state_publisher]: got segment base_link
  [robot_state_publisher-4] [INFO] [1665141725.245095952] [robot_state_publisher]: got segment base_link_inertia
  [robot_state_publisher-4] [INFO] [1665141725.245102076] [robot_state_publisher]: got segment flange
  [robot_state_publisher-4] [INFO] [1665141725.245107339] [robot_state_publisher]: got segment forearm_link
  [robot_state_publisher-4] [INFO] [1665141725.245112556] [robot_state_publisher]: got segment ft_frame
  [robot_state_publisher-4] [INFO] [1665141725.245118283] [robot_state_publisher]: got segment shoulder_link
  [robot_state_publisher-4] [INFO] [1665141725.245123880] [robot_state_publisher]: got segment tool0
  [robot_state_publisher-4] [INFO] [1665141725.245129500] [robot_state_publisher]: got segment upper_arm_link
  [robot_state_publisher-4] [INFO] [1665141725.245134877] [robot_state_publisher]: got segment world
  [robot_state_publisher-4] [INFO] [1665141725.245140572] [robot_state_publisher]: got segment wrist_1_link
  [robot_state_publisher-4] [INFO] [1665141725.245146177] [robot_state_publisher]: got segment wrist_2_link
  [robot_state_publisher-4] [INFO] [1665141725.245151839] [robot_state_publisher]: got segment wrist_3_link
  [ur_ros2_control_node-1] [INFO] [1665141725.260809789] [resource_manager]: Initialize hardware 'ur10e' 
  [ur_ros2_control_node-1] [INFO] [1665141725.261713747] [resource_manager]: Successful initialization of hardware 'ur10e'
  [ur_ros2_control_node-1] [INFO] [1665141725.262401802] [resource_manager]: 'configure' hardware 'ur10e' 
  [ur_ros2_control_node-1] [INFO] [1665141725.262813194] [resource_manager]: Successful 'configure' of hardware 'ur10e'
  [ur_ros2_control_node-1] [INFO] [1665141725.263258918] [resource_manager]: 'activate' hardware 'ur10e' 
  [ur_ros2_control_node-1] [INFO] [1665141725.263700556] [URPositionHardwareInterface]: Starting ...please wait...
  [ur_ros2_control_node-1] [INFO] [1665141725.264129849] [URPositionHardwareInterface]: Initializing driver...
  [ur_ros2_control_node-1] Warning: No realtime capabilities found. Consider using a realtime system for better performance
  [ur_ros2_control_node-1]          at line 396 in ./include/ur_client_library/comm/pipeline.h
  [spawner-11] [INFO] [1665141725.466917643] [spawner_scaled_joint_trajectory_controller]: Waiting for '/controller_manager' node to exist
  [spawner-7] [INFO] [1665141725.472303745] [spawner_io_and_status_controller]: Waiting for '/controller_manager' services to be available
  [spawner-10] [INFO] [1665141725.477918425] [spawner_forward_position_controller]: Waiting for '/controller_manager' services to be available
  [spawner-8] [INFO] [1665141725.486147805] [spawner_speed_scaling_state_broadcaster]: Waiting for '/controller_manager' node to exist
  [spawner-6] [INFO] [1665141725.492462916] [spawner_joint_state_broadcaster]: Waiting for '/controller_manager' services to be available
  [spawner-9] [INFO] [1665141725.497533681] [spawner_force_torque_sensor_broadcaster]: Waiting for '/controller_manager' node to exist
  [rviz2-5] [INFO] [1665141725.527426982] [rviz2]: Stereo is NOT SUPPORTED
[... cut ...]
[spawner-7] [INFO] [1665412266.149468059] [spawner_io_and_status_controller]: Configured and activated io_and_status_controller
[INFO] [spawner-8]: process has finished cleanly [pid 20137]
[INFO] [spawner-6]: process has finished cleanly [pid 20131]
[INFO] [spawner-10]: process has finished cleanly [pid 20161]
[INFO] [spawner-11]: process has finished cleanly [pid 20187]
[INFO] [spawner-9]: process has finished cleanly [pid 20139]
[INFO] [spawner-7]: process has finished cleanly [pid 20133]
```
</details>

fmauch, 2022-10-11:
@JSchaaf2 did you try the solution mentioned there to install `ros-humble-backward-ros`?

I just checked again and on my testing container a binary installation seems to work fine.

JSchaaf2, 2022-10-11:
Yes but got `ros-humble-backward-ros is already the newest version (1.0.2-3jammy.20220519.235224).`. Or should i remove my ros install before doing the backward install?

About the binary install: we have 3 computers 2 identical ones with my issue, the third one does not have dying controller issue.

fmauch, 2022-10-11:
> Yes but got ros-humble-backward-ros is already the newest version (1.0.2-3jammy.20220519.235224).. Or should i remove my ros install before doing the backward install?

No. But that means that this is probably not the source of #438 after all

> About the binary install: we have 3 computers, 2 identical ones with my issue, the third one does not have dying controller issue.

This is strange. Are they all up to date? Without being able to reproduce this, it is hard to find the source of the problem...

JSchaaf2, 2022-10-13:
> This is strange. Are they all up to date? Without being able to reproduce this, it is hard to find the source of the problem...

As far as i can tell everything is up to date. I did a fresh build of moveit before making this issue, maybe something went wrong there, but I am not (yet) having any issues elsewhere. I do understand the problems with trying to debug something you can't reproduce.

It does seem that if we install everything using the binary installs the issues don't happen. I did not test it on my specific system but have others where it does work. I guess I will close the issue for now.

Thanks for your help

closed 2022-10-14 by JSchaaf2

JSchaaf2, 2022-10-27:
In case other ppl end up here, yes installing everything from binary did solve the issues. I still don't know what caused the issues though.